gphoto/libgphoto2 #652: Canon 5D IV can not save picture on sd card

mailbaoer, 2021-04-21:
**Describe the bug**

no picture saved on camera sd card when use `gp_camera_wait_for_event` method, it only send picture to my phone

**Name the camera**
Canon 5D Mark IV

**libgphoto2 and gphoto2 version**
libgphoto2 2.5.26

**To Reproduce**
Steps to reproduce the behavior:
I use `gp_camera_wait_for_event` method to receive GP_EVENT_FILE_ADDED event, then download the capture file to my phone, but the camera sd card has no file saved, how can I get the camera capture event and let picture saved on camera sd card. I've try to set `capturetarget` but it has no effect, maybe I have not set it correctly.

msmeissn, 2021-04-21:
where is capturetarget set to ?  can you try setting it to card ?

mailbaoer, 2021-04-21:
Thank you for your quick reply ！I refered the code here `https://github.com/gphoto/libgphoto2/blob/031e3a1a4b1adbd42dbcc5f74e9089c92f1006c3/examples/preview.c#L30`, I have tried to set capturetarget to 0 or 1,but has not effect， so I set the capturetarget wrong way？

mailbaoer, 2021-04-23:
I found the method to set `capturetarget` correctly, I use `gp_setting_set(camera, context, 'card')`, It worked! bringing us such a good solution!

closed 2021-04-23 by mailbaoer

msmeissn, 2021-04-24:
very good! thx for feedback!

SiuyuenChow, 2021-06-27:
> I found the method to set `capturetarget` correctly, I use `gp_setting_set(camera, context, 'card')`, It worked! Thank you for bringing us such a good solution !

Maybe it's the same question you're responding to

GP_SETTING_SET (CAMERA, CONTEXT, 'CARD') = CAMERA *, CONTEXT = GPCONTEXT *?

After I connected the camera and called gp_camera_wait_for_event, the photo could not be written and the indicator light of the camera did not flicker when I took the picture

Set gp_setting_set("gphoto2", "--set-config", "capturetarget=1 "); Not much use

Could you please tell me if you have the same specific solution as me

thanks

mailbaoer, 2021-06-29:
you can try `int result = gp_setting_set("ptp2", "capturetarget", "card");` @SiuyuenChow