gphoto/libgphoto2 #655: Canon EOS Full-Press failed (0x2019: PTP Device Busy)

emc2studiofr, 2021-04-29:
Hello,
On minisforum PC with Ubuntu 20.04

When I try to use 
`env LANG=C gphoto2 --debug --debug-logfile=gphoto2.log --capture-image-and-download --force-overwrite --filename /home/frizbox/photo.jpg
`
after 10/11 seconds I've got this error message 
```
*** Error ***
Canon EOS Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-110: 'I/O in progress') ***
```

**Name the camera**
`Canon EOS 1500D                usb:002,012`
But it's a Canon EOS **2000D**

**libgphoto2 and gphoto2 version**
`gphoto2         2.5.23         gcc, popt(m), exif, cdk, aa, jpeg, readline`
`libgphoto2      2.5.25         standard camlibs (SKIPPING lumix), gcc, ltdl, EXIF`
`libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking`

The debug file is here : 
[gphoto2.log](https://github.com/gphoto/libgphoto2/files/6400155/gphoto2.log)

Thanks

Guillaume

hfiguiere, 2021-04-29:
> **Name the camera**
> `Canon EOS 1500D usb:002,012`
> But it's a Canon EOS **2000D**

Blame Canon for naming cameras differently. They are the same. Also Rebel T7.

msmeissn, 2021-04-30:
there is a single jump with a 10 second timeout just before this error.

0.449403 ptp_usb_getresp             (2): Reading PTP_OC 0x9128 (EOS_RemoteReleaseOn) response...
0.449416 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
10.447048 gp_port_read                (3): Read    16 = 0x10 out of 1024 bytes from port: (hexdump of 16 bytes)
0000  10 00 00 00 03 00 19 20-20 00 00 00 00 00 00 00  .......  .......

this is presing the shutter full until the camera reacts.

- Do you have "manual focus" enabled? Is it better with autofocusing enabled?
- can you try to add --wait-event=3s  before the --capture-image-and-download    ?

emc2studiofr, 2021-04-30:
> * Do you have "manual focus" enabled? Is it better with autofocusing enabled?

Manual because it seems faster 

> * can you try to add --wait-event=3s  before the --capture-image-and-download    ?

indeed ` --wait-event=500ms` works perfect !

Thanks

closed 2021-04-30 by emc2studiofr