gphoto/libgphoto2 #656: Nikon Coolpix P510 (PTP mode): No Image Capture

schneider8357, 2021-04-30:
I can't capture image using my Nikon Coolpix P510.
Thanks in advance for any help.

<details><summary>gphoto2 --summary --debug</summary>

<p>

```
0.000173 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000210 main                        (2): gphoto2 2.5.23
0.000219 main                        (2): gphoto2 has been compiled with the following options:
0.000224 main                        (2):  + gcc (C compiler used)
0.000228 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000232 main                        (2):  + exif (for displaying EXIF information)
0.000237 main                        (2):  + cdk (for accessing configuration options)
0.000246 main                        (2):  + aa (for displaying live previews)
0.000251 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000255 main                        (2):  + readline (for easy navigation in the shell)
0.000275 main                        (2): libgphoto2 2.5.25
0.000282 main                        (2): libgphoto2 has been compiled with the following options:
0.000287 main                        (2):  + standard camlib set SKIPPING SOME (ax203 canon digigr8 dimagev directory jl2005a jl2005c kodak_dc240 mars pentax ptp2 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801 SKIPPING lumix)
0.000297 main                        (2):  + gcc (C compiler used)
0.000306 main                        (2):  + ltdl (for portable loading of camlibs)
0.000314 main                        (2):  + EXIF (for special handling of EXIF files)
0.000346 main                        (2): libgphoto2_port 0.12.0
0.000354 main                        (2): libgphoto2_port has been compiled with the following options:
0.000359 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.000364 main                        (2):  + gcc (C compiler used)
0.000370 main                        (2):  + ltdl (for portable loading of iolibs)
0.000378 main                        (2):  + EXIF (for vusb)
0.000386 main                        (2):  + USB (libusb1, for USB cameras)
0.000398 main                        (2):  + serial (for serial cameras)
0.000405 main                        (2):  + no resmgr (serial port access and locking)
0.000416 main                        (2):  + no ttylock (serial port locking)
0.000422 main                        (2):  + no lockdev (serial port locking)
0.000427 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000433 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000438 main                        (2): invoked with following arguments:
0.000443 main                        (2):   --summary
0.000448 main                        (2):   --debug
0.000468 load_settings               (2): Creating gphoto config directory ('/root/.gphoto')
0.000524 load_settings               (2): Loading settings from file '/root/.gphoto/settings'.
0.000666 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.000678 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/lib/x86_64-linux-gnu/libgphoto2_port/0.12.0'...
0.000771 foreach_func                (2): Called for filename '/usr/lib/x86_64-linux-gnu/libgphoto2_port/0.12.0/disk'.
0.001071 gp_port_library_list        (2): found fstab fsname UUID=4076709f-5f8e-4573-b118-650f77036960
0.001094 gp_port_library_list        (2): found fstab fsname UUID=8244-6C13
0.001113 gp_port_library_list        (2): found fstab fsname /swapfile
0.001216 gp_port_library_list        (2): found mtab fsname sysfs
0.001227 gp_port_library_list        (2): found mtab fsname proc
0.001234 gp_port_library_list        (2): found mtab fsname udev
0.001270 gp_port_library_list        (2): found mtab fsname devpts
0.001292 gp_port_library_list        (2): found mtab fsname tmpfs
0.001315 gp_port_library_list        (2): found mtab fsname /dev/sda2
0.001334 gp_port_library_list        (2): found mtab fsname securityfs
0.001346 gp_port_library_list        (2): found mtab fsname tmpfs
0.001376 gp_port_library_list        (2): found mtab fsname tmpfs
0.001395 gp_port_library_list        (2): found mtab fsname tmpfs
0.001417 gp_port_library_list        (2): found mtab fsname cgroup2
0.001427 gp_port_library_list        (2): found mtab fsname cgroup
0.001437 gp_port_library_list        (2): found mtab fsname pstore
0.001458 gp_port_library_list        (2): found mtab fsname efivarfs
0.001515 gp_port_library_list        (2): found mtab fsname none
0.001537 gp_port_library_list        (2): found mtab fsname cgroup
0.001547 gp_port_library_list        (2): found mtab fsname cgroup
0.001558 gp_port_library_list        (2): found mtab fsname cgroup
0.001568 gp_port_library_list        (2): found mtab fsname cgroup
0.001578 gp_port_library_list        (2): found mtab fsname cgroup
0.001589 gp_port_library_list        (2): found mtab fsname cgroup
0.001599 gp_port_library_list        (2): found mtab fsname cgroup
[... 716 lines cut ...]
0.237849 _cb_irq                     (2): 0x564442351af0 with status 3
0.237856 _cb_irq                     (2): Transfer 0x564442351af0 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
0.237864 _cb_irq                     (2): 0x564442368ee0 with status 3
0.237871 _cb_irq                     (2): Transfer 0x564442368ee0 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
0.238080 gp_filesystem_reset         (2): resetting filesystem
0.238089 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
0.238094 gp_filesystem_lru_clear     (2): fscache LRU list already empty
0.238100 delete_all_folders          (2): Internally deleting all folders from '/'...
0.238105 lookup_folder               (2): Lookup folder '/'...
0.238111 lookup_folder               (2): Found! / is 0x56444233a8c0
0.238117 recurse_delete_folder       (2): Recurse delete folder 0x56444233a8c0//
0.238124 delete_all_files            (2): Delete all files in folder 0x56444233a8c0//
0.238130 gp_port_free                (2): Freeing port...
0.238137 gp_port_close               (2): Closing port...
0.238363 gp_filesystem_reset         (2): resetting filesystem
0.238378 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
0.238382 gp_filesystem_lru_clear     (2): fscache LRU list already empty
0.238386 delete_all_folders          (2): Internally deleting all folders from '/'...
0.238391 lookup_folder               (2): Lookup folder '/'...
0.238396 lookup_folder               (2): Found! / is 0x56444233a8c0
0.238400 recurse_delete_folder       (2): Recurse delete folder 0x56444233a8c0//
0.238405 delete_all_files            (2): Delete all files in folder 0x56444233a8c0//
```
</p>
</details>

msmeissn, 2021-05-01:
It does not advertise any picture capture capabilities.

Just for kicks, can you try

gphoto2 --set-config opcode=0x100e,0,0

and see if some capture thing happens?

schneider8357, 2021-05-01:
Thanks for the quick reply!

No luck with that either:
```
*** Error ***              
Failed to set new configuration value 0x100e,0,0 for configuration entry opcode.
*** Error (-2: 'Bad parameters') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --set-config opcode=0x100e,0,0

Please make sure there is sufficient quoting around the arguments.
```

Do I have any hope of getting the Image Capture to work? Or I will be obligated to install Windows? :sweat_smile:

msmeissn, 2021-05-04:
Any proof it would work under Windows? ;)

can you also try:
gphoto2 --set-config opcode=0x90cb
gphoto2 --set-config opcode=0x90c0,0xffffffff

but I fear neither will work.

100e is the generic PTP capture which nikons support, the 90cb and 90c0 are 2 nikon capture methods

schneider8357, 2021-05-04:
Yeah, no luck at all with those. I did test on Windows and guess what? You're totally right. It does not work as I expected it would. 
Oh well. Guess I'm going to need a new camera :disappointed: 

Thanks a lot for the help, though!

closed 2021-05-04 by schneider8357