gphoto/libgphoto2 #657: New camera support - in a difference sense

chri2, 2021-05-10:
I intentionally misunderstood the category "New camera support" as support for buying a new camera. I hope this interpretation can be **smile**d at - **if not, please close this issue and forgive me**.

I'd like to buy a camera to exclusively use with linux and libgphoto2 seems to be the software to go with.

I'm looking for help to choose a camera which meets the following specification (ordered by priority):

- **manufacturer does directly support (lib)gphoto2 for supporting their models (buying the camera ideally would support this project)**
- picture download by wifi or bluetooth (when I get near my computer I'd like to be able to download pictures wireless without opening the camera case automatically or with some simple commands)
- shooting macros (mostly documentation purposes, no high quality is needed, could be an additional lens to fit on the camera when needed)
- compact, small (not to big to carry while biking, running, skiing)
- protected (dust, shock, humidity, water)
- I guess I do not have to say that I'd be happy to have a decent quality, but the above really is more important (I assume that most cameras sold today do a lot better than a pinephone)

I'd be really thankful for some personal hands-on experience.

The list of cameras generally is very helpful, but I didn't understand how to extract the information about being able to download pictures stored on the camera via wifi/bluetooth and if there are manufacturers supporting this product it would be helpful information for people looking for a camera.

msmeissn, 2021-05-16:
hubert closed it without comment, but let me add some comment here:

Currently no vendor supports libgphoto2 directly, either money or information wise.

The best information available for cameras regarding libgphoto2 I consider Nikon DSLR and Z series, as Niikon publishes their PTP documentation.

msmeissn, 2021-05-16:
hubert closed it without comment, but let me add some comment here:

Currently no vendor supports libgphoto2 directly, either money or information wise, so your point 1 cannot be met.
For the rest it is always tricky there, not sure we can really help much.

The best information available for cameras regarding libgphoto2 I consider Nikon DSLR and Z series, as Niikon publishes their PTP documentation.

chri2, 2021-06-08:
Thanks for the comment! I looked into the camera program of Nikon again, but didn't find a camera that does have the features or at least some of them I'm looking for. Sad that there is no support from camera makers.