gphoto/libgphoto2 #658: Fujifilm X-M1 can't capture and set shtterspeed

ikbumheo, 2021-05-10:
Dear

I am using Fuji X-M1. I can detect Fuji X-M1 and get the files from X-M1
$gphoto2 --auto-detect
$gphoto2 --get-all-files

But I can't capture or set shutterspeed of Fuji x-m1 with gphoto2. can you check about that?

*** Error ***              
Sorry, your camera does not support generic capture
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***       

*** Error ***              
shutterspeed not found in configuration tree.
*** Error (-1: 'Unspecified error') ***  

Thanks for all your support

ikbumheo, 2021-05-11:
Hi 

I find out This old Fuji X-M1 support just USB mass storage but not PTP. 
Therefore, X-M1 can't capture the image and can't set shutter speed because it can't use PTP.
If it it right, please confirm?

Thanks

ikbumheo, 2021-05-12:
I'm understood.
Thanks for your support

msmeissn, 2021-05-15:
Only some of the higher price Fujis allow remote capture.

X-T<signledigit> and some others.

With the Fuji Webcam software some more allow liveview streaming.

Can you also check that you have current libgphoto2 2.5.27 installed?

And perhaps provide a logfile of above try?

gphoto2 --capture-image --debug --debug-logfile=capture.log

msmeissn, 2021-06-19:
feel free to reopen if there is more input

closed 2021-06-19 by msmeissn

auesro, 2023-02-01:
Would it be possible, at least in theory, to use gphoto to just fire the shuter of the so far non-supported Fuji cameras (thinking specifically of X-A1, X-M1 and X-T100?

msmeissn, 2023-02-01:
they do not offer that, usually if they would offer shutter firing, they would also allow more remote control and downloading

auesro, 2023-02-01:
However, I have a intervalometer which can trigger the shutter through the USB connector...so surely it is possible to send a signal to the USB port to take a picture, right? Maybe I am missing somehting...

msmeissn, 2023-02-01:
usually this is some electrical level trigger, not a usb level trigger, depending on camera. I would expect it not to be USB like.

auesro, 2023-02-01:
Specifically speaking about the X-A1, X-M1 and X-T100, they are triggered through a micro USB plug...but maybe thats only the physical form of the connector and they do not have real USB capabilities... however, I can connect the camera (X-A1 right now) directly through that USB to my computer and see the internal or SD card content (Windows 11 here). Indeed, Windows 11 just showed "PTP camera ready to use"...