gphoto/libgphoto2 #660: "Device busy" error after 32 captures with Fuji XT-3

Prototypers-lair, 2021-05-12:
I have my Fujifilm XT-3 set up to photograph time-lapses on Octolapse using a Gphoto2 script. The setup works, but after exactly 32 captures, I get an error saying the device is busy. I've varied the command, interval timer, and image quality. I've tried capturing and downloading, with and without --keep, I've tried only capturing, and I've tried the trigger-capture option. I've tried capturing by hand, and with the script described on [this page](https://github.com/FormerLurker/Octolapse/wiki/Configuring-an-External-Camera) I always get this error after 32 successful captures:

2021-05-12 18:13:41,737 - octolapse.script - ERROR - Error output (stderr) for 'Fuji XT-3 - Snapshot Camera Script':
	
	*** Error ***              
	PTP Device Busy
	ERROR: Could not capture image.
	ERROR: Could not capture.
	*** Error (-110: 'I/O in progress') ***

I'm not sure if this is a bug in libgphoto2, or a bug in the camera.

**Name the camera**
Fujifilm XT-3

**libgphoto2 and gphoto2 version**
gphoto2         2.5.27         gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.27         standard camlibs (SKIPPING lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

**To Reproduce**

Perform >32 captures from command line or script with the combination of the above versions, and a Fujifilm XT-3 ruinning 4.1 firmware.

smaun92, 2022-05-10:
@msmeissn 
Hi Marcus,

Its me again. I have a Fujifilm GFX100s 

Model                          Port                                            
----------------------------------------------------------
Fuji Fujifilm GFX 100S (2nd)   usb:001,014     

I have the following versions:

gphoto2         2.5.28         gcc, popt(m), exif, cdk, no aa, jpeg, no readline
libgphoto2      2.5.28         standard camlibs (SKIPPING docupen lumix), gcc, no ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking

Its working like a charm. however it can not take more than 10 image in one go either with --capture-image or --trigger-capture. I get the same "PTP Device Busy" error

*** Error ***              
PTP Device Busy
ERROR: Could not trigger capture.
*** Error (-110: 'I/O in progress') ***   

**This is my observation that as soon first image is taken, camera goes in to busy state. which can be seen from Blinking orange and green indicator lamp. I know this could be because of the camera as it is in a busy state, but is there any way I can disconnect the camera when its not taking picture**

I am using the following code to mimic AEB as it changes the shutter speed after taking each picture

`# AEB

import RPi.GPIO as GPIO

import logging
import os
import subprocess
import sys
import gphoto2 as gp

GPIO.setmode(GPIO.BCM)
GPIO.setup(22, GPIO.IN)
GPIO.setup(22, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)

while True:
    if GPIO.input(22):

        from subprocess import call
        call(["gphoto2", "--trigger-capture"])
        call(["gphoto2", "--set-config", "shutterspeed=30s"])
        call(["gphoto2", "--trigger-capture"])
        call(["gphoto2", "--set-config", "shutterspeed=15s"])
        call(["gphoto2", "--trigger-capture"])
        call(["gphoto2", "--set-config", "shutterspeed=8s"])
        call(["gphoto2", "--trigger-capture"])
        call(["gphoto2", "--set-config", "shutterspeed=4s"])
        call(["gphoto2", "--trigger-capture"])
        call(["gphoto2", "--set-config", "shutterspeed=1s"])
        call(["gphoto2", "--trigger-capture"])
        call(["gphoto2", "--set-config", "shutterspeed=1/2"])
        call(["gphoto2", "--trigger-capture"])
        call(["gphoto2", "--set-config", "shutterspeed=1/4"])
        call(["gphoto2", "--trigger-capture"])
        call(["gphoto2", "--set-config", "shutterspeed=1/8"])
        call(["gphoto2", "--trigger-capture"])
        call(["gphoto2", "--set-config", "shutterspeed=2s"])`

msmeissn, 2023-11-10:
i think you will need to download the images always immediately, as they are stored in camera RAM and that runs full.