gphoto/libgphoto2 #663: --trigger-capture delay before capture

danielsvane, 2021-05-18:
**Describe the bug**

When using `--trigger-capture` on my Canon m200, it takes between 1.4 and 8 seconds for a capture. The log file shows a rather large delay around saving some settings and initializing camera:
[image]
The camera is in manual mode, manual focus.

Can I do anything to make the capture faster?

**Name the camera**
Canon m200, displayed as "USB PTP Class Camera"

**libgphoto2 and gphoto2 version**
gphoto2: 2.5.20
libgphoto2: 2.5.22
libgphoto2_port: 0.12.0

**To Reproduce**
Attach a Canon m200 and run `gphoto2 --trigger-capture`
[debug.log](https://github.com/gphoto/libgphoto2/files/6501613/debug.log)

danielsvane, 2021-05-18:
I somewhat solved this, even though I'm still not sure what causes the long delay when opening the connection.

To take a picture instantly I start a process with `gphoto2 --capture-image -I -1`. Calling `kill -USR1 PID` with the PID of the started process then takes the picture instantly. This is maybe the way it's intended.

You can close this issue if the startup time is normal.

msmeissn, 2021-05-23:
the quoted 1.4s delay is really before we do anything relevant... is the harddisk access to ~/.gphoto/settings over remote fs or so?
it likely is a write delay on this file.

msmeissn, 2021-06-19:
likely solved?