gphoto/libgphoto2 #666: gphoto2 --capture-image An error occurred in the io-library ('Timeout reading from or writing to the port'): No error description available

yanncarlier, 2021-06-06:
**Describe the bug**

A clear and concise description of what the bug is.

gphoto2 --capture-image
An error occurred in the io-library ('Timeout reading from or writing to the port'): No error description available

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs

Sony Alpha-A9 (Control)        usb:001,003     

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version
gphoto2 2.5.23

Copyright (c) 2000-2019 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.25         standard camlibs (SKIPPING lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

**To Reproduce**
Steps to reproduce the behavior:

Cant: gphoto2 --capture-image

When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.

0.000339 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000439 main                        (2): gphoto2 2.5.23
0.000459 main                        (2): gphoto2 has been compiled with the following options:
0.000471 main                        (2):  + gcc (C compiler used)
0.000481 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000492 main                        (2):  + exif (for displaying EXIF information)
0.000502 main                        (2):  + cdk (for accessing configuration options)
0.000512 main                        (2):  + aa (for displaying live previews)
0.000522 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000531 main                        (2):  + readline (for easy navigation in the shell)
0.000554 main                        (2): libgphoto2 2.5.25
0.000595 main                        (2): libgphoto2 has been compiled with the following options:
0.000623 main                        (2):  + standard camlib set SKIPPING SOME (ax203 canon digigr8 dimagev directory jl2005a jl2005c kodak_dc240 mars pentax ptp2 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801 SKIPPING lumix)
0.000639 main                        (2):  + gcc (C compiler used)
0.000654 main                        (2):  + ltdl (for portable loading of camlibs)
0.000669 main                        (2):  + EXIF (for special handling of EXIF files)
0.000696 main                        (2): libgphoto2_port 0.12.0
0.000711 main                        (2): libgphoto2_port has been compiled with the following options:
0.000721 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.000731 main                        (2):  + gcc (C compiler used)
0.000741 main                        (2):  + ltdl (for portable loading of iolibs)
0.000751 main                        (2):  + EXIF (for vusb)
0.000761 main                        (2):  + USB (libusb1, for USB cameras)
0.000771 main                        (2):  + serial (for serial cameras)
0.000781 main                        (2):  + no resmgr (serial port access and locking)
0.000791 main                        (2):  + no ttylock (serial port locking)
0.000801 main                        (2):  + no lockdev (serial port locking)
0.000812 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000822 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000833 main                        (2): invoked with following arguments:
0.000843 main                        (2):   --capture-image
0.000854 main                        (2):   --debug
0.000905 load_settings               (2): Creating gphoto config directory ('/root/.gphoto')
0.001020 load_settings               (2): Loading settings from file '/root/.gphoto/settings'.
0.001092 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.001113 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/lib/x86_64-linux-gnu/libgphoto2_port/0.12.0'...
[... 447 lines cut ...]
48.109497 _cb_irq                     (2): 0x5651b3fce9a0 with status 3
48.109519 _cb_irq                     (2): Transfer 0x5651b3fce9a0 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
48.109549 _cb_irq                     (2): 0x5651b3fe0d10 with status 3
48.109572 _cb_irq                     (2): Transfer 0x5651b3fe0d10 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
48.109603 _cb_irq                     (2): 0x5651b3fe3210 with status 3
48.109626 _cb_irq                     (2): Transfer 0x5651b3fe3210 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
48.109657 _cb_irq                     (2): 0x5651b3fce170 with status 3
48.109744 _cb_irq                     (2): Transfer 0x5651b3fce170 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
48.109838 _cb_irq                     (2): 0x5651b3fd4560 with status 3
48.109956 _cb_irq                     (2): Transfer 0x5651b3fd4560 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
48.111953 gp_context_error            (0): An error occurred in the io-library ('Timeout reading from or writing to the port'): No error description available

*** Error ***              
An error occurred in the io-library ('Timeout reading from or writing to the port'): No error description available
48.114015 gp_camera_free              (2): Freeing camera...
48.114077 gp_port_free                (2): Freeing port...
48.114099 gp_port_close               (2): Closing port...
48.114725 gp_filesystem_reset         (2): resetting filesystem
48.114758 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
48.114772 gp_filesystem_lru_clear     (2): fscache LRU list already empty
48.114785 delete_all_folders          (2): Internally deleting all folders from '/'...
48.114800 lookup_folder               (2): Lookup folder '/'...
48.114815 lookup_folder               (2): Found! / is 0x5651b3fc58c0
48.114829 recurse_delete_folder       (2): Recurse delete folder 0x5651b3fc58c0//
48.114843 delete_all_files            (2): Delete all files in folder 0x5651b3fc58c0//

msmeissn, 2021-06-20:
logfile shows: camera does not react to any command sent, only times out. hard to say where the reason is.

is this the first run after plugging it in? perhaps also another process gvfs blocking it, even though its not reported.

gillohner, 2021-10-17:
Having the same issue with my Alpha-6500.

jtorrex, 2024-02-14:
Having the same issue here: not sure if it is already solved or not.

```
8.129636 ptp                         (2): PTP: Opening session
8.129641 ptp_usb_sendreq             (2): Sending PTP_OC 0x1002 (Open session) (0x1) request...
8.129644 gp_port_write               (3): Writing 16 = 0x10 bytes to port...
16.129815 gp_libusb1_write [libusb1.c:578](0): 'libusb_bulk_transfer (port->pl->dh, port->settings.usb.outep, (unsigned char*)bytes, size, &curwritten, port->timeout)' failed: Operation timed out (-7)
16.129842 gp_port_write [gphoto2-port.c:405](0): Writing 16 = 0x10 bytes to port failed: Timeout reading from or writing to the port (-10)
16.129846 ptp_usb_sendreq [usb.c:109] (0): PTP_OC 0x1002 sending req failed: Timeout reading from or writing to the port (-10)
16.129849 camera_init [library.c:9864](0): 'ptp_opensession (params, sessionid)' failed: PTP Timeout (0x02fa)
16.129866 ptp                         (2): PTP: Opening session
16.129870 ptp_usb_sendreq             (2): Sending PTP_OC 0x1002 (Open session) (0x1) request...
16.129871 gp_port_write               (3): Writing 16 = 0x10 bytes to port...
24.129992 gp_libusb1_write [libusb1.c:578](0): 'libusb_bulk_transfer (port->pl->dh, port->settings.usb.outep, (unsigned char*)bytes, size, &curwritten, port->timeout)' failed: Operation timed out (-7)
24.130014 gp_port_write [gphoto2-port.c:405](0): Writing 16 = 0x10 bytes to port failed: Timeout reading from or writing to the port (-10)
24.130018 ptp_usb_sendreq [usb.c:109] (0): PTP_OC 0x1002 sending req failed: Timeout reading from or writing to the port (-10)
24.130021 camera_init [library.c:9864](0): 'ptp_opensession (params, sessionid)' failed: PTP Timeout (0x02fa)
24.130022 camera_init [library.c:9892](0): 'ret' failed: 'PTP Timeout' (0x02fa)
24.130025 gp_context_error            (0): PTP Timeout
24.130036 gp_port_close               (2): Closing port...
```

jacket, 2025-03-04:
I also have this problem.

I am using gphoto2 on Ubuntu 24.04 and Google Pixel 8 Pro.

The device is connected by USB 2.0 and on Ubuntu I can browse the filesystem of the phone.

I can use gphoto2 to detect the camera:

`user@my-laptop-name:~/Desktop$ gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Google Inc Nexus/Pixel (MTP)   usb:001,006`

But when I try to get information or do anything with the device, I get the following error:

`user@my-laptop-name:~/Desktop$ gphoto2 --summary
                                                                               
*** Error ***              
PTP Timeout

*** Error ***              
An error occurred in the io-library ('Timeout reading from or writing to the port'): No error description available
*** Error (-10: 'Timeout reading from or writing to the port') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --summary

Please make sure there is sufficient quoting around the arguments.`

I have tried restarting the laptop; restarting the phone; changing the USB cable to the one that came with the phone; changing to a second laptop. In all cases, I get the same error above.