gphoto/libgphoto2 #670: gp_camera_wait_for_event make camera busy

SiuyuenChow, 2021-06-23:
Hi,
It's a great project

I have some questions to ask for help,

When polling for a new file, the camera has pressed the shutter to take a picture, the number of photos has not been added, nor has it received a return from GP_EVENT_FILE_ADDED. The camera seems to be busy all the time, and the camera model is EOS 6D

 while (1) {
        retval = gp_camera_wait_for_event (camera, 1, &evttype, &evtdata, context);
}

Thanks

SiuyuenChow, 2021-06-24:
When I poll the gp_camera_wait_for_event method,
Press the shutter of the camera to take a picture. There is a sound of the camera taking a picture, but the indicator light is not flashing. Look at the folder, and there is no new picture taken

SiuyuenChow, 2021-06-24:
15 seconds after I have successfully connected the camera, call ret = gp_camera_init (camera, context);
The camera is not added to the photo, and then follow up by pressing the shutter to take the photo and there is no new photo
So here's the log that's printed out

camera_trigger_capture
Triggering capture to , autofocus=0
camera_trigger_canon_eos_capture
prepare_capture
preparing EOS capture...
Sending PTP_OC 0x9114 (EOS_SetRemoteMode) (0x1) request...
Reading PTP_OC 0x9114 (EOS_SetRemoteMode) response...
PTPIP_CMD_RESPONSE
Sending PTP_OC 0x9115 (EOS_SetEventMode) (0x1) request...
Reading PTP_OC 0x9115 (EOS_SetEventMode) response...
PTPIP_CMD_RESPONSE
Sending PTP_OC 0x913d (EOS_SetRequestOLCInfoGroup) (0x1fff) request...
Reading PTP_OC 0x913d (EOS_SetRequestOLCInfoGroup) response...
PTPIP_CMD_RESPONSE
Sending PTP_OC 0x9116 (EOS_GetEvent) request...
Reading PTP_OC 0x9116 (EOS_GetEvent) data...
Reading PTP_OC 0x9116 (EOS_GetEvent) response...
PTPIP_CMD_RESPONSE
event 0: unknown EOS event c1f4
    8: 02
    9: 00
    10: 00
    11: 00
event 1: EOS prop d194 desc record, datasize 16, propxtype 3
event 1: propdesc d194, default value not found.
event 2: EOS prop d1a3 info record, datasize is 4
event 2: Unknown EOS property d1a3, datasize is 4, using uint32
    0: 0x       1
event 2: currentvalue of d1a3 is 1
event 3: EOS prop d194 desc record, datasize 16, propxtype 3
event 3: propdesc d194, default value not found.
event 4: EOS prop d1a3 info record, datasize is 4
event 4: Unknown EOS property d1a3, datasize is 4, using uint32
    0: 0x       0
event 4: currentvalue of d1a3 is 0
event 5: EOS prop d105 info record, datasize is 4
event 5: currentvalue of d105 is 1
event 6: EOS prop d108 info record, datasize is 4
event 6: currentvalue of d108 is 3
event 7: EOS prop d106 info record, datasize is 4
event 7: currentvalue of d106 is 0
event 8: EOS prop d107 info record, datasize is 4
event 8: currentvalue of d107 is 3
event 9: EOS prop d109 info record, datasize is 4
event 9: currentvalue of d109 is 0
event 10: EOS prop d10a info record, datasize is 4
event 10: currentvalue of d10a is 1450
event 11: EOS prop d10b info record, datasize is 4
event 12: EOS prop d10c info record, datasize is 4
event 13: EOS prop d10d info record, datasize is 4
event 13: currentvalue of d10d is 0
event 14: EOS prop d10e info record, datasize is 4
event 14: currentvalue of d10e is 0
event 15: EOS prop d10f info record, datasize is 4
event 15: currentvalue of d10f is 1
event 16: EOS prop d11b info record, datasize is 4
event 16: currentvalue of d11b is 21d99
event 17: EOS prop d114 info record, datasize is 4
event 17: currentvalue of d114 is 0
event 18: EOS prop d116 info record, datasize is 4
event 18: currentvalue of d116 is 80000302
event 19: EOS prop d119 info record, datasize is 4
event 19: currentvalue of d119 is 100
event 20: EOS prop d110 info record, datasize is 4
event 20: currentvalue of d110 is 87
event 21: EOS prop d101 info record, datasize is 4
[... 2814 lines cut ...]
    6: 00
    7: 00
    8: 04
    9: 00
olcinfoversion is 7
event 5: EOS event OLCInfoChanged (size 20)
    0: 0c
    1: 00
    2: 00
    3: 00
    4: 10
    5: 00
    6: 00
    7: 00
    8: 04
    9: 02
    10: 0f
    11: b8
Sending PTP_OC 0x9116 (EOS_GetEvent) request...
Reading PTP_OC 0x9116 (EOS_GetEvent) data...
Reading PTP_OC 0x9116 (EOS_GetEvent) response...
PTPIP_CMD_RESPONSE
Sending PTP_OC 0x9129 (EOS_RemoteReleaseOff) (0x1) request...
Reading PTP_OC 0x9129 (EOS_RemoteReleaseOff) response...
PTPIP_CMD_RESPONSE

msmeissn, 2021-06-24:
can you try out if you get it working with the commandline?

gphoto2 --trigger-capture --wait-event-and-download=20s

SiuyuenChow, 2021-06-24:
> can you try out if you get it working with the commandline?
> 
> gphoto2 --trigger-capture --wait-event-and-download=20s

The download command is OK.
Using gp_camera_trigger_capture alone, the camera has the shutter sound, but there is still no new photo
After calling gp_camera_trigger_capture and manually pressing the camera shutter, the indicator light does not flicker and does not add to the photo. This situation lasts for 5-8 minutes before pressing the shutter again and the indicator light flashes

SiuyuenChow, 2021-06-24:
This is my code:

self.protocol = @"ptpip";
self.cameraModel = @"Canon EOS (WLAN)";
NSString *connectionStr = [NSString stringWithFormat:@"%@:%@",self.protocol,cameraIP];
int        ret,indexCamera,indexPort;
CameraAbilitiesList    *abilities;
CameraAbilities    a;

GPPortInfoList        *portinfolist = NULL;
GPPortInfo    pi;

gp_log_add_func(GP_LOG_ERROR, errordumper, NULL);
gp_log_add_func(GP_LOG_DEBUG, logdumper, NULL);

context = gp_context_new();

gp_camera_new (&camera);

gp_abilities_list_new (&abilities);

ret = gp_abilities_list_load (abilities, context);

indexCamera = gp_abilities_list_lookup_model (abilities, [self.cameraModel UTF8String]);

if (indexCamera>=0) {
   
    gp_abilities_list_get_abilities (abilities, indexCamera, &a);
    
    gp_camera_set_abilities (camera, a);
}

gp_port_info_list_new (&portinfolist);

ret = gp_port_info_list_load (portinfolist);

ret = gp_port_info_list_count (portinfolist);

indexPort = gp_port_info_list_lookup_path (portinfolist, [connectionStr UTF8String]);
if (indexPort>=0) {
 
    gp_port_info_list_get_info (portinfolist, indexPort, &pi);
    gp_camera_set_port_info (camera, pi);
}

gp_port_info_list_free(portinfolist);
gp_abilities_list_free(abilities);

gp_setting_set("ptpip", "hostname", "test");

ret = gp_camera_init (camera, context);
if (ret == GP_OK) {
    gp_camera_trigger_capture(camera, context);
}

SiuyuenChow, 2021-06-24:
here is the log:

camera_trigger_capture
Triggering capture to , autofocus=0
camera_trigger_canon_eos_capture
Card value is 2
optimized ... setdevicepropvalue of capturetarget to 0x4 not done as it was set already.
Sending PTP_OC 0x9116 (EOS_GetEvent) request...
Reading PTP_OC 0x9116 (EOS_GetEvent) data...
Reading PTP_OC 0x9116 (EOS_GetEvent) response...
PTPIP_CMD_RESPONSE
event 0: unknown EOS event c1f4
    8: 01
    9: 00
    10: 00
    11: 00
event 1: unknown EOS event c1e8
    8: 02
    9: 00
    10: 00
    11: 00
    12: 00
    13: 00
    14: 00
    15: 00
    16: 00
    17: 00
    18: 00
    19: 00
event 2: EOS prop d103 desc record, datasize 40, propxtype 3
event 2: propxtype is 3, prop is 0xd103, data type is 0x0004, propxcnt is 10.
event 2: suppval[0] of d103 is 0x0.
event 2: suppval[1] of d103 is 0x48.
event 2: suppval[2] of d103 is 0x50.
event 2: suppval[3] of d103 is 0x58.
event 2: suppval[4] of d103 is 0x60.
event 2: suppval[5] of d103 is 0x68.
event 2: suppval[6] of d103 is 0x70.
event 2: suppval[7] of d103 is 0x78.
event 2: suppval[8] of d103 is 0x80.
event 2: suppval[9] of d103 is 0x88.
event 3: EOS prop d1d9 desc record, datasize 28, propxtype 3
event 3: propxtype is 3, prop is 0xd1d9, data type is 0x0004, propxcnt is 7.
event 3: suppval[0] of d1d9 is 0x0.
event 3: suppval[1] of d1d9 is 0x3.
event 3: suppval[2] of d1d9 is 0x5.
event 3: suppval[3] of d1d9 is 0x8.
event 3: suppval[4] of d1d9 is 0xb.
event 3: suppval[5] of d1d9 is 0xd.
event 3: suppval[6] of d1d9 is 0x10.
event 4: EOS prop d1dc desc record, datasize 8, propxtype 3
event 4: propxtype is 3, prop is 0xd1dc, data type is 0x0006, propxcnt is 2.
event 4: suppval[0] of d1dc is 0x0.
event 4: suppval[1] of d1dc is 0x1.
event 5: EOS prop d1c3 info record, datasize is 4
event 5: Unknown EOS property d1c3, datasize is 4
    0: 00
    1: 00
    2: 00
    3: 00
olcinfoversion is 18
event 6: EOS event OLCInfoChanged (size 31)
    0: 1b
    1: 00
    2: 00
    3: 00
    4: 41
    5: 02
    6: 00
    7: 00
[... 469 lines cut ...]
focusing - read event type 3
focusing - read event type 3
focusing - read event type 3
focusing - read event type 5
focusinfo content: 000000000200
Canon EOS Capture failed to release: Perhaps no focus?
Canon EOS Capture failed to release: Perhaps no focus?

*** Contexterror ***              
Canon EOS Capture failed to release: Perhaps no focus?
focusing - read event type 0
focusing - read event type 0
focusing - read event type 0
focusing - read event type 3
focusing - read event type 0
focusing - read event type 0
focusing - read event type 3
focusing - read event type 5
focusinfo content: 000000000000
focusing - read event type 0
Sending PTP_OC 0x9129 (EOS_RemoteReleaseOff) (0x1) request...
Reading PTP_OC 0x9129 (EOS_RemoteReleaseOff) response...
PTPIP_CMD_RESPONSE
'camera->functions->trigger_capture (camera, context)' failed: -1
'camera->functions->trigger_capture (camera, context)' failed: -1

SiuyuenChow, 2021-06-24:
info update.

while (1) {
gp_camera_wait_for_event(camera, 1000, &eType, &dData, context);
}

I found that when I polled the gp_camera_wait_for_event method, only this GP_EVENT_FILE_ADDED was not accepted.
The EOS 6D camera screen shows BUSY every time.

SiuyuenChow, 2021-06-24:
info update again
-------------------
The connection was successful,

sleep(10);

call

while (1) {
int ret =  gp_camera_wait_for_event(newCamera, 10, &evttype, &evtdata, ctx);
}

In sleep (10); So that's 10 seconds. I pressed the shutter on my camera, so let's say I pressed the shutter 4 times in 10 seconds to take a picture

GP_CAMERA_WAIT_FOR_EVENT returns GP_EVENT_FILE_Added four times

But when you press the shutter after sleep(10) to take a picture, you never get back to GP_EVENT_FILE_ADDED

I can only get GP_EVENT_FILE_ADDED if I turn off the WiFi inside the camera and turn it back on again

It's not quite clear why that is.

SiuyuenChow, 2021-06-27:
hi @msmeissn again,

I noticed a pattern,
When I call gp_camera_wait_for_event once, the camera is fine
When I call gp_camera_wait_for_event the second time or later, the photo taken by the camera will never be written. It will always be like that
Student: I think what is blocking the camera from taking a picture and writing a picture
Is there something that needs to be released somewhere? gp_camera_wait_for_event

thanks
Siu

msmeissn, 2021-06-28:
Please do not spam other tickets!

for EOS cameras, you need to run wait:_for_event a lot of times ... EOS cameras generate around a 100+ initial events to drain.

Unless all are drained, the camera still might be blocked.

msmeissn, 2021-06-28:
it would be good to see if you can reproduce this wioth the cmmandline tool 

gphoto2 --wait-event-and-download=100s 

to have a 100s second wiatloop, includuing file download.

Sija, 2021-06-28:
@SiuyuenChow Please, use triple backticks for the output - and possibly wrap it into [collapsible section](https://gist.github.com/pierrejoubert73/902cc94d79424356a8d20be2b382e1ab).

SiuyuenChow, 2021-06-28:
I've solved the problem,

thanks for help  @msmeissn

closed 2021-06-28 by SiuyuenChow

papa-alex, 2024-01-30:
> I've solved the problem,
> 
> thanks for help @msmeissn

How was it resolved? I encountered the same problem

papa-alex, 2024-01-30:
I have encountered the same problem. How did I solve it？

papa-alex, 2024-02-01:
I've solved the problem

Sija, 2024-02-01:
@papa-alex Would you share your solution?

Adityadikonda10, 2025-06-17:
@SiuyuenChow, @papa-alex could you help me? what's the solution problem? I'm unable to make the camera write image on 

its sd card when the gp_camera_wait_for_event records an event. The camera returns image file as event_data. I'm polling 

every 500ms.

Adityadikonda10, 2025-06-17:
issue resolved!