gphoto/libgphoto2 #671: Downloading capt0000.jpg

SiuyuenChow, 2021-06-26:
Camera model: EOS 200DII
When I call gp_camera_wait_for_event, I take the picture and return the following

  Downloading capt0000.jpg / 0...
  Downloading capt0001.jpg / 0...
  Downloading capt0002.jpg / 0...

msmeissn, 2021-06-28:
capt0000.jpg is the SDRAM stored image, you need to download it immediately.

gp_setting_set("gphoto2", "--set-config", "capturetarget=1");  ... no, wrong syntax.

use something like
gp_setting_set("ptp2","capturetarget","card");

or edit ~/.gphoto/settings  directly and adjust the capturetarget line or add

ptp2=capturetarget=card

SiuyuenChow, 2021-06-28:
> capt0000.jpg is the SDRAM stored image, you need to download it immediately.
>
> gp_setting_set("gphoto2", "--set-config", "capturetarget=1"); ... no, wrong syntax.
>
> use something like [...]

To understand the
Cheers 🍻
Thank you very much
Best wishes,
Siu

closed 2021-06-28 by SiuyuenChow