gphoto/libgphoto2 #684: Duplicate "Synchronize camera date and time with PC"

RReverser, 2021-07-19:
**Describe the bug**

When using PTP, Canon cameras (or, at least, EOS 250D) return two properties both labelled as "Synchronize camera date and time with PC" in the config:

```
/main/actions/syncdatetimeutc
Label: Synchronize camera date and time with PC
Readonly: 0
Type: TOGGLE
Current: 0
END
/main/actions/syncdatetime
Label: Synchronize camera date and time with PC
Readonly: 0
Type: TOGGLE
Current: 0
END
```

**Name the camera**

```
Model                          Port            
----------------------------------------------------------
USB PTP Class Camera           usb:001,006     
```

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version

```
gphoto2         2.5.20         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking
```

---

Both definitions are here: https://github.com/gphoto/libgphoto2/blob/233569b17322a702aeab8abecc63e37d50de44e2/camlibs/ptp2/config.c#L9966-L9967

A simple fix would be to rename the first one to have `... (UTC)` in the end so that they're distinguishable in UI, but ideally we would only show one of those as a time as the 2nd duplicate is somewhat useless if they do the same thing.

@msmeissn what do you think?

msmeissn, 2021-07-20:
i did this change, thanks!

RReverser, 2021-07-20:
> but ideally we would only show one of those as a time as the 2nd duplicate is somewhat useless if they do the same thing

Ah, so you think we shouldn't hide the other one?

msmeissn, 2021-07-20:
hmm. might also be better. also timeone handling with the eos was weird too, not sure if both work?

RReverser, 2021-07-20:
> hmm. might also be better. also timeone handling with the eos was weird too, not sure if both work?

Yeah I'm not entirely sure what the difference is either and which one users might actually want...

msmeissn, 2021-07-22:
so both use time(NULL), either one mifght not have the right timezone ... ,)

RReverser, 2021-07-22:
Hmm. I think `time(NULL)` returns a UTC time. As long as we don't convert the result to local timezone, it sounds like it's best to expose just the UTC one?