gphoto/libgphoto2 #685: Some Sony properties are reported as readonly but actually aren't

RReverser, 2021-07-19:
**Describe the bug**

I've noticed that some Sony properties on Sony a6600 are reported as readonly, but if I recompile code to ignore that, they are actually correctly writable.

In particular, I ran into this when trying to change "Exposure Compensation Mode" and "Focus Mode". I believe they are writable because they have custom stepping handlers on a6600, and I checked that those handlers do work correctly. However, the underlying PTP property they're assigned to are read-only, and so the corresponding widgets are incorrectly reported by gphoto2 as read-only too and, as a result, it forbids changes to those widgets.

This is the full dump of `--list-all-config`:

```
/main/actions/autofocus
Label: Auto-Focus
Readonly: 0
Type: TOGGLE
Current: 2
END
/main/actions/manualfocus
Label: Manual-Focus
Readonly: 0
Type: RANGE
Current: 0
Bottom: -7
Top: 7
Step: 1
END
/main/actions/capture
Label: Capture
Readonly: 0
Type: TOGGLE
Current: 2
END
/main/actions/bulb
Label: Bulb Mode
Readonly: 0
Type: TOGGLE
Current: 2
END
/main/actions/movie
Label: Movie Capture
Readonly: 0
Type: TOGGLE
Current: 2
END
/main/actions/opcode
Label: PTP Opcode
Readonly: 0
Type: TEXT
Current: 0x1001,0xparam1,0xparam2
END
/main/settings/capturetarget
Label: Capture Target
Readonly: 0
Type: RADIO
Current: sdram
Choice: 0 sdram
Choice: 1 card+sdram
END
/main/status/serialnumber
Label: Serial Number
Readonly: 1
Type: TEXT
Current: 00000000000000003285171006381391
END
/main/status/manufacturer
Label: Camera Manufacturer
Readonly: 1
Type: TEXT
Current: Sony Corporation
END
/main/status/cameramodel
Label: Camera Model
[... 1456 lines cut ...]
```

**Name the camera**

Sony a6600 (connected via USB PTP)

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version

```
gphoto2         2.5.27.1       gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.27.1       standard camlibs (SKIPPING docupen lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

**To Reproduce**
Steps to reproduce the behavior:

Try to list config using `--list-all-config` or change one of the properties from the description above via `set-config`. You'll get an error when trying to change those properties.

However, when changes are made to the code to ignore the readonly status (in gphoto2 https://github.com/gphoto/gphoto2/blob/1cec66dc7b078c470b28cab21bb9f998010e4d87/gphoto2/actions.c#L1827 and in libgphoto2 https://github.com/gphoto/libgphoto2/blob/233569b17322a702aeab8abecc63e37d50de44e2/camlibs/ptp2/config.c#L11156), then those properties written successfully.

---

I believe we need some way to recognise those properties and exclude from readonly list. One way would be to literally hardcode vendor + property combo if the `_get_config`, but perhaps there's a cleaner way I didn't think of? @msmeissn

msmeissn, 2021-07-20:
currently we set the readonly from the weird sony property blob.
ptp-pack.c:: ptp_unpack_Sony_DPD  ... the switch case

the use of these values is guessed. we can globally disabled it again in the code if it does not match reality.

RReverser, 2021-07-20:
> we can globally disabled it again in the code if it does not match reality.

Oh interesting, I didn't see that code. Let me play around and see if it reports anything different for those properties internally that would allow to distinguish those from the rest of actually readonly props.

In most cases it seems to report status correctly, and it would be a shame to lose readonly indicators for all properties when it's only a few that are wrong.

Worst case scenario, I believe we should just whitelist those few properties for which we have custom "stepping" handlers as writable even if they're reported as readonly.

msmeissn, 2021-08-01:
i meanwhile disabled this logic until we find a better / more correct way.
so all props are read/write now.

RReverser, 2021-08-01:
Hmm, I'm not sure if that's better. Most properties were reported as readonly correctly, and those few are rather niche. In general, it's very useful to have readonly status reflected in UI for properties you can't change and as long as it works for most common properties, I'd rather leave the current state of things.

I think we just need to exclude those by hardcoding vendor/propid combo in readonly check in the end.

Btw, we might want to do the same for opposite - e.g., as far as I can tell, there is no way to change "Exposure Program" via PTP API, as those are controlled by physical wheel on the camera - yet it's reported as writable.

msmeissn, 2021-08-03:
ok,. if we have a list then we can add these as overrides...

do you have a list?

and do we report exposure program as ptp prop?

RReverser, 2021-08-03:
> ok,. if we have a list then we can add these as overrides...

For now it's just the props from the issue description, I didn't notice more.

Exposure Program comes from PTP as far as I can tell, yeah. From the log above:

```
/main/other/500e
Label: Exposure Program Mode
Readonly: 0
Type: MENU
Current: 1
Choice: 0 32768
...
```

I believe this is the source that is (incorrectly) reported as writable.

flavioperucchi, 2021-08-31:
Hi, we are currently running into the same problem. It would help a lot if the following setting on a Sony RX100M7 is editable over libgphoto:

```
/main/settings/capturetarget
Label: Capture Target
Readonly: 0
Type: RADIO
Current: card+sdram
Choice: 0 sdram
Choice: 1 card+sdram
Choice: 2 card
END
```

Thanks a lot.

ChristophEngelHH, 2022-06-10:
Any news on this? I ran into that error today. It's working with another version and same camera model. I'll figure out later which versions I'm running.