gphoto/libgphoto2 #687: how to get low lag using canon eos r as webcam??

pootle, 2021-07-21:
Running a canon EOS R as a webcam using
`sudo modprobe v4l2loopback exclusive_caps=1 max_buffers=2`

and 

`gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video0`

results in much more lag than using the canon webcam utility on windows.

Its a fairly beastie laptop (Ryzen 7 5800h) so it's not cpu limited.

Is there anything I can do to get nearer the performance of the canon webcam utility?

msmeissn, 2021-07-21:
what amount of frames are reported by 

gphoto2 --capture-movie=10s

(Just to get a feeling of the FPS we can deliver)

pootle, 2021-07-21:
Capturing preview frames as movie to 'movie.mjpg' for 10 seconds. 
    Movie capture finished (416 frames)

 that seems quite a lot!

msmeissn, 2021-07-21:
ok, this gets 40fps... so the lag is from the rest of the pipeline :/

milovangudelj, 2022-03-28:
Hi @pootle.
Maybe a little late, but if you still haven't managed to find a better way of doing it here's a [gist](https://gist.github.com/Sopamo/e3c0f714432659171074dc3e78e254ec) where instead of FFmpeg they use GStreamer. I haven't been able to use it, but that may be related to missing configuration on my system (I use Fedora).