gphoto/libgphoto2 #688: Cameras inconsistently get stuck on "Capturing frame #..." when tethered and triggerred using sigkill

bkapadia01, 2021-07-23:
**Describe the bug**

We have 2 cameras that are tethered and we're using KILL SIGUSR PID# to trigger the of capture image and download the image.
We have recently noticed that either one or both cameras get stuck on "Capturing frame #.." for period of time after which the camera begins accepting trigger signals again but eventually get stuck again. 

We had this working before without any issues but now we are consistently seeing the issue and easily reproducible.

**Name the camera**
x2  Canon EOS 5DS R 

**libgphoto2 and gphoto2 version**
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.27        
libgphoto2      2.5.27         
libgphoto2_port 0.12.0
Using MacOS 11.5

**To Reproduce**
Steps to reproduce the behavior:
Setup tethering to both cameras and retrieve the PIDs of each of the cameras.
Every few seconds send a sigkill signal to both of the cameras. 
 
see below for after `Capturing frame #47...`
```Capturing frame #45...
New file is in location /capt0041.cr2 on the camera
Saving file as 042-BOT_2021-07-22_21-59-59.cr2
Deleting file /capt0041.cr2 on the camera
Awakened by SIGUSR1...
Capturing frame #46...
New file is in location /capt0042.cr2 on the camera
Saving file as 043-BOT_2021-07-22_22-00-14.cr2
Deleting file /capt0042.cr2 on the camera
Awakened by SIGUSR1...
Capturing frame #47...
ERROR: Could not capture image.
ERROR: Could not capture.
Awakened by SIGUSR1...
Capturing frame #48...
New file is in location /capt0043.cr2 on the camera
Saving file as 044-BOT_2021-07-22_22-01-59.cr2```

vepicurean, 2021-07-23:
Some additional info. It's easy to reproduce this strange behaviour. If people are trying to reproduce this behaviour, please make sure to use a non-Canon prime lens (35mm or 50mm) first.  The strange behaviour with the Capturing Frame # will show up within the first 10 captures 80% of the time.  Then use a Canon lens and see if it happens the same. 

I don't have any Canon lenses as the Sigma ART prime lens picture quality is far better than Canon's expensive prime lenses. Picture quality is the most important thing for me.

msmeissn, 2021-07-23:
i currently just have canon kit lenses for my EOS. perhaps someone else has?

would be nice to have a debuglogfile, not sure if it will help though

vepicurean, 2021-07-23:
> i currently just have canon kit lenses for my EOS. perhaps someone else has?
> 
> would be nice to have a debuglogfile, not sure if it will help though

If you can try to duplicate with the Canon lens, it would be appreciated. I'll get you my debuglogfile from my experiments with the Sigma lens. Will send later today. Thanks.

vepicurean, 2021-07-24:
More info. I borrowed some old Canon zoom lenses and ran experiments today. Using  the best long (15ft) USB3 cables (with boosters) that money can buy, I was able to get the same strange "Capturing frame #" stalling behavior on two Canon 5DS R camera bodies. Same as with the Sigma ART lenses.  Make sure you focus the lenses and run the camera and the lenses both on manual mode once you have the lenses in focus. 

Same behaviour.......  Within the first ten captures, it's intermittent, but both camera bodies stalled/paused on the "Capturing Frame #". Then a minute (random amount of time) later it releases, and then you may be good for another 10 or so captures, but eventually the "Capturing Frame #" stall comes back. 

[image]

[image]

This never happened before the last two days, with this exact same equipment (though before I had seen the stalls with Sigma 50mm prime lenses, and that's why I was forced to only use Sigma 35mm prime lens which were stable up til two days ago).

Something seems to impact the camera body on the SIGKILL as I notice sometimes the mirror will stay up. I hear the mirror come back down several seconds after you expect it to. Maybe it's the mirror staying up too long that causes this intermittent behaviour. 

Tomorrow we'll run these same experiments with both the SIGMA and Canon lenses with the ghoto2 debug logfile on and send the logs. But I wanted to share this Canon experiment results. Very frustrating.

vepicurean, 2021-07-26:
Hi:  

Here's a debug logfiile that we put together. Let us know if we should put anything else together to see if we can figure out why this is happening on both of the Canon 5DS Rs with the capture and download tether and SIGKILL method. I've seen this setup work flawlessly before for hours before, so it's got to be a specific variable that is applicable to both of these 5DS R cameras causing the "capturing frame" instability and freezing. 

[https://drive.google.com/file/d/1NqF6jHxtFZJMNii3biEGt_F6hEusBi9G/view?usp=sharing](url). (Debug session 1)

[https://drive.google.com/file/d/1aAiSlwYdkgWRt3USzBQ-vlDx8isK8Qef/view?usp=sharing](url) (Debug session 2)

Thanks for the help. Appreciated.

vepicurean, 2021-08-04:
Still no change on this. gphoto and the cameras continue to behave the some way with the tether and we get the "Capturing Frame" failure intermittently every few captures. 

Thanks for any suggestions on this.

vepicurean, 2021-08-17:
I figured out that using the gphoto2 --capture-image-and-download -interval 1 causes the freezing behavior on the cameras. I can recreate this anytime on our Canon EOS RDS R cameras.  

It seems to have something to do with the deleting of the files and needing to be awakened by the -SIGUSR1 command. 

--capture-tethered works perfectly. But the issue is I have to walk over to the cameras to trigger them with this --capture-tethered way of doing things.... As stable as it is, it's of no practical use to us (maybe I'm missing how to trigger the capture via a command when using --capture-tethered). 

What is an alternative way to set up a stable tether similar to --capture-tethered and be able to trigger the cameras from a command using command line like the SIGUSER process? Capture-image-and-download is not available to us as it freezes the Canon cameras. Seems others have had issues we the --capture-image-and-download as well. 

Thanks for your help and time.

vepicurean, 2021-08-17:
Would it make sense to go back to the previous version of gphoto2 ( libgphoto2 2.5.26 and gphoto2 2.5.26) to solve/fix this issue with the --capture-image-and-download ??

It was working before using capture-image-and-download, so I'm hoping going back to the previous version will solve the issue?

msmeissn, 2021-08-18:
One thing to try:

Can you go to library.c and add for your EOS R line
        {"Canon:EOS R",                         0x04a9, 0x32da, PTP_CAP|PTP_CAP_PREVIEW|PTP_DONT_CLOSE_SESSION},

The additional flag PTPBUG_DELETE_SENDS_EVENT, like this:

        {"Canon:EOS R",                         0x04a9, 0x32da, PTP_CAP|PTP_CAP_PREVIEW|PTP_DONT_CLOSE_SESSION|PTPBUG_DELETE_SENDS_EVENT},

(if you have R5 or R6, adjust that line)

vepicurean, 2021-08-18:
Thanks. We will edit the it on the test environment and see how that helps.

We have Canon 5DS R camera bodies. Not the mirrorless cameras (R5, R6). How do we write the syntax that line of code in library.c for the 5DS R camera body?

Thanks for your help and time.

msmeissn, 2021-08-19:
just append this flag to the respective line of your camera
|PTPBUG_DELETE_SENDS_EVENT

vepicurean, 2021-08-19:
> just append this flag to the respective line of your camera
> |PTPBUG_DELETE_SENDS_EVENT

I see. Thanks. One more clarification. Is this additional code targeted at fixing the freezing we are seeing when using --capture-image-and-download parameter?  

Thanks for your help and time.

vepicurean, 2021-08-21:
Before I mess with they gphoto code... I've noticed that Terminal and MacOS 11.XX may not be stable. Especially for gphoto related commands and bash scripts that utilize gphoto commands, which we use bash scripts extensively for our gphoto related apps (these apps using the bash scripts used to work perfectly, now they are almost unusable). 

Has anyone else noticed that Terminal behaves in an unstable manner on MacOS 11.XX?

I have also seen some people complaining about about Terminal not working on MacOS 11.XX. Maybe Apple did not fix Terminal on MacOS 11 (Big Sur). I would not be surprised if this is the case. Apple has some issues these days.

vepicurean, 2021-08-21:
Something is not right (maybe it's something we've done wrong) with either Terminal on the MacOS platform, or it's gphoto2 and Terminal on the MacOS platform. 

To see if the issue was isolated to the iMac and MacOS 11 or not, I got out the Mac laptop that has MacOS 10.15.7 running on it. I know this Mac laptop runs perfectly. 

Installed gphoto2 using Homebrew which is at gphoto2 version 2.5.27 

Manually set up a `--capture-image-and download --interval -1 ` session. Right away I knew it would fail as you can see just from the first capture that `--capture-image-and-download --interval -1 ` acts very differently on the Linux computer vs the MacOS computers. 

I'm at a loss. Gphoto was working just fine on the Macs up until about a month ago when everything associated with gphoto2 based on --capture-image-and-download on the Macs became unstable. 

You can see here just a couple shots in, and the whole "capture image and download" session fails. 

[image]

vepicurean, 2021-08-23:
I got out another of the Macs with MacOS 11.5 on it. 

Installed everything from scratch. 

Made the edits to the the code for the 5DS R camera, and executed the build from source procedure for libgphoto2 which seems to install as it should, but it's hard to check if it installed properly. 

Got to the gphoto2 build from source, but that generates the following errors. 

`Making check in gphoto-m4
make[1]: Nothing to be done for `check'.
Making check in contrib
make[1]: Nothing to be done for `check'.
Making check in doc
make[1]: Nothing to be done for `check'.
Making check in gphoto2
gcc -DHAVE_CONFIG_H -I. -I..  -DLOCALEDIR=\"/Users/t2tresearch/.local/share/locale\" -D_GPHOTO2_INTERNAL_CODE  -I.. -I../gphoto2 -I/usr/local/include/gphoto2 -I/usr/local/Cellar/libexif/0.6.22/include -I/usr/local/include/gphoto2 -DLOCALEDIR=\"/Users/t2tresearch/.local/share/locale\" -D_GPHOTO2_INTERNAL_CODE    -I/usr/local/Cellar/libexif/0.6.22/include  -I/usr/local/Cellar/popt/1.18/include -g -O2 -Wall -Wmissing-declarations -Wmissing-prototypes -g -MT gphoto2-shell.o -MD -MP -MF .deps/gphoto2-shell.Tpo -c -o gphoto2-shell.o `test -f 'shell.c' || echo './'`shell.c
shell.c:422:21: error: implicit declaration of function 'rl_copy_text' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
                current = strdup (rl_copy_text (0, end));
                                  ^
shell.c:422:21: note: did you mean 'rl_kill_text'?
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/readline/history.h:210:7: note: 'rl_kill_text' declared here
int              rl_kill_text(int, int);
                 ^
shell.c:422:21: warning: incompatible integer to pointer conversion passing 'int' to parameter of type 'const char *' [-Wint-conversion]
                current = strdup (rl_copy_text (0, end));
                                  ^~~~~~~~~~~~~~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/string.h:117:26: note: passing argument to parameter '__s1' here
char    *strdup(const char *__s1);
                            ^
1 warning and 1 error generated.
make[1]: *** [gphoto2-shell.o] Error 1
make: *** [check-recursive] Error 1

msmeissn, 2021-08-23:
this is readline-devel not the right version... or something is too new? hmm.

msmeissn, 2021-08-23:
i do not have such an EOS camera, i have some "lower end" EOS ones... 

It is something I would like to try out, as we saw it earlier that Canon cameras got stuck due to interrupt delivery issues and this flag helped.

as for stability on MacOS, so far it seems on-par with Linux... Have not heard complaints lately.

vepicurean, 2021-08-23:
> this is readline-devel not the right version... or something is too new? hmm.

I'm not sure what mean on the "readline-devel" not the right version.  

I followed these instructions here for building gphoto2 

[image]

There is no popt-devel, etc for the brew package manager on MacOS (just install for popt library on brew package manager). But from reading the messages during the gphoto2 build process, popt seems OK. 

How do you tell if libgphoto2 is install OK?

vepicurean, 2021-08-23:
> i do not have such an EOS camera, i have some "lower end" EOS ones...
>
> It is something I would like to try out, as we saw it earlier that Canon cameras got stuck due to interrupt delivery issues and [...]

I have used capture-image-and-download on the newer EOS 5DS R and also the much older (mid 2000s) Canon EOS 5D Mark II (Mark 2). A month ago capture-image-and-download was working the same on both the newer EOS 5DS Rs and older 5D Mark II cameras. No difference.  All were very stable. Triggering via capture-image-and-download using the -SIGUSR1 to trigger the cameras via USB3 was working just fine. I was actually very surprised how stable it was.  

I don't know whether or not the MacOS brew package manager last month updated the version to the current 2.5.27 (from a previous version of gphoto2) which is/was the cause of the issues we're now seeing with capture-image-and-download on all the Macs.  It has nothing to do with MacOS 11 vs MacOS 10.15.7. gphoto2 capture-image-and-download behaves erratically on multiple Macs have 2.5.27 installed on and managed via the brew package manager.   

Good to hear that you are seeing the same stability on MacOS as with Linux. I would expect that to be the case. 

I'm thinking somehow, something changed on gphoto2 that is impacting specifically capture-image-and-download on the Macs. But I'm at a loss what changed. 

I'm trying to get gphoto2 built and compiled on a completely fresh Mac running 11.5 that has never had gphoto on it before, and compile it from source using your PTPBUG_DELETE_SENDS_EVENT code edit. 

But the make process has a couple errors.

vepicurean, 2021-08-29:
> > > this is readline-devel not the right version... or something is too new? hmm.
> > 
> > 
> > I'm not sure what you mean on the "readline-devel" not the right version?? Sorry for not understanding. But there is obviously something wrong related to the readline program which was installed by the brew package manager on this new Mac. As you know, readline is needed to as a dependency for gphoto2.
> 
> If you are not using the interactive gphoto2 shell (`gphoto2 --shell`), you should be able to `configure --without-readline` and build without any readline related compilation.
> 
> Unfortunately, I do not have (access to) a Mac to run builds on MacOS.
> 
> > There is no popt-devel, etc for the brew package manager on MacOS (just install for popt library on brew package manager). But from reading the messages during the gphoto2 build process, popt seems OK.
> 
> The specific package names differ from system to system, whether that is `popt-dev` or `popt-devel` or `popt` - you need the package containing the popt development files like `popt.h`, that is all.
> 
> > ```
> > T2T-M3s-Mac-Pro:gphoto2 t2tresearch$ make check
> > Making check in gphoto-m4
> > make[1]: Nothing to be done for `check'.
> > Making check in contrib
> > make[1]: Nothing to be done for `check'.
> > Making check in doc
> > make[1]: Nothing to be done for `check'.
> > Making check in gphoto2
> > gcc -DHAVE_CONFIG_H -I. -I..  -DLOCALEDIR=\"/Users/t2tresearch/.local/share/locale\" -D_GPHOTO2_INTERNAL_CODE  -I.. -I../gphoto2 -I/usr/local/include/gphoto2 -I/usr/local/Cellar/libexif/0.6.22/include -I/usr/local/include/gphoto2 -DLOCALEDIR=\"/Users/t2tresearch/.local/share/locale\" -D_GPHOTO2_INTERNAL_CODE    -I/usr/local/Cellar/libexif/0.6.22/include  -I/usr/local/Cellar/popt/1.18/include -g -O2 -Wall -Wmissing-declarations -Wmissing-prototypes -g -MT gphoto2-shell.o -MD -MP -MF .deps/gphoto2-shell.Tpo -c -o gphoto2-shell.o `test -f 'shell.c' || echo './'`shell.c
> > shell.c:422:21: **error: implicit declaration of function 'rl_copy_text' is invalid in C99 [-Werror,-Wimplicit-function-declaration]**
> >                 current = strdup (rl_copy_text (0, end));
> >                                   ^
> > shell.c:422:21: note: did you mean 'rl_kill_text'?
> > /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/readline/history.h:210:7: note: 'rl_kill_text' declared here
> > int              rl_kill_text(int, int);
> >                  ^
> > shell.c:422:21: warning: incompatible integer to pointer conversion passing 'int' to parameter of type 'const char *' [-Wint-conversion]
> >                 current = strdup (rl_copy_text (0, end));
> >                                   ^~~~~~~~~~~~~~~~~~~~~
> > /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/string.h:117:26: note: passing argument to parameter '__s1' here
> > char    *strdup(const char *__s1);
> >                             ^
> > 1 warning and 1 error generated.
> > make[1]: *** [gphoto2-shell.o] Error 1
> > make: *** [check-recursive] Error 1
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > If we can fix this error about the "implicit declaration of function "rl_copy_text" is invalid in C99" then I believe we can get gphoto2 to build properly, and get this new code in libgphoto2 linked to gphoto2. Then it's a quick test to see if this new code in libgphoto2 fixes the issue with -`-capture-image-and-download` parameter
> 
> Try compiling after `configure --without-readline`.

Some good news. I ran the regular `./configure PKG_CONFIG_PATH="$HOME/.local/lib/pkgconfig${PKG_CONFIG_PATH+":${PKG_CONFIG_PATH}"}" --prefix="$HOME/.local"`

then I ran the suggested `./configure --without-readline`

We are just using command line. Never even knew there was a gphoto2 --shell. 

Here's the result of the "make check". A lot of warnings and it's a bit confusing as some of the warnings get reported as errors. 

```
T2T-M3s-Mac-Pro:gphoto2 t2tresearch$ make check
Making check in gphoto-m4
make[1]: Nothing to be done for `check'.
Making check in contrib
make[1]: Nothing to be done for `check'.
Making check in doc
[... 100 lines cut ...]
make[1]: *** [check-recursive] Error 1
make: *** [check-recursive] Error 1
```

I did this on the test Mac. I have two versions of gphoto on it now. One 2.5.26 managed by Homebrew and then this version 2.5.27.1 that I built from the source code on GitHub. The key as mentioned is to run  `./configure --without readline` (if you're not using the gphoto shell). 

```
T2T-M3s-Mac-Pro:gphoto2 t2tresearch$ gphoto2 --version
gphoto2 2.5.27.1

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.27.1       gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.27.1       standard camlibs (SKIPPING docupen), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1, gcc, ltdl, EXIF, USB, serial without locking
```

Now, I don't see any difference in performance between this version using the libphoto2 2.5.27.1 where we edited the 5DS R  code with the  `PTPBUG_DELETE_SENDS_EVENT` vs the version installed by Homebrew. 

I did some more testing with the camera and lens on manual vs auto mode this weekend. I'll have some more notes to share with everyone on that in a couple days once I'm sure of the behaviours.

vepicurean, 2021-08-31:
Hi:

The is a key setting to using `--capture-image-and-download --interval -1`, then open another terminal session and control the shutter trigger signal by issuing command `kill -SIGUSR1 PID#`.  You can have the camera body in "M" or manual mode. But you must ensure the physical lens switch setting for the Canon 5DS R is on automatic focus (AF). If not, --capture-image-and-download will be inconsistent/unstable..... and eventually you get the "capturing frame" freeze issue. The will camera stop triggering and eventually gphoto is useless from the instability of the camera. 

I can get only `--capture-image-and-download --interval -1` to work (with the lens switch set to AF), but the capture and download sequence is not quick (the time to capture and download an image varies noticeably), and the process unnecessarily focuses the lens with each capture. For our application, we focus the lens at the start of the capture session and that's all we need until the next time we need to focus the lens. 

If anyone knows of a way to prevent the lens from focusing on each signal (SIGUSR1) trigger using --capture-image-and-download --interval -1 it would be appreciated. 

Thanks for your help and time.

vepicurean, 2021-09-04:
Do you think this is the alternative way to achieve remote `--capture-image-and download --interval -1` via the computer without autofocus each time? In notice the PID for this process stops after full execution of this command. We'll also need to manually increment the sequence ID of the downloaded file, or you get the "overwrite existing" question each time. Also no kill -SIGUSR1 PID needed with the below. Which makes things even simpler. 

I can't tell if the autofocus step is executing every time or not when using the below, but this triggers the full press of the shutter remotely and downloads the RAW image to the computer immediately. Takes about 3 to 4 seconds for the entire process through to "CAPTURECOMPLETE". As long as it's under 5 seconds, it works for us. 

`gphoto2 --set-config eosremoterelease="Immediate" --set-config eosremoterelease="Release Full" --wait-event-and-download="CAPTURECOMPLETE"`

This is not freezing the camera or the lens (lens switch is set to AF), which is great.

Can someone that understands what is happening here, please give some expert input where the autofocus is still happening or not.

vepicurean, 2021-09-08:
Update:

When we run our script at a 7 second shutter trigger interval using the `gphoto2 --set-config eosremoterelease="Immediate" --set-config eosremoterelease="Release Full" --wait-event-and-download="FILEADDED"` command we able to achieve stable operation of the cameras and the images are good quality. But when we increase the interval of our remote shutter trigger script to 5 seconds, this seems to cause issues (conflicts between the sequential remote triggers) as the wait time for the "FILEADDED" event is too long.

Any suggestions to be able to decrease the entire remote shutter trigger and download processing time down to 2 or 3 seconds? That's what we need to get the remote shutter trigger capture and download processing time down to to operate our 5 second interval script. Thanks for your help and time.

farhi, 2024-03-23:
Hello, I have exactly the same error with a Sony Alpha 6000. gphoto 2.5.27, camera plugged via a USB cable. The camera is in "S" program (speed).

The first few shots work, then the camera gets stuck:
```
gphoto2 --shell --force-overwrite
gphoto2: {/home/farhi} /> capture-image                                        
New file is in location /capt0000.jpg on the camera                            
gphoto2: {/home/farhi} /> capture-image
New file is in location /capt0001.jpg on the camera                            
New file is in location /capt0002.jpg on the camera                            
gphoto2: {/home/farhi} /> capture-image
New file is in location /capt0003.jpg on the camera                            
gphoto2: {/home/farhi} /> capture-image
New file is in location /capt0004.jpg on the camera                            
gphoto2: {/home/farhi} /> capture-image
ERROR: Could not capture image.
ERROR: Could not capture.
gphoto2: {/home/farhi} /> capture-image
ERROR: Could not capture image.
ERROR: Could not capture. 
```
If I turn the camera off/on, the issue is still there, which shows this is purely gphoto related.

**A tip on the issue:**
This test was done with the lens cap on (black image), and I found out the issue somewhat resides in the focusing. Indeed, the camera can not focus, which blocks the capture command. Then all subsequent capture commands from gphoto2 fail, even when further removing the cap from the lens.
If I exit/restart gphoto2, cap lens off, then all is fine.

My conclusion is that the issue is indeed related to gphoto2 only (the camera works fine by itself), and when a focusing error/timeout occurs, gphoto2 can not grab anymore pictures.