gphoto/libgphoto2 #690: Track down leaked `ptp_getobjectinfo` results

RReverser, 2021-07-23:
Originally reported in https://github.com/gphoto/libgphoto2/pull/689#issuecomment-885602845, copying here for a separate tracking issue: most usages of `ptp_getobjectinfo` seem to leak the resulting heap-allocated strings and must call `ptp_freeobjectinfo`.

---

> mainly the objectinfos go into the PTPObjects and are freed from there.

Maybe I missed a few, but I haven't seem examples of that.

E.g. from the same file, the Fuji switch below: https://github.com/gphoto/libgphoto2/blob/fcdf03182febb8ced418c8a002ca04682b200e86/camlibs/ptp2/library.c#L3692. As far as I can tell, this `oi` will also be leaked like the Sony one.

Same for this one: https://github.com/gphoto/libgphoto2/blob/fcdf03182febb8ced418c8a002ca04682b200e86/camlibs/ptp2/library.c#L4489

Or here: https://github.com/gphoto/libgphoto2/blob/fcdf03182febb8ced418c8a002ca04682b200e86/camlibs/ptp2/library.c#L4893

...and there are many others.

msmeissn, 2021-07-23:
i did a first pass. I however need to now test as many of those cameras as I can in case I made a mistake ;)

RReverser, 2021-07-23:
Nice, thanks! Yeah, testing was my concern too - unfortunately, I don't have many cameras to verify it on.

notnot, 2021-07-31:
Since a few weeks i installed 2.5.27, and have also discovered a growing memory usage when running live previews. I have a machine vision system that is running non-stop and this memory leak will eventually lead to a fatal Out of Memory error. That system can't run for more than two days now... 
I will download and install libgphoto2 master today and see how it goes. Camera used is the Sony A6200, so i can't contribute to wider testing on other camera types.

RReverser, 2021-07-31:
@notnot That one should be fixed by #689, at least I didn't find any more leaks for Sony a6600 live view afterwards.

RReverser, 2021-07-31:
It's strange that this issue went unnoticed for such a long time though... as far as I can tell, most of the relevant code has been unchanged for years?

notnot, 2021-07-31:
> @notnot That one should be fixed by #689, at least I didn't find any more leaks for Sony a6600 live view afterwards.

Have you also looked for leaks associated with capture and download of images? My application does that a lot as well, if it sees something interesting in the preview. I will monitor memory usage and hopefully all the memory leaks are gone now!

notnot, 2021-07-31:
> It's strange that this issue went unnoticed for such a long time though... as far as I can tell, most of the relevant code has been unchanged for years?

What is the magnitude of the leaked memory you were seeing? I am seeing on average 10.5 MB per hour, when capturing 25 previews a second. I measured the memory use of my application every 5 minutes, with the (ubuntu) system monitor, and the memory growth isn't linear. 

For example:
00 minutes : 26.2 MB
05 minutes : 27.9 MB (+ 1.7 MB)
10 minutes : 30.0 MB (+ 2.1 MB)
15 minutes : 30.6 MB (+ 0.6 MB)
20 minutes : 30.9 MB (+ 0.3 MB)
25 minutes : 31.1 MB (+ 0.2 MB)
30 minutes : 33.7 MB (+ 2.6 MB)
35 minutes : 34.3 MB (+ 0.6 MB)
40 minutes : 34.3 MB (0)
45 minutes : 34.9 MB (+ 0.6 MB)
50 minutes : 35.3 MB (+ 0.4 MB)
55 minutes : 36.0 MB (+ 0.7 MB)
60 minutes : 36.7 MB (+ 0.7 MB)

All this time the camera is stationary, pointed at a scene that doesn't change, so the preview data is very similar in size on each new frame. 

There is a complication in my setup: i am programming in Go, and i use a wrapper around the C libgphoto2 library. It is possible that there is a flaw in that wrapper. Maybe it doesn't free everything that needs to be freed. I think my next experiment would be to write a pure C program that captures previews as fast as possible and see how the memory usage is, over longer periods of time.

msmeissn, 2021-08-01:
various live methods have only been added recently in the last 3 years or so FWIW.
the leaks should increate linear with amount of captured frames if the fix is not applied. Please test with fix applied.

RReverser, 2021-08-01:
> I will monitor memory usage and hopefully all the memory leaks are gone now!

@notnot FWIW it's a very slow and unreliable method. Best to compile your code with leak sanitizers or run with Valgrind (I'm not entirely sure which one integrates with Go better). That's how I found the leak above, and ensured that at least my program didn't have any others, no matter how small.

notnot, 2021-08-01:
> @notnot FWIW it's a very slow and unreliable method. Best to compile your code with leak sanitizers or run with Valgrind (I'm not entirely sure which one integrates with Go better). That's how I [...]

Yes this is a crude way of monitoring memory. I will learn about the best ways to do this accurately with Go executables. Go is a garbage collecting language, which is not so simple to reason about :(

In the meantime i did write a minimal C program that captures previews 25 times a second, and i don't see any memory leaks when running it. The huge memory leak that i was seeing in my application must thus be on the Go side. I will investigate the gphoto2 wrapper in detail to find the flaw. Sorry for the noise!

There is still a mystery though: with libgphoto2 2.5.20, which i have been using for some years, my application/machine could run for months without an OOM error. With 2.5.27 it runs for at most 2 days, then it hits an OOM. I didn't change my code, i only upgraded libgphoto2. 
I tend to believe that this is not a flaw in libgphoto2, but a flaw in the Go wrapper and/or the way Go handles memory that is allocated externally.

notnot, 2021-08-03:
I spoke too soon... when running the memory leak test for a longer time, i do see a serious leak related to preview captures. This happens in my C test program, so it has nothing to do with the Go wrapper. Note that the older 2.5.20 that i used to use in production doesn't suffer from it. 2.5.27  has it. I am running this test on arm cpu, maybe that matters. 
I did notice something peculiar that might explain why a longer test run is needed to see the leak. 
In the beginning of preview streaming, the preview data is stored at the same address each time. Usually, after a few hundred previews, the data address changes, it is incrementing. For instance:

frame 346: 26343 bytes of data @ 0x55AFF6CFD0
frame 347: 26262 bytes of data @ 0x55AFF6CFD0
frame 348: 26254 bytes of data @ 0x55AFF6CFD0
frame 349: 26273 bytes of data @ 0x55AFF6CFD0
frame 350: 26350 bytes of data @ 0x55AFF6CFD0
frame 351: 26271 bytes of data @ 0x55AFF6CFD0
frame 352: 26323 bytes of data @ 0x55AFF6CFD0
frame 353: 26309 bytes of data @ 0x55AFF6CFF0
frame 354: 26314 bytes of data @ 0x55AFF6D050
frame 355: 26299 bytes of data @ 0x55AFF6D090
frame 356: 26251 bytes of data @ 0x55AFF6D0D0
frame 357: 26318 bytes of data @ 0x55AFF6D130
frame 358: 26317 bytes of data @ 0x55AFF6D170
frame 359: 26345 bytes of data @ 0x55AFF6D1D0
frame 360: 26344 bytes of data @ 0x55AFF6D210
frame 361: 26289 bytes of data @ 0x55AFF6D270
frame 362: 26325 bytes of data @ 0x55AFF6D2B0

In 2.5.20 the preview  data is always stored at the same address, at least it is on another of my machines where i tested on too. No memory leaks there, even after an hour of previews at 25 FPS.  

It is possible that my C test program is not correct, i include it here, so that you can verify it:
----------------------------------------
// gphoto2.c, jpad 2021
// Test libgphoto2 live preview

#include <stdio.h>
#include <stdlib.h>

#include <gphoto2/gphoto2.h>

int main(int argc, char **argv)
{
    printf("libgphoto2 live preview\n");

    // initialize a context
    GPContext* context = gp_context_new();
    if (context == NULL) {
        printf("gp_context_new() failed\n");
        return 1;
    }
    printf("gphoto2 context initialized\n");

    // initialize a camera
    Camera* camera; 
    gp_camera_new(&camera);
    if (camera == NULL) {
        printf("gp_camera_new() failed\n");
        return 1;
    }
    int r = gp_camera_init(camera, context);
    if (r != GP_OK) {
        printf("gp_camera_init(): %d\n", r);
        return 1;
    }
    printf("initialized camera\n");

    // get previews
    int count = 0;
    while (count < 1000000000) {
        CameraFile *file;
        int r = gp_file_new(&file);
        if (r != GP_OK) {
            printf("gp_file_new(): %d\n", r);
            return 1;
        }

        r = gp_camera_capture_preview(camera, file, context);
        if (r != GP_OK) {
            printf("gp_camera_capture_preview(): %d\n", r);
            return 1;
        }

        // peek at the data
        const char* data;
        ulong len;
        gp_file_get_data_and_size(file, &data, &len);
        printf("frame %d: %d bytes of data @ 0x%lX\n",
            count, (int)len, (ulong)data);

        //gp_file_unref(file);
        gp_file_free(file);

        count++;
    }

    // free camera
    gp_camera_exit(camera, context);
    gp_camera_unref(camera);

    // free context
    gp_context_unref(context);

    return 0;
}
----------------------------------------

 If you need more info on my setup, please tell me how to provide it. My computer vision machine is running on a nVidia Jeston Nano computer.

notnot, 2021-08-03:
Note that i tried both gp_file_unref() and gp_file_free(). No difference. Or should i use both?

RReverser, 2021-08-03:
> 2.5.27 has it.

Just to be clear - 2.5.27 release or the master of the repo (2.5.27.1)? Only the master has the fix for the leak.

notnot, 2021-08-03:
Of course i have applied the changes. 

The leak is serious, for example

time     mem

00.00 : 0.8 MB
05:00 : 1.5 MB
10:00 : 2.6 MB
15:00 : 3.2 MB

RReverser, 2021-08-03:
Can you build with `-g -fsanitize=leak` so that it can give you stacktraces for what's actually leaking?

notnot, 2021-08-03:
i can try
how to get the stack trace?

RReverser, 2021-08-03:
It will print it for you upon exit automatically. If you haven't used it before, the docs are pretty good: https://clang.llvm.org/docs/LeakSanitizer.html

You can also reduce number of iterations - just run like 100 previews or so, if there's a leak, it doesn't matter how many you run, it should show up.

RReverser, 2021-08-03:
> If you haven't used it before, the docs are pretty good: [clang.llvm.org/docs/LeakSanitizer.html](https://clang.llvm.org/docs/LeakSanitizer.html)

Er, actually, that page doesn't show just `-fsanitize=leak` usage, but I guess `-fsanitize=address` is even better, just in case.

notnot, 2021-08-03:
> It will print it for you upon exit automatically. If you haven't used it before, the docs are pretty good: https://clang.llvm.org/docs/LeakSanitizer.html
>
> You can also reduce number of iterations - [...]

The thing is... i think the leak comes later on... usually after several hundred previews.. see my earlier discovery of the way that the data location changes... probably related to that

hfiguiere, 2021-08-03:
> The thing is... i think the leak comes later on...

Maybe you just notice it at that point, but LeakSanitizer will tell you if you leak 4 bytes...

RReverser, 2021-08-03:
> see my earlier discovery of the way that the data location changes

Data location can change due to allocator fragmentation, it's not much of a symptom by itself usually.

notnot, 2021-08-03:
frame 90: 26413 bytes of data @ 0x7F7E700000
frame 91: 26409 bytes of data @ 0x7F7E700000
frame 92: 26466 bytes of data @ 0x7F7E700000
frame 93: 26392 bytes of data @ 0x7F7E700000
frame 94: 26387 bytes of data @ 0x7F7E700000
frame 95: 26383 bytes of data @ 0x7F7E700000
frame 96: 26391 bytes of data @ 0x7F7E700000
frame 97: 26390 bytes of data @ 0x7F7E700000
frame 98: 26461 bytes of data @ 0x7F7E700000
frame 99: 26446 bytes of data @ 0x7F7E700000

=================================================================
==11311==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 4096 byte(s) in 1 object(s) allocated from:
    #0 0x7f844f376f in malloc (/usr/lib/aarch64-linux-gnu/liblsan.so.0+0xe76f)
    #1 0x7f83fdc77f  (/usr/local/lib/libgphoto2.so.6.1.0+0x5a77f)
    #2 0x7f83fdfed3  (/usr/local/lib/libgphoto2.so.6.1.0+0x5ded3)
    #3 0x7f83fe3e87  (/usr/local/lib/libgphoto2.so.6.1.0+0x61e87)
    #4 0x7f83fe42df  (/usr/local/lib/libgphoto2.so.6.1.0+0x622df)
    #5 0x7f83fe611f  (/usr/local/lib/libgphoto2.so.6.1.0+0x6411f)
    #6 0x7f83fe6433  (/usr/local/lib/libgphoto2.so.6.1.0+0x64433)
    #7 0x7f83fe7ae7  (/usr/local/lib/libgphoto2.so.6.1.0+0x65ae7)
    #8 0x7f84117143  (<unknown module>)
    #9 0x7f8411593b  (/lib/aarch64-linux-gnu/libpthread-2.27.so+0x2093b)
    #10 0x7f8410d057  (/lib/aarch64-linux-gnu/libc-2.27.so+0x16057)
    #11 0x7f841556a7  (<unknown module>)
    #12 0x7f84265807 in foreach_func /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2_port/libgphoto2_port/gphoto2-port-info-list.c:234
    #13 0x7f8424dab7  (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x3ab7)
    #14 0x7f8424d4a3  (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x34a3)
    #15 0x7f8424e0b3 in lt_dlforeachfile (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x40b3)
    #16 0x7f84265cbf in gp_port_info_list_load /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2_port/libgphoto2_port/gphoto2-port-info-list.c:286
    #17 0x7f8445d19b in gp_camera_init /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2/gphoto2-camera.c:724
    #18 0x555b98bd7b in main (/home/jpad/prog/C/test/gphoto2/gphoto2+0xd7b)
    #19 0x7f843196df in __libc_start_main (/lib/aarch64-linux-gnu/libc.so.6+0x206df)
    #20 0x555b98bc13  (/home/jpad/prog/C/test/gphoto2/gphoto2+0xc13)

Direct leak of 4096 byte(s) in 1 object(s) allocated from:
    #0 0x7f844f376f in malloc (/usr/lib/aarch64-linux-gnu/liblsan.so.0+0xe76f)
    #1 0x7f83fb677f  (/usr/local/lib/libgphoto2.so.6.1.0+0x3477f)
    #2 0x7f83fb9ed3  (/usr/local/lib/libgphoto2.so.6.1.0+0x37ed3)
    #3 0x7f83fbde87  (/usr/local/lib/libgphoto2.so.6.1.0+0x3be87)
    #4 0x7f83fbe2df  (/usr/local/lib/libgphoto2.so.6.1.0+0x3c2df)
    #5 0x7f83fc011f  (/usr/local/lib/libgphoto2.so.6.1.0+0x3e11f)
    #6 0x7f83fc0433  (/usr/local/lib/libgphoto2.so.6.1.0+0x3e433)
    #7 0x7f83fc1ae7  (/usr/local/lib/libgphoto2.so.6.1.0+0x3fae7)
    #8 0x7f83fea143  (/usr/local/lib/libgphoto2.so.6.1.0+0x68143)
    #9 0x7f83fe893b  (/usr/local/lib/libgphoto2.so.6.1.0+0x6693b)
    #10 0x7f83fe0057  (/usr/local/lib/libgphoto2.so.6.1.0+0x5e057)
    #11 0x7f84153deb  (<unknown module>)
    #12 0x7f84268337 in gp_port_init /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2_port/libgphoto2_port/gphoto2-port.c:118
    #13 0x7f84268337 in gp_port_set_info /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2_port/libgphoto2_port/gphoto2-port.c:199
    #14 0x7f84459f23 in gp_abilities_list_detect /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2/gphoto2-abilities-list.c:422
    #15 0x7f8445d1b7 in gp_camera_init /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2/gphoto2-camera.c:725
    #16 0x555b98bd7b in main (/home/jpad/prog/C/test/gphoto2/gphoto2+0xd7b)
    #17 0x7f843196df in __libc_start_main (/lib/aarch64-linux-gnu/libc.so.6+0x206df)
    #18 0x555b98bc13  (/home/jpad/prog/C/test/gphoto2/gphoto2+0xc13)

Direct leak of 4096 byte(s) in 1 object(s) allocated from:
    #0 0x7f844f376f in malloc (/usr/lib/aarch64-linux-gnu/liblsan.so.0+0xe76f)
    #1 0x7f83fdc77f  (/usr/local/lib/libgphoto2.so.6.1.0+0x5a77f)
    #2 0x7f83fdfb97  (/usr/local/lib/libgphoto2.so.6.1.0+0x5db97)
    #3 0x7f8411711b  (<unknown module>)
    #4 0x7f8411593b  (/lib/aarch64-linux-gnu/libpthread-2.27.so+0x2093b)
    #5 0x7f8410d057  (/lib/aarch64-linux-gnu/libc-2.27.so+0x16057)
    #6 0x7f841556a7  (<unknown module>)
    #7 0x7f84265807 in foreach_func /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2_port/libgphoto2_port/gphoto2-port-info-list.c:234
    #8 0x7f8424dab7  (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x3ab7)
    #9 0x7f8424d4a3  (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x34a3)
    #10 0x7f8424e0b3 in lt_dlforeachfile (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x40b3)
[... 85 lines cut ...]
    #4 0x7f83fe893b  (/usr/local/lib/libgphoto2.so.6.1.0+0x6693b)
    #5 0x7f83fe0057  (/usr/local/lib/libgphoto2.so.6.1.0+0x5e057)
    #6 0x7f84153deb  (<unknown module>)
    #7 0x7f84268337 in gp_port_init /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2_port/libgphoto2_port/gphoto2-port.c:118
    #8 0x7f84268337 in gp_port_set_info /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2_port/libgphoto2_port/gphoto2-port.c:199
    #9 0x7f84459f23 in gp_abilities_list_detect /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2/gphoto2-abilities-list.c:422
    #10 0x7f8445d1b7 in gp_camera_init /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2/gphoto2-camera.c:725
[... cut ...]
    #4 0x7f7fe47d47  (<unknown module>)
    #5 0x7f8446014f in gp_camera_capture_preview /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2/gphoto2-camera.c:1403
    #6 0x555b98bdf7 in main (/home/jpad/prog/C/test/gphoto2/gphoto2+0xdf7)
    #7 0x7f843196df in __libc_start_main (/lib/aarch64-linux-gnu/libc.so.6+0x206df)
    #8 0x555b98bc13  (/home/jpad/prog/C/test/gphoto2/gphoto2+0xc13)

SUMMARY: LeakSanitizer: 35927 byte(s) leaked in 251 allocation(s).
makefile:16: recipe for target 'run' failed
make: *** [run] Error 23

RReverser, 2021-08-03:
Nit: it's best to always wrap code, logs, etc. on Github with

````
```
...
```
````

Otherwise you just accidentally cc'd a bunch of old issues.

notnot, 2021-08-03:
The above was just 100 preview frames
I hope someone can use this report to pinpoint the problem!

RReverser, 2021-08-03:
@notnot You can pinpoint it quite easily yourself. Ignore everything coming from `gp_port_init` - obviously those are not really "leaks" but a normal global init. The last one, however, is more suspicious as it comes from gp_camera_capture_preview:

```
Direct leak of 3159 byte(s) in 243 object(s) allocated from:
#0 0x7f844f376f in malloc (/usr/lib/aarch64-linux-gnu/liblsan.so.0+0xe76f)
#1 0x7f8437413f in strdup (/lib/aarch64-linux-gnu/libc.so.6+0x7b13f)
#2 0x7f7fe2c9c7 ()
#3 0x7f7fe33947 ()
#4 0x7f7fe47d47 ()
#5 0x7f8446014f in gp_camera_capture_preview /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2/gphoto2-camera.c:1403
#6 0x555b98bdf7 in main (/home/jpad/prog/C/test/gphoto2/gphoto2+0xdf7)
#7 0x7f843196df in __libc_start_main (/lib/aarch64-linux-gnu/libc.so.6+0x206df)
#8 0x555b98bc13 (/home/jpad/prog/C/test/gphoto2/gphoto2+0xc13)
```

Unfortunately, looks like your libgphoto2 is built with optimisations and not just a debug build, so it doesn't provide much more info (missing function names).

RReverser, 2021-08-03:
Make sure to rebuild everything - clean & make libgphoto2 as well as rebuild your own wrapper - with just `CFLAGS=-g` to disable optimizations and enable debug info.

RReverser, 2021-08-03:
> Ignore everything coming from `gp_port_init` - obviously those are not really "leaks" but a normal global init.

Btw, if you free camera itself before exit, those spurious leaks should go away too.

notnot, 2021-08-03:
sorry... i'm quite new to this, second attempt with better formatting

```
frame 90: 26413 bytes of data @ 0x7F7E700000
frame 91: 26409 bytes of data @ 0x7F7E700000
frame 92: 26466 bytes of data @ 0x7F7E700000
frame 93: 26392 bytes of data @ 0x7F7E700000
frame 94: 26387 bytes of data @ 0x7F7E700000
frame 95: 26383 bytes of data @ 0x7F7E700000
frame 96: 26391 bytes of data @ 0x7F7E700000
frame 97: 26390 bytes of data @ 0x7F7E700000
frame 98: 26461 bytes of data @ 0x7F7E700000
frame 99: 26446 bytes of data @ 0x7F7E700000

=================================================================
==11311==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 4096 byte(s) in 1 object(s) allocated from:
    #0 0x7f844f376f in malloc (/usr/lib/aarch64-linux-gnu/liblsan.so.0+0xe76f)
    #1 0x7f83fdc77f  (/usr/local/lib/libgphoto2.so.6.1.0+0x5a77f)
    #2 0x7f83fdfed3  (/usr/local/lib/libgphoto2.so.6.1.0+0x5ded3)
    #3 0x7f83fe3e87  (/usr/local/lib/libgphoto2.so.6.1.0+0x61e87)
    #4 0x7f83fe42df  (/usr/local/lib/libgphoto2.so.6.1.0+0x622df)
    #5 0x7f83fe611f  (/usr/local/lib/libgphoto2.so.6.1.0+0x6411f)
    #6 0x7f83fe6433  (/usr/local/lib/libgphoto2.so.6.1.0+0x64433)
    #7 0x7f83fe7ae7  (/usr/local/lib/libgphoto2.so.6.1.0+0x65ae7)
    #8 0x7f84117143  (<unknown module>)
    #9 0x7f8411593b  (/lib/aarch64-linux-gnu/libpthread-2.27.so+0x2093b)
    #10 0x7f8410d057  (/lib/aarch64-linux-gnu/libc-2.27.so+0x16057)
    #11 0x7f841556a7  (<unknown module>)
    #12 0x7f84265807 in foreach_func /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2_port/libgphoto2_port/gphoto2-port-info-list.c:234
    #13 0x7f8424dab7  (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x3ab7)
    #14 0x7f8424d4a3  (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x34a3)
    #15 0x7f8424e0b3 in lt_dlforeachfile (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x40b3)
    #16 0x7f84265cbf in gp_port_info_list_load /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2_port/libgphoto2_port/gphoto2-port-info-list.c:286
    #17 0x7f8445d19b in gp_camera_init /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2/gphoto2-camera.c:724
    #18 0x555b98bd7b in main (/home/jpad/prog/C/test/gphoto2/gphoto2+0xd7b)
    #19 0x7f843196df in __libc_start_main (/lib/aarch64-linux-gnu/libc.so.6+0x206df)
    #20 0x555b98bc13  (/home/jpad/prog/C/test/gphoto2/gphoto2+0xc13)

Direct leak of 4096 byte(s) in 1 object(s) allocated from:
    #0 0x7f844f376f in malloc (/usr/lib/aarch64-linux-gnu/liblsan.so.0+0xe76f)
    #1 0x7f83fb677f  (/usr/local/lib/libgphoto2.so.6.1.0+0x3477f)
    #2 0x7f83fb9ed3  (/usr/local/lib/libgphoto2.so.6.1.0+0x37ed3)
    #3 0x7f83fbde87  (/usr/local/lib/libgphoto2.so.6.1.0+0x3be87)
    #4 0x7f83fbe2df  (/usr/local/lib/libgphoto2.so.6.1.0+0x3c2df)
    #5 0x7f83fc011f  (/usr/local/lib/libgphoto2.so.6.1.0+0x3e11f)
    #6 0x7f83fc0433  (/usr/local/lib/libgphoto2.so.6.1.0+0x3e433)
    #7 0x7f83fc1ae7  (/usr/local/lib/libgphoto2.so.6.1.0+0x3fae7)
    #8 0x7f83fea143  (/usr/local/lib/libgphoto2.so.6.1.0+0x68143)
    #9 0x7f83fe893b  (/usr/local/lib/libgphoto2.so.6.1.0+0x6693b)
    #10 0x7f83fe0057  (/usr/local/lib/libgphoto2.so.6.1.0+0x5e057)
    #11 0x7f84153deb  (<unknown module>)
    #12 0x7f84268337 in gp_port_init /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2_port/libgphoto2_port/gphoto2-port.c:118
    #13 0x7f84268337 in gp_port_set_info /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2_port/libgphoto2_port/gphoto2-port.c:199
    #14 0x7f84459f23 in gp_abilities_list_detect /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2/gphoto2-abilities-list.c:422
    #15 0x7f8445d1b7 in gp_camera_init /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2/gphoto2-camera.c:725
    #16 0x555b98bd7b in main (/home/jpad/prog/C/test/gphoto2/gphoto2+0xd7b)
    #17 0x7f843196df in __libc_start_main (/lib/aarch64-linux-gnu/libc.so.6+0x206df)
    #18 0x555b98bc13  (/home/jpad/prog/C/test/gphoto2/gphoto2+0xc13)

Direct leak of 4096 byte(s) in 1 object(s) allocated from:
    #0 0x7f844f376f in malloc (/usr/lib/aarch64-linux-gnu/liblsan.so.0+0xe76f)
    #1 0x7f83fdc77f  (/usr/local/lib/libgphoto2.so.6.1.0+0x5a77f)
    #2 0x7f83fdfb97  (/usr/local/lib/libgphoto2.so.6.1.0+0x5db97)
    #3 0x7f8411711b  (<unknown module>)
    #4 0x7f8411593b  (/lib/aarch64-linux-gnu/libpthread-2.27.so+0x2093b)
    #5 0x7f8410d057  (/lib/aarch64-linux-gnu/libc-2.27.so+0x16057)
    #6 0x7f841556a7  (<unknown module>)
    #7 0x7f84265807 in foreach_func /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2_port/libgphoto2_port/gphoto2-port-info-list.c:234
[... 89 lines cut ...]
    #5 0x7f83fe0057  (/usr/local/lib/libgphoto2.so.6.1.0+0x5e057)
    #6 0x7f84153deb  (<unknown module>)
    #7 0x7f84268337 in gp_port_init /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2_port/libgphoto2_port/gphoto2-port.c:118
    #8 0x7f84268337 in gp_port_set_info /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2_port/libgphoto2_port/gphoto2-port.c:199
    #9 0x7f84459f23 in gp_abilities_list_detect /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2/gphoto2-abilities-list.c:422
    #10 0x7f8445d1b7 in gp_camera_init /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2/gphoto2-camera.c:725
    #11 0x555b98bd7b in main (/home/jpad/prog/C/test/gphoto2/gphoto2+0xd7b)
    #12 0x7f843196df in __libc_start_main (/lib/aarch64-linux-gnu/libc.so.6+0x206df)
[... cut ...]
    #5 0x7f8446014f in gp_camera_capture_preview /home/jpad/Downloads/installers/libgphoto2-2.5.27/libgphoto2/gphoto2-camera.c:1403
    #6 0x555b98bdf7 in main (/home/jpad/prog/C/test/gphoto2/gphoto2+0xdf7)
    #7 0x7f843196df in __libc_start_main (/lib/aarch64-linux-gnu/libc.so.6+0x206df)
    #8 0x555b98bc13  (/home/jpad/prog/C/test/gphoto2/gphoto2+0xc13)

SUMMARY: LeakSanitizer: 35927 byte(s) leaked in 251 allocation(s).
makefile:16: recipe for target 'run' failed
make: *** [run] Error 23
```

RReverser, 2021-08-03:
```
    #2 0x7f7fe2c9c7  (<unknown module>)
    #3 0x7f7fe33947  (<unknown module>)
    #4 0x7f7fe47d47  (<unknown module>)
    ```

suggests there is still some code not rebuilt with debug symbols.

But at least you can take a look at the line it gives at gphoto2-camera.c:1403 right before that.

RReverser, 2021-08-03:
> Maybe you can reproduce it?

As I said above, I didn't see any more leaks in my app, but it could very well be camera-specific.

notnot, 2021-08-03:
> > Maybe you can reproduce it?
> 
> As I said above, I didn't see any more leaks in my app, but it could very well be camera-specific.

I am running this with a Sony A6300, which uses ptp. I figure your A6600 will use the same code paths?

RReverser, 2021-08-03:
Unfortunately, that line - 1403 - does not correspond to anything on current master, so it appears your copy is still out of date: https://github.com/gphoto/libgphoto2/blob/04fed07020ccf25b2a5c6d730e707e01e49eb02a/libgphoto2/gphoto2-camera.c#L1403

But, either way, mind you, I'm not a maintainer or anything - just a fellow user who happened to debug issues that arose with my camera and contributed some fixes and trying to help the best I can :)

I'm sure you'll be able to find out what's causing leaks in your app too, you just need some time to go over those codepaths the leak indicated with a debugger or by inserting some logs.

notnot, 2021-08-03:
> Unfortunately, that line - 1403 - does not correspond to anything on current master, so it appears your copy is still out of date:

Yes that is true, i manually applied the changes in the official 2.5.27, because at the location where my machine is, the internet is very very bad. I was even lucky to get the diff page on screen! I should download the current master at home, save it to stick and install it on site. Tomorrow, or the day after...

Thanks for your concerns, i hope Marcus will have a look at it. I lost most of my C skills lol. It feels like i entered the rabbit hole! Good thing is that i can now produce leak sanitizer logs, thanks to you :)

notnot, 2021-08-03:
Line 1403 in my gphoto2-camera-c source:
CHECK_RESULT_OPEN_CLOSE (camera, camera->functions->capture_preview (

RReverser, 2021-08-03:
> Line 1403 in my gphoto2-camera-c source:
> CHECK_RESULT_OPEN_CLOSE (camera, camera->functions->capture_preview (

Ah right, that's not very helpful then. A bit too high-level API to tell anything useful. You'll need to clean & recompile the rest with debug flags and without opts as described above to see missing frames and figure out where the leak _actually_ occurs.

msmeissn, 2021-08-04:
the strdup in the leak trace could just be this ptp_getobjectinfo thing.

build libgphoto2 also with CFLAGS="-O2 -DVALGRIND -g"   .... the -DVALGRIND disables the unloading of the camera drivers so their symvbols stay in memory.

notnot, 2021-08-05:
I can confirm that today's master has no preview capture memory leak any longer. 

I still have two (not so important) questions:

1) Why did my earlier 'manual fix' attempt not work? Applying the changes to ptp2/library.c, running 'make' and 'sudo make install'. All the libgphoto2 things that were installed were new (looking at the timestamp). Should i have done 'make clean' as a first step?

2) How to get rid of the gp_camera_init() related leaks? I am doing gp_camera_exit(camera, context);
gp_camera_free(camera); but the leaks stay.

RReverser, 2021-08-05:
> I can confirm that today's master has no preview capture memory leak any longer.

Great, glad to hear! Sounds like it should've been came codepath then.

>     1. Why did my earlier 'manual fix' attempt not work?

You might have missed some commits as the fixes were split between Sony and other cameras. That's why I asked earlier about `master`:

> Just to be clear - 2.5.27 release or the master of the repo (2.5.27.1)? Only the master has the fix for the leak.

> How to get rid of the gp_camera_init() related leaks? I am doing gp_camera_exit(camera, context);
gp_camera_free(camera); but the leaks stay.

Judging by the fact those all are coming from `gp_port_init`, I suspect it's just one-off initialization for the lifetime of the program, and not really a leak in traditional sense, and so can be ignored (@msmeissn please correct me if I'm wrong).

msmeissn, 2021-08-06:
it is probably just 1 time initialization, yes. without better debug symbols its a bit hard to say.

(But I see similar one time leaks here.)

RReverser, 2021-08-07:
I think it's fair to close this issue for now; testing-wise I assume people will report new issues when they run into problems.

It could be really nice if gphoto tests could be automated somehow, but I doubt it's possible without some sort of full-grown camera emulators (or a server with a bunch of cameras connected somewhere).

closed 2021-08-07 by RReverser

msmeissn, 2021-08-07:
yes

RReverser, 2021-08-08:
I wonder if it makes sense to at least build and ship gphoto2 CLI with `-fsanitize=leak`.

Unlike full-on address sanitizer, it has very low overhead (definitely negligible compared to USB I/O speed as well as those `sleep`s here and there) and it could help people report any leaks they run into with their cameras in the wild.

(The false leaks above can be configured to be excluded so people won't see any leaks that are not real.)