gphoto/libgphoto2 #693: First capture preview takes a very long time on Sony a6600

RReverser, 2021-07-27:
**Describe the bug**

First call to `--capture-preview` or its API variant takes a very long time on Sony a6600 if the camera wasn't connected before.

**Name the camera**

Sony a6600 (USB PTP)

**libgphoto2 and gphoto2 version**

2.5.27.1 (master)

**To Reproduce**

1. Turn on the Sony camera.
2. Connect it via USB to the computer.
3. Message "Connecting..." and a USB logo appear on the camera screen.
4. Run `time gphoto2 --auto-detect --capture-preview`. Application is stuck for a while, and eventually saves the capture preview in 20 seconds.

Log attached: [debug.log](https://github.com/gphoto/libgphoto2/files/6887836/vm2.log). 

Judging by the log, the underlying `gp_libusb1_read` operation times out after 20 seconds. Is that its actual timeout? if so, it seems like a lot.

When experimenting with libgphoto2 API, I found that even waiting 1 second after creating the context & camera and before capturing the actual preview seems to return result successfully and a lot faster, but I'm not sure how reliable that specific number is, plus it introduces unnecessary delay if camera was already connected.

RReverser, 2021-08-07:
@msmeissn I don't know what's the reason for this one or how to actually fix it... does anything in the logs stand out to you by any chance? Or is the only solution really just waiting 1 second before starting any preview, "just in case"?

msmeissn, 2021-08-25:
Sorry missed this ticket earlier ... hmm

Some other newer Sony cameras have a mandatory wait of 2.5 - 3 seconds before they can startup preview and other things ,.. might be needed for the 6600 too.

What I see it times out on the query of the preview picture objectinfo... not sure how to avoid this, except by putting a small wait in front like you said (we have something similar in camera_sony_capture already, we can add similar to preview 6600 capture)

msmeissn, 2021-08-25:
the 20 seconds is the default timeout, it could be reduced temporary during this operation too.

RReverser, 2021-08-25:
> not sure how to avoid this

I wonder if there's any operation or property we can poll to detect what camera has fully "warmed up". Or, I guess, if the initial delay works for capture image, then yeah, we can add it to the preview too. Just need to be careful to make sure it only happens once and not for each preview.

msmeissn, 2021-08-25:
i tried to find this with the Sony ZV1 then, not much luck of any indicators. hmm.

i added a patch that reduces the timeout to 1 second fpr this function, can you check if it helps?

RReverser, 2021-08-25:
Can't check right now - currently busy with some other stuff at work - but I'll try as soon as I return to this project.

RReverser, 2021-09-14:
Hm, now just getting transfer error if camera wasn't previously connected (same repro steps as described in the issue above). If it was already connected, then preview succeeds.