gphoto/libgphoto2 #694: Can't set shutter speed to Bulb on Sony a6600

RReverser, 2021-07-27:
**Describe the bug**

If current shutter speed is not Bulb already, it's not possible to set the Bulb one via libgphoto2 / gphoto2.

**Name the camera**

Sony a6600

**libgphoto2 and gphoto2 version**

2.5.27.1 (master)

**To Reproduce**

Try `gphoto2 --auto-detect --set-config shutterspeed=Bulb` on a Sony camera. It will get stuck for a bit, and eventually give up.

[Logs](https://github.com/gphoto/libgphoto2/files/6887902/vm2.log)

As far as I can tell, this is very similar to the issue I fixed in https://github.com/gphoto/libgphoto2/pull/678, but shutter speed doesn't reuse same functions. Perhaps it should / could? If not, similar fix needs to be implemented in `put_Sony_ShutterSpeed` too (I can try to do this later, but for me it's a lower priority for now).

changedsoul, 2022-12-12:
Same problem on a7RII.
Camera gets set to 30" and not bulb.
issuing a -B 60 also will not open shutter for 60 seconds. Seems the camera knows its only set for 30" and ignores what its being told.

msmeissn, 2022-12-19:
I put in a temporary hack to libgphoto2 master., perhaps you can test

changedsoul, 2022-12-21:
Would be glad to. 
How do I test it out once I clone the git repo?

msmeissn, 2022-12-21:
it should pick the closest one. we have a fixed list inside libgphoto2, as the sony does not expose the property,  which we might need to adjust too. debuglogfile there would be helpful.
thanks for the feedback that bulb now works :)

changedsoul, 2022-12-22:
[log.txt](https://github.com/gphoto/libgphoto2/files/10283399/log.txt)
So I did some testing and it still seems to have some bugs, but the BULB mode is at least getting set.
What I have found is an issue with setting the shutter speed seems strange. For example, My camera has 1", 1.3", 1.6", 2", 2.5", 3.2"..... If I am above 3.2, and issue a command to set 3.2, it will set the camera to 2.5 and not 3.2. 
This happens with other speeds as well. 

When I issue: gphoto2 --get-config=shutterspeed
It returns this as my shutter speeds. Not sure if this helps.

Label: Shutter Speed                                                           
Readonly: 0
Type: RADIO
Current: 1
Choice: 0 30
Choice: 1 25
Choice: 2 20
Choice: 3 15
Choice: 4 13
Choice: 5 10
Choice: 6 8
Choice: 7 6
Choice: 8 5
Choice: 9 4
Choice: 10 32/10
Choice: 11 25/10
Choice: 12 2
Choice: 13 16/10
Choice: 14 13/10
Choice: 15 1
Choice: 16 8/10
Choice: 17 6/10
Choice: 18 5/10
Choice: 19 4/10
Choice: 20 1/3
Choice: 21 1/4
Choice: 22 1/5
Choice: 23 1/6
Choice: 24 1/8
Choice: 25 1/10
Choice: 26 1/13
Choice: 27 1/15
Choice: 28 1/20
Choice: 29 1/25
Choice: 30 1/30
Choice: 31 1/40
Choice: 32 1/50
Choice: 33 1/60
Choice: 34 1/80
Choice: 35 1/100
Choice: 36 1/125
Choice: 37 1/160
Choice: 38 1/200
Choice: 39 1/250
Choice: 40 1/320
Choice: 41 1/400
Choice: 42 1/500
Choice: 43 1/640
Choice: 44 1/800
Choice: 45 1/1000
Choice: 46 1/1250
Choice: 47 1/1600
Choice: 48 1/2000
Choice: 49 1/2
Choice: 50 1/3200
Choice: 51 1/4000
Choice: 52 1/5000
Choice: 53 1/6400
Choice: 54 1/8000
Choice: 55 1/10000
Choice: 56 1/12500
Choice: 57 1/16000
Choice: 58 1/20000
Choice: 59 1/25000
Choice: 60 1/32000
Choice: 61 Bulb
END