gphoto/libgphoto2 #695: Capture Image with out triggering autofocus

jamesCreateAndConnect, 2021-07-27:
Hello, we're using a Canon Rebel SL3 with a Raspberry Pi 4B+ to first take one image to focus then take 81 photos in succession on a moving platform to get different lighting angles. Throughout this process, it is necessary for the focus to remain the exact same, however, each photo triggers the autofocus which becomes apparent over time. 

I've done quite a bit of searching and testing but have yet found a way to capture an image without triggering the autofocus.

Am I just missing something and if not will this ever be a feature?

hfiguiere, 2021-07-27:
> I've done quite a bit of searching and testing but have yet found a way to capture an image without triggering the autofocus.

Doesn't the lens allow disabling autofocus directly? All my EF lenses have a MF button that disable autofocus.

jamesCreateAndConnect, 2021-07-27:
Yes, it does! I should have clarified. The machine uses autofocus on a "setup" image in order to lock in where it must remain focused for the rest of the pictures. The goal is complete autonomy.

jamesCreateAndConnect, 2021-08-09:
I think a better question is, is there a way to lock focus

msmeissn, 2021-08-10:
hmm, there is focusmode option which can be set to Manual in P / A/ V or M modes I think.

not sure if we expose the focuslock property correctly.

msmeissn, 2021-08-10:
or use eosremoterelease=Immediate   --wait-event-and-download=5s this will emulate a ful shutterpress without half-press (focusing)

thexangle, 2021-08-10:
Not sure you have the option in the custom functions of the SL3 but on the SL1, you can disable the AF on halfpress using the AE lock/AF option. 
Check https://www.dummies.com/photography/cameras/canon-camera/customize-ae-lock-shutter-buttons-canon-rebel-t61300d/ for more details.

jamesCreateAndConnect, 2021-08-11:
@msmeissn The eosremoterelease=Immediate works. After setting it, we are able to trigger another single capture without the half-press triggering autofocus. However, it only allows for the single next trigger to behave in that way. So we have to set it to immediate each time we want to trigger the capture. But it works! Is this intended to work until changed or is the current behaviour to be expected?

@thexangle Just checked and tested that manually, you're correct! It disables the autofocus on half-press. I'm going to try to access the custom functions from gphoto2.

thexangle, 2021-08-11:
From what we gathered, the custom function string for the SL3 is surprisingly small.
I dont have any SL3 available right now but i noted that it looked like that: 
20,1,7,14,1,60f,1,1,
Meaning only 1 custom func is mapped (the Mirror lockup if I'm not mistaken). 

Speaking of custom_func on SL3:  We had some weird bugs when trying the set the MLU from gphoto2 via the custom_func string. 
In some random cases, it would reset the choices of shutterspeed to only values > 1 sec and the only way to revert that would be to manually reset the custom functions on the camera. We are not 100% sure this bug is related to gphoto2 but I'll open a new bug when I can get my hands on a SL3 if I manage to reproduce the issue. 
In the meantime, @msmeissn any idea why the custom_func string is so small (same goes for the newer models like 850d/t8i) and have you ever heard of a bug like the one i mentioned above ?

msmeissn, 2021-08-12:
can you capture a debug log of gphoto2 --list-config

the camera might be sending multiple customfunc strings and we only display the last or the first one :( 
meaning we might have some shortcoming there.

do you know the structure of this customdata better?

thexangle, 2021-08-13:
I'll get the logs as soon as I can get an SL3 i can torture at the office:) Thanks @msmeissn