gphoto/libgphoto2 #704: multicamera

natxopedreira, 2021-08-09:
Hello

Im trying to use 2 Canon EOS 2000D cameras with a c++ app, using only one works great but as soon as i want to use the second one i get `error 53 Could not claim the USB device`

Im able to open both no  problem, and get info like name and model from both but problem seems to happens when i want to get liveview from second camera

Im on xubuntu 20 and lib version is 2.5.25-0ubuntu0.1

Any clue on what can be wrong here?¿

Thanks!

BeebBenjamin, 2021-08-09:
For using multiple cameras, you usually have to specify '--port' after parsing out the usb port address from a '--auto-detect' response.  If your cameras are the same make, you'll also need to include '--camera' to stop force gphoto2 to only try to capture on a particular camera.  So for example 'gphoto2 --port=usb:000,001 --camera="Canon EOS 2000D" --capture-image' note you have to enclose the camera name in quotes otherwise it doesn't work.

BeebBenjamin, 2021-08-09:
Also, are the cameras unmounted when you try?

natxopedreira, 2021-08-09:
Thank you @BeebBenjamin 

Note im not using gphoto2 app, im using the library libgphoto in a custom c++ app

Yep, cameras are unmounted

natxopedreira, 2021-08-09:
Solved, you give me the clue, before init the camera i set the port data.

Thanks

closed 2021-08-09 by natxopedreira