gphoto/libgphoto2 #708: Integrate libgphoto2/libgphoto2_port/configure.ac into libgphoto2/configure.ac

ndim, 2021-08-13:
**Is your feature request related to a problem? Please describe.**
Running `configure --help` in a libgphoto2 source tree only shows the configure options for libgphoto2, but not for libgphoto2_port. That is confusing, and bad documentaiton.

**Describe the solution you'd like**
I would like to integrate `libgphoto2/libgphoto2_port/configure.ac` into `libgphoto2/configure.ac` so that a single `configure --help` run shows all configuration options. That also should accelerate the time needed to run `configure` for libgphoto2.

**Describe alternatives you've considered**
Keep explaining libgphoto2_port options to libgphoto2 users, and keep waiting longer for `configure` to finish.

**Additional context**
This is completely unrelated to whether the binaries built (`libgphoto2.so` and `libgphoto2_port.so`) are two separate ones or one integrated one.

This issue is so that I remember to do this, and so I can collect related comments here.

ndim, 2021-08-13:
One interesting part will be whether the translations for libgphoto2 an libgphoto2_port can be kept separate as they are with a single common integrated configure.ac.

jtojnar, 2023-04-19:
Alternative might be switching to something like [Meson build system](https://mesonbuild.com/), which has native support for subprojects. That way you could maintain a clean separation of the libraries while able to pass options to `libgphoto2_port`.

ndim, 2024-01-30:
JFTR, it is possible to keep the libgphoto2 `po/` subdir separate from the libgphoto2_port `po/` subdir even with a single `configure.ac`.