gphoto/libgphoto2 #709: Resolution of --capture-movie is odd.

PugzAreCute, 2021-08-14:
**Describe the bug**

Resolution of `gphoto2 --stdout --capture-movie` is 640x424, instead of full hd (Setting set on camera)

**Name the camera**
Nikon DSC D3200

**libgphoto2 and gphoto2 version**
gphoto2         2.5.20
libgphoto2      2.5.22
libgphoto2_port 0.12.0 

**To Reproduce**
Steps to reproduce the behavior:
Run command `gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video2`, open a software like OBS, the resolution is not FullHD. I have only tested it on a Nikon D3200 As of now.

Log : Currently too big, will send if requested

msmeissn, 2021-08-14:
is gphoto2 --summary | grep d1ac 
showing anything?
I fear it might not, so the liveview size might not be controllable...

the d1ac property, is also available decoded as  "liveviewsize"  in libgphoto2 2.5.26 and newer.

PugzAreCute, 2021-08-14:
```bash
pugzarecute@debinax:~$ gphoto2 --summary | grep d1ac
pugzarecute@debinax:~$ 
```
Nope, no output., but to be fair my camera is pretty old

msmeissn, 2021-08-14:
i fear it wont be adjustable :/

PugzAreCute, 2021-08-14:
Ah ok, i see, ill try to take a movie on the cam itself to see if it can do full hd by itself

PugzAreCute, 2021-08-14:
Looks like it can by itself

msmeissn, 2021-08-14:
yes, but just to be clear capturing a movie on camera (in h264) is different from --capture-movie , which just repeatedly captures the liveview jpeg frame ;)

PugzAreCute, 2021-08-14:
Ahh, ok!

PugzAreCute, 2021-08-14:
Any way i can get the H264 feed live or no?

msmeissn, 2021-08-14:
i am afraid i have not seen any regular camera that can stream h264 direct.

PugzAreCute, 2021-09-07:
Closing as it is a camera limitation

closed 2021-09-07 by PugzAreCute

Binklebonk, 2021-12-10:
> 

It's a non-libgphoto2 solution, but as the question was "**Any way** I can get...", I feel compelled to suggest you may be able to utilise the cameras HDMI output, connected to a USB capture device, which will typically be detected by OSX/Windows/Linux as a webcam.  I'm able to achieve true 1080p with this.  Higher may be possible depending on your camera and capture device.

I must caution, that in my testing (with Nikon D750), I can not simultaneously output HDMI from the camera (requires LiveView to be active), whilst controlling it via libgphoto2 / gphoto2, or even having USB connected to the camera.  It's an exclusively one or the other deal, it seems.