gphoto/libgphoto2 #712: Nikon D3500 Livew View

AdamLee19, 2021-08-30:
Hey there,

I am using Nikon D3500 for my project. I am wondering is there any function I could use in libgphoto2 to get live stream video/pointer from my camera and display the video in my application? I tried to find some example code online, but I could not find any.

I appreciate any hint or support. Thank you in advance.

msmeissn, 2021-08-31:
The libgphoto2 API has 

int gp_camera_capture_preview    (Camera *camera, CameraFile *file,
                                  GPContext *context);

Whichs returns 1 JPEG frame of the liveview (via a CameraFile* struct).

You can push this JPEG to your rendering pipeline.

The D3500 I have noted down as able to capture liveview.

AdamLee19, 2021-09-02:
> The libgphoto2 API has
>
> ```
> int gp_camera_capture_preview    (Camera *camera, CameraFile *file,
>                                   GPContext *context);
> ```
>
> Whichs returns 1 JPEG frame of the [...]

Thank you Marcus! That helps. I put this function in a loop. I can get a live view now.
Btw, in case some one did same thing as what I am doing, here is some tips:
1. For my Nikon D3500 camera, I turned on viewfinder first before I used this function
2. Since I need a real time live view, the gphoto2 keeps sending me frames so that gphoto2 occupies I/O. We can not control the camera while reading images. Otherwise, it will through out errors says "I/O busy"
3. To avoid it, I am trying to put the preview function to a thread. I am still working on it. I don't know if that works. I am using python-gphoto2 and PyQt5 now. (if anyone would like to share some ideas?)

But, thank you Marcus! I appreciated your help!

closed 2021-09-02 by AdamLee19

msmeissn, 2021-09-03:
you need to understand that USB devices are single-session... 

while you can write a multithreaded program, you have only 1 command/data stream to/ from the camera.

So you need to mix/interleave the preview command/data flow with other control commands.