gphoto/libgphoto2 #713: What if I got all the photos and could just show the preview or thumbnail?

SiuyuenChow, 2021-08-31:
hello，

How can I quickly get a preview or thumbnail after I use gp_camerA_file_get_info to get all the image data in all folders?
Now I'm downloading photos and using gp_camerA_file_read to get data. If I preview images using gp_camerA_file_read, it's ok, but it will affect the download of photos.
Excuse me, is there any way to get the preview quickly?

msmeissn, 2021-08-31:
you will need to run for each image gp_camera_file_get  with GP_FILE_TYPE_PREVIEW  as type. ... this should retrieve only the preview part,. so not the full pixcture and be faster.

SiuyuenChow, 2021-08-31:
> you will need to run for each image gp_camera_file_get with GP_FILE_TYPE_PREVIEW as type. ... this should retrieve only the preview part,. so not the full pixcture and be faster.

So can I only use gp_camerA_file_read if I want to show a preview or download an image? However, GP_FILE_TYPE_PREVIEW does not retrieve the data

logdumper ========== Reading 22274 bytes from file 'IMG_0992.JPG' at offset 0.
errordumper========='gp_filesystem_read_file (camera->fs, folder, file, type, offset, buf, size, context)' failed: -6
logdumper ========== 'gp_filesystem_read_file (camera->fs, folder, file, type, offset, buf, size, context)' failed: -6
logdumper ========== Getting file 'IMG_0992.JPG' in folder '/store_00020001/DCIM/100CANON'...
logdumper ========== (storage=0x00020001, handle=0x00000000)
logdumper ========== (storage=0x00020001, handle=0x90000000)
logdumper ========== (storage=0x00020001, handle=0x91900000)
logdumper ========== Getting file 'IMG_0992.JPG' in folder '/store_00020001/DCIM/100CANON'...
logdumper ========== Getting file 'IMG_0992.JPG' in folder '/store_00020001/DCIM/100CANON'...
logdumper ========== Getting file 'IMG_0992.JPG' in folder '/store_00020001/DCIM/100CANON'...
logdumper ========== Getting file 'IMG_0992.JPG' in folder '/store_00020001/DCIM/100CANON'...
logdumper ========== Getting file 'IMG_0992.JPG' in folder '/store_00020001/DCIM/100CANON'...
logdumper ========== Reading 22274 bytes from file 'IMG_0992.JPG' at offset 0.
logdumper ========== Getting file 'IMG_0992.JPG' in folder '/store_00020001/DCIM/100CANON'...
logdumper ========== Getting file 'IMG_0992.JPG' in folder '/store_00020001/DCIM/100CANON'...
logdumper ========== Getting file 'IMG_0992.JPG' in folder '/store_00020001/DCIM/100CANON'...
errordumper========='gp_filesystem_read_file (camera->fs, folder, file, type, offset, buf, size, context)' failed: -6
logdumper ========== 'gp_filesystem_read_file (camera->fs, folder, file, type, offset, buf, size, context)' failed: -6

SiuyuenChow, 2021-08-31:
Here's what I did to get all the images in the camera:
Gp_list_new -> gp_camerA_folder_list_folders -> gp_list_sort -> gp_list_count -> gp_list_reset Gp_camera_folder_list_files -> Traversal GP_list_count -> In traversal gp_CAMERA_FILe_get_info obtains information about each file

SiuyuenChow, 2021-09-01:
hello again,
When I call
gp_camera_file_get(camera, [folder UTF8String], [fileName UTF8String], GP_FILE_TYPE_PREVIEW, file, context);
gp_file_get_data_and_size(file, &imagedata, &size);
Sometimes the returned imageData == null

closed 2021-09-01 by SiuyuenChow