gphoto/libgphoto2 #714: The picture information obtained is not correct

SiuyuenChow, 2021-09-02:
When I listen to
When gp_camera_wait_for_event gets a new photo, it sometimes gets the wrong size of the photo information, as shown below

For example, like **img_2710.jpg,** the size returned is already 4.0GB,

 fileName =================== IMG_2709.JPG  ==== 1802202
 **fileName =================== IMG_2710.JPG  ==== 4394586240**
 fileName =================== IMG_2711.JPG  ==== 1865438
 fileName =================== IMG_2712.JPG  ==== 1884329

I got the photo information from gp_camera_file_get_info

SiuyuenChow, 2021-09-02:
> has the get info set the flag bit in "fields" of GP_FILE_INFO_SIZE
> 
> what camera was it?

Camera model :EOS 6D
There's nothing set up. It's an accident. I don't know what's inevitable.
At present, the observed situation is that when the continuous shooting, the probability of occurrence is relatively large

msmeissn, 2021-10-02:
tried reproducing with eos m6 mark2 - not seen so far.