gphoto/libgphoto2 #716: Can't capture (and delete files) on Nikon 1 J1 with manual lens

rainbow-app, 2021-09-06:
Hello.

I'm not sure if it's a bug, may be my problem.

I can't capture on Nikon 1 J1 with manual lens (=fully mechanical, no electronics). PTP I/O Error (see below).

Getting-files from camera works fine.

I don't have native Nikon lens, so cannot check if the problem is in the lack of communication with the lens. Capturing with the shutter button on the camera (of course) works (need to set exposure mode to Manual). I tried another cable, and other USB ports.

I'm new to gphoto. Could somebody please tell me, should it work or not?

Thanks.

```
Model                          Port
----------------------------------------------------------
Nikon J1                       usb:005,002

$ gphoto2 -v --capture-image
gphoto2 2.5.27

...

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.27         gcc, popt(m), no exif, no cdk, no aa, no jpeg, readline
libgphoto2      2.5.27         standard camlibs (SKIPPING docupen lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

*** Error ***
PTP I/O Error
ERROR: Could not capture image.
ERROR: Could not capture.

*** Error ***
PTP I/O Error

$ gphoto2  -f /store_00010001/DCIM/367_2911 --delete-file 1

*** Error ***
PTP I/O Error
*** Error (-7: 'I/O problem') ***

For debugging messages, please use the --debug option.
...
```

msmeissn, 2021-10-02:
can you attach the full debuglogfile?

gphoto2 --debug-logfile=Capture.log --debug --capture-image

rainbow-app, 2021-10-02:
0.000144 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000174 main                        (2): gphoto2 2.5.27
0.000186 main                        (2): gphoto2 has been compiled with the following options:
0.000193 main                        (2):  + gcc (C compiler used)
0.000199 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000206 main                        (2):  + no exif (for displaying EXIF information)
0.000212 main                        (2):  + no cdk (for accessing configuration options)
0.000217 main                        (2):  + no aa (for displaying live previews)
0.000223 main                        (2):  + no jpeg (for displaying live previews in JPEG format)
0.000229 main                        (2):  + readline (for easy navigation in the shell)
0.000240 main                        (2): libgphoto2 2.5.27
0.000249 main                        (2): libgphoto2 has been compiled with the following options:
0.000258 main                        (2):  + standard camlib set SKIPPING SOME (ax203 canon digigr8 dimagev directory jl2005a jl2005c kodak_dc240 mars pentax ptp2 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801 SKIPPING docupen lumix)
0.000264 main                        (2):  + gcc (C compiler used)
0.000271 main                        (2):  + ltdl (for portable loading of camlibs)
0.000277 main                        (2):  + EXIF (for special handling of EXIF files)
0.000283 main                        (2): libgphoto2_port 0.12.0
0.000292 main                        (2): libgphoto2_port has been compiled with the following options:
0.000298 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.000304 main                        (2):  + gcc (C compiler used)
0.000310 main                        (2):  + ltdl (for portable loading of iolibs)
0.000316 main                        (2):  + EXIF (for vusb)
0.000322 main                        (2):  + USB (libusb1, for USB cameras)
0.000328 main                        (2):  + serial (for serial cameras)
0.000333 main                        (2):  + no resmgr (serial port access and locking)
0.000339 main                        (2):  + no ttylock (serial port locking)
0.000345 main                        (2):  + no lockdev (serial port locking)
0.000351 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000357 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000362 main                        (2): invoked with following arguments:
0.000368 main                        (2):   --debug-logfile=n1.txt
0.000374 main                        (2):   --debug
0.000380 main                        (2):   --capture-image
0.000439 load_settings               (2): Creating gphoto config directory ('/home/username/.gphoto')
0.000523 load_settings               (2): Loading settings from file '/home/username/.gphoto/settings'.
0.000615 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.000633 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/local/lib/libgphoto2_port/0.12.0'...
0.000806 foreach_func                (2): Called for filename '/usr/local/lib/libgphoto2_port/0.12.0/disk'.
0.001101 gp_port_library_list        (2): found fstab fsname UUID=531c6bc6-56dc-4b7a-bec7-f2bb591867cb
0.001242 gp_port_library_list        (2): found mtab fsname sysfs
0.001258 gp_port_library_list        (2): found mtab fsname proc
0.001269 gp_port_library_list        (2): found mtab fsname udev
0.001297 gp_port_library_list        (2): found mtab fsname devpts
0.001308 gp_port_library_list        (2): found mtab fsname tmpfs
0.001335 gp_port_library_list        (2): found mtab fsname /dev/sda1
0.001362 gp_port_library_list        (2): found mtab fsname none
0.001388 gp_port_library_list        (2): found mtab fsname none
0.001406 gp_port_library_list        (2): found mtab fsname tmpfs
0.001442 gp_port_library_list        (2): found mtab fsname pstore
0.001474 gp_port_library_list        (2): found mtab fsname tmpfs
0.001502 gp_port_library_list        (2): found mtab fsname cgroup
0.001512 gp_port_library_list        (2): found mtab fsname systemd
0.001522 gp_port_library_list        (2): found mtab fsname tmpfs
0.001559 gp_port_library_list        (2): found mtab fsname /dev/sda2
0.001588 gp_port_library_list        (2): found mtab fsname /dev/sdb3
0.001614 gp_port_library_list        (2): found mtab fsname tmpfs
0.001685 foreach_func                (2): Loaded '' ('^disk:') from '/usr/local/lib/libgphoto2_port/0.12.0/disk'.
0.001694 foreach_func                (2): Called for filename '/usr/local/lib/libgphoto2_port/0.12.0/ptpip'.
0.001932 foreach_func                (2): Loaded 'PTP/IP Connection' ('ptpip:') from '/usr/local/lib/libgphoto2_port/0.12.0/ptpip'.
0.001944 foreach_func                (2): Loaded '' ('^ptpip:') from '/usr/local/lib/libgphoto2_port/0.12.0/ptpip'.
0.001952 foreach_func                (2): Loaded 'IP Connection' ('ip:') from '/usr/local/lib/libgphoto2_port/0.12.0/ptpip'.
0.001960 foreach_func                (2): Loaded '' ('^ip:') from '/usr/local/lib/libgphoto2_port/0.12.0/ptpip'.
0.001967 foreach_func                (2): Called for filename '/usr/local/lib/libgphoto2_port/0.12.0/serial'.
[... cut ...]
60.327819 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
60.327833 gp_filesystem_lru_clear     (2): fscache LRU list already empty
60.327846 delete_all_folders          (2): Internally deleting all folders from '/'...
60.327861 lookup_folder               (2): Lookup folder '/'...
60.327875 lookup_folder               (2): Found! / is 0x556b48129410
60.327890 recurse_delete_folder       (2): Recurse delete folder 0x556b48129410//
60.327905 delete_all_files            (2): Delete all files in folder 0x556b48129410//

rainbow-app, 2021-10-02:
Spaces were lost in the above, hope it's ok.

Not sure if it will be helpful, just in case: delete log on google [drive](https://drive.google.com/file/d/1JxQptoGrkw5QIW-id4HhGVRCaL7X5DJc/view?usp=sharing) (160KB; the above was 45KB).

msmeissn, 2021-10-03:
camera seems to have crashed on changecameramode.
do other lenses work actually?

msmeissn, 2021-10-03:
ok, i did now similar changes for the J1 as we did for the V1.

In current libgphoto2 git.

rainbow-app, 2021-10-05:
Now works :) Both capture and delete. Thank you very much!

msmeissn, 2021-10-06:
very good, thx for feedback!

closed 2021-10-06 by msmeissn