gphoto/libgphoto2 #717: libgphoto2+gphoto2 builds fail looking for gphoto2/gphoto2-foo.h include file

ndim, 2021-09-09:
**Describe the bug**

When there are no gphoto2/*.h include files in the C preprocessor's default include path, building gphoto2 fails trying to `#include` a `gphoto2/gphoto2-foo.h` include file.

```
In file included from actions.c:45:
./actions.h:24:10: fatal error: 'gphoto2/gphoto2-camera.h' file not found
#include <gphoto2/gphoto2-camera.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
```

The reason is that libgphoto2.pc and libgphoto2_port.pc both contain

    Cflags: -I${includedir}/gphoto2

but the gphoto2 source code uses

    #include <gphoto2/gphoto2-camera.h>

which we have declared the standard way of including libgphoto2 headers.

We have removed the symlink which used to link `${includedir}/gphoto2/gphoto2` to `${includedir}/gphoto2` in https://github.com/gphoto/libgphoto2/pull/452 after https://github.com/gphoto/libgphoto2/issues/444, and that symlink used to hack this to work.

**libgphoto2 and gphoto2 version**

**To Reproduce**

On a system which does not have any gphoto2 headers installed in the C preprocessor path,

```
cd
cd libgphoto2
./configure --prefix=$HOME/.local
make all
make install

cd
cd gphoto2
./configure --prefix=$HOME/.local PKG_CONFIG_PATH=$HOME/.local/lib/pkgconfig
make all
```

ndim, 2021-09-09:
The "proper" way to fix this would be to change the `libgphoto2.pc` and `libgphoto2_port.pc` files to both read

    Cflags: -I${includedir}

instead of

    Cflags: -I${includedir}/gphoto2

msmeissn, 2021-09-15:
hmm. kamera uses <gphoto2.h>  , needs adaption then.
digikam has also a single <gphoto2.h> include I think.

our tooling (gphoto2, gtkam) uses gphoto2/gphoto2.h already.
python bindings seem to actually work around the above already.
darktable uses gphoto2/gphoto2.h already
wine is ok
... cant check the external tools ...

but i think the change is okay, some packages might need to adjust

ndim, 2021-09-15:
Or we change to `Cflags: -I${includedir}/gphoto-2.5` in both `libgphoto2_port.pc` and `libgphoto2.pc`, but change the installation location of the header files from `/usr/include/gphoto2/gphoto2-camera.h` to `/usr/include/gphoto-2.5/gphoto2/gphoto2-camera.h`, similar to `/usr/include/gtk-3.0/gtk/gtkwindow.h`.

ndim, 2022-01-21:
Or we can have both, basically going back to the `/usr/include/gphoto2` directory days with the `ln -s . gphoto2` symlink.

Anyway, I would really like to be able to build `gphoto2` against a `libgphoto2` I have just installed and fix this.

closed 2022-01-23 by ndim