gphoto/libgphoto2 #718: Canon PowerShot SX740 HS error setting zoom

gabrielesartor, 2021-09-15:
**Describe the bug**
Using Canon PowerShot SX740 HS I am not able to set the zoom using gphoto2 with the terminal (Linux, Raspian OS).

Below is the info about the camera and the error.

Model                          Port                                            
----------------------------------------------------------
Canon PowerShot SX740 HS       usb:001,003  

**libgphoto2 and gphoto2 version**
gphoto2         2.5.27.1       gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.27.1       standard camlibs (SKIPPING lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

**To Reproduce**
gphoto2 --set-config zoom=50
                                                                               
*** Error ***              

The property 'Zoom' / 0xd055 was not set (0x2019: PTP Device Busy).

*** Error ***              
Failed to set new configuration value 50 for configuration entry zoom.
*** Error (-110: 'I/O in progress') ***       

**Debug flag file**
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/7170361/my-logfile.txt)

**Further info**
I noticed that, using the command "gphoto2 --list-all-config", the zoom is of the type RADIO, could it be the problem?

/main/capturesettings/zoom
Label: Zoom
Readonly: 0
Type: RADIO
Current: 0
Choice: 0 127
END

Thank you in advance!

msmeissn, 2021-09-24:
i put in a change that you can send values between 0 and 1000 or so... can you try curent libgpoto2 git and send values between 0 and 255 or so?
cant test myself sadly, as my sx720hs does not do that

gabrielesartor, 2021-09-24:
Running _gphoto2 --set-config zoom=50 --debug --debug-logfile=debug.log_ the problem is still there. Also adding _--capture-image_ the problem remains.
[debug.log](https://github.com/gphoto/libgphoto2/files/7224996/debug.log)

However, it seems it works correctly using the Python bindings python-gphoto2! :)

msmeissn, 2021-09-24:
hmm, can you try

gphoto2 --capture-preview --set-config zoom=50 --capture-preview 
or so and see if that works?
(it might need liveview enabled)

gabrielesartor, 2021-09-28:
Unfortunately, It still doesn't work using _gphoto2 --capture-preview --set-config zoom=50 --capture-preview --debug --debug-logfile=debug.log._

*** Error ***              
The property 'Zoom' / 0xd055 was not set (0x2019: PTP Device Busy).

*** Error ***              
Failed to set new configuration value 50 for configuration entry zoom.
*** Error (-110: 'I/O in progress') ***

[debug.log](https://github.com/gphoto/libgphoto2/files/7242232/debug.log)

bhvi150, 2023-01-25:
I am working on altering Zoom when USB tethering of canon G7 max mark iii and this doesn't work for g7mark iii as well. I get the exact same error as @gabrielesartor following the steps he is doing.

Please let me know how to set 0xd055

blackmole, 2025-08-09:
I've got the same issue with the Canon PowerShot SX70 HS. Is there any solution or workaround for this issue?