gphoto/libgphoto2 #720: Nikon Z6 Programmode (A,M,S,P,...) not changeable....(?)

sepl1977, 2021-09-21:
Hi!
i am a great fan of your work! Using gphoto for a long time with a lot of nikon cameras (DSLRs) and your software really appears always to me rock-solid and reliable! Before i worked with the Z6 i never had a problem with changing the Programmode via gphoto (it worked with the D200, D300 and D700) but now (on the Z6) it is a read-only parameter and i cant change it any more...
I guess this is a camea-issue and has nothing to do with your software -but if there is a chance to get this working i will be very thankful! Maby there is a trick to achieve this (ie set program-dial to U2 and activate setting X bla)? i read something about the "controlmode" setting -but after setting this parameter to "1" (set-config) the value stays on 0 when i readout the parameter after setting it(get-config)... -i found an other parameter called "applicationmode" but this had no effect too (when i set it to 1)
I use the latest Nikon Firmware: v3.31
and the latest libgphoto2: v2.5.27 (compiled)
OS is debian 10 (buster)

big thx in davance!

best regards!

sepl

msmeissn, 2021-10-02:
gphoto2 --set-config controlmode=1 --set-config expprogram=S  ... more things

however as soon as gphoto2 session ends the camera set value will be restored

sepl1977, 2021-10-04:
it works!!!! thank you so much!
and thx for careing for this project for so long!
best regards!
sepl

msmeissn, 2021-10-04:
you are welcome

closed 2021-10-04 by msmeissn