gphoto/libgphoto2 #721: SIGSEGV after capturing 20+ images for Sony A7

knro, 2021-09-24:
**Describe the bug**

Crash when capturing many images with Sony A7. It appears to sigfault after capturing anywhere between 20-80 images. Using latest libgphoto2 from Github at the time of this bug. Please note this crash is new, but I wasn't able to bisect the issue. All I know is that it didn't occur in previous versions of libgphoto2.

**Name the camera**
Sony Alpha-A7 (Control)        usb:001,022 

lsusb

`Bus 001 Device 022: ID 054c:094c Sony Corp. `

**libgphoto2 and gphoto2 version**
gphoto2 2.5.20

Copyright (c) 2000-2018 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.20         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.27.1       standard camlibs (SKIPPING docupen lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

**To Reproduce**
I used this step to reproduce the behavior, it crashes randomly between 20-80 images.

```
#!/bin/bash

for i in {1..100}
do
    rm *.arw
    echo "Capturing image #$i"

    gphoto2 --debug --debug-logfile=gphoto.txt --capture-image-and-download || { echo "gphoto failed"; exit 1; }
done

```

Attached are:
1. Debug log [debug_gphoto.txt](https://github.com/gphoto/libgphoto2/files/7227728/debug_gphoto.txt)
2. Backtrace [sony_crash.txt](https://github.com/gphoto/libgphoto2/files/7227727/sony_crash.txt)

msmeissn, 2021-09-25:
memory corruption, not good. likely caused by wait_for_event reporting more data now.

can you check if e.g. eef6ef81399ed9b06ea8132ab6e02b04403e97e1 is still good?

I will add more error handling to git, but the crash itself is already caused by earlier memory corruption i think.

is there any possibility to run this within valgrind?

knro, 2021-09-27:
Yes tested and that version is good, no crashes after 100 captures in a row.

msmeissn, 2021-09-27:
ok, first known bad might be 81cca23257aca3b25a338d31630a462c74fadb93 if you want to test that

msmeissn, 2021-10-02:
i found a bug that might have caused this and fixed it

msmeissn, 2021-10-03:
(would be good if you could verify ;)

knro, 2021-10-11:
So I don't have this camera myself and I asked the user to test again and he said it was crashing on master. But I'll ask him to participate in this discussion directly better.

glundby, 2021-11-11:
SIGSEGV after capturing still present in newest updated GPHOTO2!
Tested on Sony a5000 two different and on a7. Craches INDI then Kstars - Ekos!
Had to go back to last good version patched by KNRO on my system while testing the problem. With this patch it runs perfect!
Please fix so I can allow my system to upgrade!

msmeissn, 2021-11-12:
can you get a backtrace or valgrind trace? hm.. i need to rtest it again here

msmeissn, 2021-12-29:
i was not able to reproduce with a sony slt a58v at current git level ( i did various fixes inbetween)

knro, 2022-02-03:
We have more users reporting this crash with Sony, will try to get more logs.