gphoto/libgphoto2 #722: Nikon 1 mirrorless support

vidrazor, 2021-09-30:
**Describe the bug**
Camera cannot be controlled.
A clear and concise description of what the bug is.
Using the the Nikon driver in Ekos, camera is recognized, but can only live view, cannot take pictures
**Name the camera**
Nikon 1 V3 and Nikon 1 J4
Camera name as shown by gphoto2 --auto-detect or USB IDs
In Ekos, it just lists as Nikon DSLR, generic listing for all Nikons.
**libgphoto2 and gphoto2 version**
Sorry, don't have specific number, but most recent as per Astroberry update 9/21
output of: gphoto2 --version
I don't understand this question.
**To Reproduce**
Steps to reproduce the behavior:
Using Ekos in Astroberry, camera capture control does not work.
When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.
Sorry, I'm not a programmer, was sent to this site from the http://www.gphoto.org/doc/remote/ page, I hope my information is adequate. Would love to see the Nikon 1 V3 and J4 fully controllable from Ekos.
Thanks for your time.

ndim, 2021-10-01:
To save everybody on the gphoto side the research about what "Ekos" and "Astroberry" are: It looks as if "Astroberry" is a Linux distro for Raspberry Pi for people doing astronomy and astrophotography.

It appears "Ekos" means https://www.indilib.org/about/ekos.html, and the kstars that page mentions is probably https://edu.kde.org/kstars/. As to where libgphoto2 comes into play... apparently there is a `libgphoto2` interface `indi-gphoto` available at https://github.com/indilib/indi-3rdparty/tree/master/indi-gphoto but only bundled together with amongst other things nonfree blobs in the https://github.com/indilib/indi-3rdparty/ repository.

I could find Fedora packages `kstars` and `indi-gphoto`, but have not found anything in Fedora about `ekos`. I have not looked into the package situation on Raspberry OS on which I presume Astroberry is based on.

ndim, 2021-10-01:
@vidrazor It is very helpful for us to know which version exactly of our software you are running.

For a first bunch of information, you could run `apt list 'libgphoto2*' gphoto2`. This probably looks something like the following:

```
user@host:~$ apt list 'libgphoto2*' gphoto2
Listing... Done
gphoto2/stable 2.5.20-3 amd64
libgphoto2-6/stable,now 2.5.22-3 amd64 [installed,automatic]
libgphoto2-dev-doc/stable 2.5.22-3 all
libgphoto2-dev/stable 2.5.22-3 amd64
libgphoto2-l10n/stable,now 2.5.22-3 all [installed,automatic]
libgphoto2-port12/stable,now 2.5.22-3 amd64 [installed,automatic]
user@host:~$ _
```

We are interested in the version numbers behind `gphoto2/stable` and `libgphoto2-6/stable`. I presume you have only the `libgphoto2-6`, `libgphoto2-port12`, and `libgphoto2-l10n` packages installed, and probably are missing the `gphoto2` package.

If you could install the `gphoto2` package and post the output of `gphoto2 --version`, we will know even more details about the version of libgphoto2 which you are running, however, that might not strictly be necessary yet at this point in time.

```
user@host:~$ sudo apt install gphoto2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libcdk5nc6
Suggested packages:
  gthumb
The following NEW packages will be installed:
  gphoto2 libcdk5nc6
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 320 kB of archives.
After this operation, 1352 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
[... deleted ...]
user@host:~$ gphoto2 --version
gphoto2 2.5.20

Copyright (c) 2000-2018 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.20         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking
user@host:~$ _
```

ndim, 2021-10-01:
Just BTW, judging from a look into `indi-3rdparty/indi-gphoto/gphoto_driver.cpp`, it should be possible to switch `indi-gphoto` into a debug mode to have it log the `libgphoto2` version information through the indi-gphoto debug output:

```
1556     DEBUGDEVICE(device, INDI::Logger::DBG_DEBUG, "libgphoto2 info:");
1557     const char **libgphotoVerionInfo = gp_library_version(GP_VERSION_SHORT);
1558     for (const char **verionInfo = libgphotoVerionInfo; *verionInfo; verionInfo++)
1559     {                                              
1560         const char *pInfo = *verionInfo;           
1561         DEBUGFDEVICE(device, INDI::Logger::DBG_DEBUG, "%s", pInfo);
1562     }                                              
```

Whether that debug mode is easy to set up, and whether the debug output is easy to access or not, I do not know.

msmeissn, 2021-10-01:
in general Nikon 1 have been supported for a while, and this should be available also in libgphoto2 2.5.22 already.
good to see preview works, but also regular capture should work, and at lerast does with my nikon 1 j3/j5

Perhaps the cameras do not have the full coverage needed by the astro tooling.-.. more debug would be helpful

vidrazor, 2021-10-01:
Thanks for your replies. My apologies, as I mentioned I'm not a programmer, I was merely vectored to your site via the http://www.gphoto.org/doc/remote/ link. 
Astroberry is a prepackaged astronomy ISO for the Raspberry Pi, which runs on Raspbian, a Debian derivative. It's merely a Raspbian ISO with a series of astronomy related apps preinstalled. 

My connection with all this is merely as an end-user of the Astroberry to do astrophotography, specifically with that of the app KStars, which has a sub app embedded Ekos, which is a one-stop astrophotography and observatory control center, which works in conjunction with an "INDI server", which is to Linux what ASCOM is to Windows.
https://indilib.org/what-is-indi.html

In the Ekos sub-app is where you set up the hardware you will use to for your astrophography session, and this is where you choose, among other things, your camera. In the camera choices are generic brand names (such as Nikon), as well as a generic GPhoto CCD driver.

Apparently the problem is Ekos related, as I just installed an app called ENTANGLE, and it appears to have no problem controlling the Nikon 1 V3. So I suppose this isn't really your problem LOL.
 
But for completeness sake, I tried your suggestion. Being as I was brought into the Linux world though the Astroberry system for the purposes of astrophotography, I'm not really verse on Linux as well. However I ran the command you requested in the terminal and got the data listed below:

astroberry@astroberry:~ $ apt list 'libgphoto2*' gphoto2
Listing... Done
gphoto2/oldstable 2.5.20-3 armhf
libgphoto2-6-dbgsym/unknown 2.5.27 armhf
libgphoto2-6/unknown,now 2.5.27 armhf [installed,automatic]
libgphoto2-dev-doc/unknown 2.5.27 all
libgphoto2-dev/unknown 2.5.27 armhf
libgphoto2-l10n/unknown 2.5.27 all
libgphoto2-port12-dbgsym/unknown 2.5.27 armhf
libgphoto2-port12/unknown,now 2.5.27 armhf [installed,automatic]
astroberry@astroberry:~ $ 

As I said, being as ENTANGLE has no problem accessing the camera, this isn't really your issue. I never did get a solution from the INDI/Astroberry group on this however, which has forced me to use a Windows setup for the Nikon 1 via ASCOM, which is it's own can of worms. :-)

Thanks again for your time.

msmeissn, 2021-10-03:
lets close on our side then

closed 2021-10-03 by msmeissn