gphoto/libgphoto2 #723: Should the Python/Raspbian '-dev' wheel(?) be updating? Still 2.5.22

greiginsydney, 2021-10-01:
**Describe the bug**

I'm installing 'libgphoto-dev' on a Raspberry Pi as required by [python-gphoto2](https://github.com/jim-easterbrook/python-gphoto2) using `sudo apt-get install libgphoto2-dev`.

This installs 2.5.22, however it looks to me like the latest '-dev' build is 2.5.27, and as a result I'm missing support for some of the cameras added in the ensuing releases (e.g. the Canon 250D).

Is this still the best way to get the latest -dev release on a Pi, should I be doing something differently, or is the -dev release just being held back for some reason?

**Name the camera**
Canon 250D

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version

I've only installed 'libgphoto-dev' in this example build, and in /lib/arm-linux-gnueabihf/libgphoto2-dev/bin if I `cat gphoto2-config` it includes this:

```
--version)
        echo "libgphoto2" "2.5.22"
```

**To Reproduce**
Steps to reproduce the behavior:

`sudo apt-get install libgphoto2-dev`

&nbsp;<br>
Thanks.

msmeissn, 2021-10-01:
this is nothing we are responsible for ... can you tell the maitnainers of these repos?

hfiguiere, 2021-10-01:
The problem is that Raspian packages are frozen in the past, and these are the one you get with `apt` And that's a Raspian (RaspberryPi OS) problem.

ndim, 2021-10-01:
As Raspberry Pi OS (Raspian) is a Debian derivative, here is my understanding based on mainly some experience with Debian from 15  to 20 years ago on my daily use desktop system, and an occasional VM or Raspberry in the last 5 years.

Debian has basically four tiers: unstable, testing, stable, and oldstable (with increasing age). Debian stable (could also be called stale in many ways) is software which is a few years old, which is good when you want stability, but bad if you want to benefit from the last few years of work on a specific piece of software.

You can update your system to more current tier of Debian like testing (the oldest I would try for desktops) or unstable (which might occasionally break somewhere). However, what Debian is (or at least used to be) very good at was mixing packages from different tiers.

So if you are interested in more current gphoto2 and libgphoto2 on your stale or testing system, pick the gphoto2 and libgphoto2 packages from testing or unstable, which might pull in some other packages from testing or unstable, on a system which is otherwise completely stable or testing. Don't ask me how exactly to do that... I did that more that 15 years ago.

greiginsydney, 2021-10-03:
Thanks for enlightening me. I didn't realise these builds were created and managed by the Raspbian folk and not @msmeissn directly through this repo.

Forcing "N-5" and not offering a more recent build stream seems an extreme approach to the release tiering, but I guess that's their decision.

As a consequence I've decided to abandon the `apt` install process and manage my own builds.

\- G.

closed 2021-10-03 by greiginsydney

msmeissn, 2021-10-04:
I think they just do not care / do not have resources to update their packages :(