gphoto/libgphoto2 #724: Can we require C99 for building libgphoto2?

ndim, 2021-10-03:
**Is your feature request related to a problem? Please describe.**

The [HACKING.md](https://github.com/gphoto/libgphoto2/blob/master/HACKING.md) file says libgphoto2 code is supposed to compile with a C89 compiler.

However, since 2014-07, some parts of the libgphoto2 repo only compile with support for variadic macros which are a language feature not available in C90. Variadic macros have only been introduced in C99. Note that even the libgphoto2 fallback code uses variadic macros there:

https://github.com/gphoto/libgphoto2/blob/e2c0497dbe6059e888682923495d2333ef6ef655/libgphoto2_port/gphoto2/gphoto2-port-log.h#L137

```
#define GP_LOG_D(...) /* no-op */
#define GP_LOG_E(...) /* no-op */
#define GP_LOG_DATA(DATA, SIZE, ...) /* no-op */
```

Given this has apparently not been an issue since mid 2014, can we require C99 now for building libgphoto2 itself?

This would give us (amongst other things)
  * variadic macros for the debug messages (no need for an `#else` branch when defining variadic macros)
  * inline definition of for loop iteration variables (keep iteration variables local)
  * stdint.h (provided by the compiler)

**Describe the solution you'd like**
Require a C99 compiler to build libgphoto2 itself, but make the libgphoto2 public header files in such a way that they can also be included from a compilation unit which is compiled in C90 mode.

**Describe alternatives you've considered**
Continue carrying around unused and untested branches which nobody has been using for the last 7 years and probably will not use in the future.

**Additional context**
If we change this, all the repos of the https://github.com/gphoto/ project which use the C language should follow.

hfiguiere, 2021-10-03:
I'd say yes. Even more since it's already implicit.

msmeissn, 2021-10-04:
i would be okay with it too

closed 2021-10-31 by ndim