gphoto/libgphoto2 #741: Trigger command accepts any value for port and can trigger incorrect camera more than one camera is connected

bkapadia01, 2021-11-09:
**Describe the bug**
The trigger and capture command seems to accept any value for the `-port` portion of the script which is causing the camera to be triggered when an invalid port value is provided. This is causing an issue when we have more than one camera mounted to the system where the incorrect camera is being triggered when it should not be. 

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs
Canon EOS 5DS R 

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version
libgphoto2      2.5.27     

**To Reproduce**
Steps to reproduce the behavior:
The usb port when running gphoto2 --auto-detect for a camera is: 
`Canon EOS 5DS R                usb:020,013 `

Running command with invalid port `-port x`, note the camera should not be triggered.

`gphoto2 --port x --filename TOP_%H-%M-%S.%C --set-config eosremoterelease="Immediate" --set-config eosremoterelease="Release Full" --wait-event-and-download="FILEADDED"`

When reproducing with the command line tool, attach debug out using  --debug --debug-logfile=debug.log   if considered useful

> The port you specified ('?mh??') can not be found. Please specify one of the ports found by 'gphoto2 --list-ports' and make sure the spelling is correct (i.e. with prefix 'serial:' or 'usb:').Loading camera driveWaiting for 'FILEADDED' event from camera. Press Ctrl-C to abort.              
UNKNOWN PTP Property d105 changed
UNKNOWN PTP Property d108 changed
UNKNOWN PTP Property d106 changed
UNKNOWN PTP Property d107 changed
UNKNOWN PTP Property d109 changed
UNKNOWN PTP Property d10a changed
UNKNOWN PTP Property d10b changed
UNKNOWN PTP Property d10c changed
UNKNOWN PTP Property d10d changed
UNKNOWN PTP Property d10e changed
UNKNOWN PTP Property d10f changed
UNKNOWN PTP Property d11b changed
UNKNOWN PTP Property d114 changed
UNKNOWN PTP Property d116 changed
UNKNOWN PTP Property d119 changed
UNKNOWN PTP Property d110 changed
UNKNOWN PTP Property d101 changed
UNKNOWN PTP Property d102 changed
UNKNOWN PTP Property d103 changed
UNKNOWN PTP Property d104 changed
UNKNOWN PTP Property d11d changed
UNKNOWN PTP Property d111 changed
UNKNOWN PTP Property d112 changed
UNKNOWN PTP Property d113 changed
UNKNOWN PTP Property d17c changed
UNKNOWN PTP Property d17d changed
UNKNOWN PTP Property d17e changed
UNKNOWN PTP Property d120 changed
UNKNOWN PTP Property d121 changed
UNKNOWN PTP Property d122 changed
UNKNOWN PTP Property d15d changed
UNKNOWN PTP Property d157 changed
UNKNOWN PTP Property d158 changed
UNKNOWN PTP Property d159 changed
UNKNOWN PTP Property d15a changed
UNKNOWN PTP Property d15b changed
UNKNOWN PTP Property d15c changed
UNKNOWN PTP Property d15e changed
UNKNOWN PTP Property d163 changed
UNKNOWN PTP Property d164 changed
UNKNOWN PTP Property d165 changed
UNKNOWN PTP Property d11c changed
UNKNOWN PTP Property d1a0 changed
UNKNOWN PTP Property d1a0 changed
UNKNOWN PTP Property d1a0 changed
UNKNOWN PTP Property d1a1 changed
UNKNOWN PTP Property d1a8 changed
[... 94 lines cut ...]
UNKNOWN OLCInfo exposure indicator 1,2,-9.3 (00000000)
UNKNOWN OLCInfo event 0x0080 content 00020000
UNKNOWN Focus Mask 00010000000008
UNKNOWN OLCInfo event mask=2c1
UNKNOWN PTP Property d1c3 changed
UNKNOWN PTP Property d1bf changed
UNKNOWN Button 4
UNKNOWN OLCInfo event 0x0080 content 01000000
UNKNOWN Focus Mask 00000000000800
UNKNOWN OLCInfo event mask=281
UNKNOWN Camera Status 1
UNKNOWN PTP Property d11b changed
UNKNOWN PTP Property d1c3 changed
UNKNOWN Button 3
UNKNOWN Focus Mask 01000000000800
UNKNOWN OLCInfo event mask=201
UNKNOWN FocusInfo eosversion=8,size=8688x5792,size2=8688x5792,points={{-435,652,283,283},{-966,652,283,283},{-1425,652,283,283},{-1860,652,283,283},{-2295,652,283,283},{435,978,283,283},{0,978,283,283},{-435,978,283,283},{-1860,326,283,283},{-2295,326,283,283},{2295,652,283,283},{1860,652,283,283},{1425,652,283,283},{966,652,283,283},{435,652,283,283},{0,652,283,283},{1860,326,283,283},{1425,326,283,283},{966,326,283,283},{435,326,283,283},{0,326,283,283},{-435,326,283,283},{-966,326,283,283},{-1425,326,283,283},{435,0,283,283},{0,0,283,283},{-435,0,283,283},{-966,0,283,283},{-1425,0,283,283},{-1860,0,283,283},{-2295,0,283,283},{2295,326,283,283},{-966,-326,283,283},{-1425,-326,283,283},{-1860,-326,283,283},{-2295,-326,283,283},{2295,0,283,283},{1860,0,283,283},{1425,0,283,283},{966,0,283,283},{-2295,-652,283,283},{2295,-326,283,283},{1860,-326,283,283},{1425,-326,283,283},{966,-326,283,283},{435,-326,283,283},{0,-326,283,283},{-435,-326,283,283},{1425,-652,283,283},{966,-652,283,283},{435,-652,283,283},{0,-652,283,283},{-435,-652,283,283},{-966,-652,283,283},{-1425,-652,283,283},{-1860,-652,283,283},{435,-978,283,283},{0,-978,283,283},{-435,-978,283,283},{2295,-652,283,283},{1860,-652,283,283}},select={},unknown={00001007004000000000000000000000ffff}
Saving file as TOP_21-00-36.cr2
event found, stopping wait!
t2timac1:Arduino genuser1$ gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Canon EOS 5DS R                usb:020,013     
t2timac1:Arduino genuser1$ gphoto2 -v
gphoto2 2.5.27

vepicurean, 2021-11-29:
Does anyone have any suggestions on how to fix this issue? 

We spent a long time investigated what we are observing, thinking it was something we were doing wrong, but we realized what was causing the wrong cameras to be triggered we attributed to how the gphoto2 trigger commend works with the USB port parameter.

msmeissn, 2021-11-29:
i added some more error code handling, especially for fully invalid ports.

msmeissn, 2021-11-29:
however --port usb:xxx,yyy if the combi is invalid, will still fallback to the first valid port

msmeissn, 2021-11-29:
ok, some more code added ..

--port usb:002,007   now fails if there is only usb:002,008

lets hope i did not break any usecases relying on this lax detection ;)

vepicurean, 2021-12-03:
Understand.  Thanks. I'll try this fix on our development Mac and let you know how it goes.