gphoto/libgphoto2 #744: On Canon EOS 4D IV capture-image stops after 17 shots

mcapitanio, 2021-11-24:
**Describe the bug**

If I execute 2 times this command:

gphoto2 --set-config capturetarget=1 --set-config aperture=2.8 --wait-event 250ms --set-config iso=100 --wait-event 250ms --set-config shutterspeed=2 --capture-image --wait-event 250ms --set-config aperture=4 --wait-event 250ms --set-config shutterspeed=2 --wait-event 250ms --capture-image --wait-event 250ms --set-config aperture=5.6 --wait-event 250ms --set-config shutterspeed=2 --wait-event 250ms --capture-image --wait-event 250ms --set-config aperture=8 --wait-event 250ms --set-config shutterspeed=2 --wait-event 250ms --capture-image --wait-event 250ms --set-config aperture=11 --wait-event 250ms --set-config shutterspeed=2 --wait-event 250ms --capture-image --wait-event 250ms --set-config aperture=16 --wait-event 250ms --set-config shutterspeed=2 --wait-event 250ms --capture-image --wait-event 250ms --set-config aperture=22 --wait-event 250ms --set-config shutterspeed=2 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=1 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=0.5 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=1/4 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=1/8 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=1/15 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=1/30 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=1/60 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=1/125 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=1/250 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=1/500 --wait-event 250ms --capture-image --wait-event 250ms

the first time it takes 17 photos, the second one it does not capture any image.

If I turn off the camera I see the message "Registering.... Image left: 17". I have to remove the USB cable to see the counter going to 0 and shutting down the camera.

**Name the camera**
Canon EOS 5D Mark IV

**libgphoto2 and gphoto2 version**

gphoto2 2.5.27

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.27         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.27.1       standard camlibs, gcc, no ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking

**To Reproduce**

Execute 2 times the command:

gphoto2 --debug --debug-logfile=debug.log --set-config capturetarget=1 --set-config aperture=2.8 --wait-event 250ms --set-config iso=100 --wait-event 250ms --set-config shutterspeed=2 --capture-image --wait-event 250ms --set-config aperture=4 --wait-event 250ms --set-config shutterspeed=2 --wait-event 250ms --capture-image --wait-event 250ms --set-config aperture=5.6 --wait-event 250ms --set-config shutterspeed=2 --wait-event 250ms --capture-image --wait-event 250ms --set-config aperture=8 --wait-event 250ms --set-config shutterspeed=2 --wait-event 250ms --capture-image --wait-event 250ms --set-config aperture=11 --wait-event 250ms --set-config shutterspeed=2 --wait-event 250ms --capture-image --wait-event 250ms --set-config aperture=16 --wait-event 250ms --set-config shutterspeed=2 --wait-event 250ms --capture-image --wait-event 250ms --set-config aperture=22 --wait-event 250ms --set-config shutterspeed=2 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=1 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=0.5 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=1/4 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=1/8 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=1/15 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=1/30 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=1/60 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=1/125 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=1/250 --wait-event 250ms --capture-image --wait-event 250ms --set-config shutterspeed=1/500 --wait-event 250ms --capture-image --wait-event 250ms

I attach 2 logs:

[debug-1.log](https://github.com/gphoto/libgphoto2/files/7594171/debug-1.log): the one executing the command the first time
[debug-2.log](https://github.com/gphoto/libgphoto2/files/7594175/debug-2.log): the one executing the command the second time

msmeissn, 2021-11-24:
This message usually means we do not retrieve all images from the SDRAM of the camera.

I see unhandled 0xc1ab events, which is a FTP based transfer (?) ... what kind of image transfer mode do you have enabled?

mcapitanio, 2021-11-24:
How can I check the transfer mode I have enabled?

mcapitanio, 2021-11-24:
Ok, maybe I have understood. On camera settings I had the "Auto" setting for Transfer mode.
I have disabled "Auto" transfer mode and the problem disappeared.

msmeissn, 2021-11-24:
this is a default setting right? then it might be 1 method we do not support yet

mcapitanio, 2021-11-24:
Yes, I think that is a default setting

msmeissn, 2021-11-24:
i think i need to handle these new requests, i might need to ask you for some testing in some days (not sure when I get to it :( )

but it is good that you have a workaround

mcapitanio, 2021-11-24:
Yes, sure, consider me as a tester, no problem. Let me know when and how.