gphoto/libgphoto2 #749: Sony A7R4A not recognised as Sony camera

atebay, 2021-12-08:
Fill in fields below. If you don't know how, leave it free.

**Name of the camera*
Sony A7RM4A

Use the Marketing name.

**USB IDs**
e.g. by running:   lsusb 

[lsusb.txt](https://github.com/gphoto/libgphoto2/files/7674971/lsusb.txt)

**camera summary output**
run

  gphoto2 --summary > summary.txt

attach summary.txt here

[sony_a7rm4a.txt](https://github.com/gphoto/libgphoto2/files/7674973/sony_a7rm4a.txt)

**camera configuration output**

gphoto2 --list-all-config > list-all-config.txt

and attach list-all-config.txt here

[sony_a7rm4a_config.txt](https://github.com/gphoto/libgphoto2/files/7674976/sony_a7rm4a_config.txt)

**test capture**

test if capture perhaps already works:

Only works if 3s wait is added
  gphoto2 --wait-event=3s --capture-image-and-download

  gphoto2 --capture-image-and-download
 
  gphoto2 --capture-preview

msmeissn, 2021-12-08:
i have added it to current git for the next release, thanks for the report!

atebay, 2021-12-08:
That was quick.  I have tested the update with the camera on Ubuntu on PC and on a Raspberry PI4 and it is detected correctly and captures images. Many thanks,Andy TebaySent from Mail for Windows From: Marcus MeissnerSent: 08 December 2021 10:26To: gphoto/libgphoto2Cc: atebay; AuthorSubject: Re: [gphoto/libgphoto2] Sony A7R4A not recognised as Sony camera (Issue #749) i have added it to current git for the next release, thanks for the report!

closed 2021-12-28 by msmeissn

iliesaya, 2022-03-24:
Sorry to re-open this , I am running 
gphoto2         2.5.28.1
libgphoto2      2.5.29.1 
libgphoto2_port 0.12.0

and my Sony ILCE-7RM4A  is behaving the same as OP , 

$ gphoto2 --capture-image-and-download
ERROR: Could not capture image.                                                
ERROR: Could not capture.

$ gphoto2 --wait-event=3s --capture-image-and-download
Waiting for 3 seconds for events from camera. Press Ctrl-C to abort.           
UNKNOWN PTP Property 0000 changed, "PTP Property 0x0000" to "Unknown type 0x0000"
UNKNOWN PTP Property 0000 changed, "PTP Property 0x0000" to "Unknown type 0x0000"
UNKNOWN PTP Property 0000 changed, "PTP Property 0x0000" to "Unknown type 0x0000"
UNKNOWN PTP Property 0000 changed, "PTP Property 0x0000" to "Unknown type 0x0000"
New file is in location /capt0000.arw on the camera                            
Saving file as capt0000.arw
Deleting file /capt0000.arw on the camera

Did I miss something ? I do have your last modification in my file camlibs/ptp2/library.c: 
{"Sony:ILCE-7RM4A (PC Control)", 0x054c, 0x0d9f, PTP_CAP|PTP_CAP_PREVIEW},

thx

msmeissn, 2022-03-24:
can you capture a failure debug trace? we should be waiting 2.5 seconds already.

iliesaya, 2022-03-27:
Here is the debug trace : 

```
jetson@ubuntu:~/git/Jetson$ env LC_ALL=C gphoto2 --debug-logfile=logfile.log --debug --capture-image-and-download
ERROR: Could not capture image.                                                
ERROR: Could not capture.
```

https://raw.githubusercontent.com/iliesaya/TmpFileHostingForGithubIssue/main/Sony_ILCE-7RM4A_gphoto2_logfile.log

msmeissn, 2022-07-23:
hmm, i added this only in April 20th.... can you try 2.5.30 or current git again?