gphoto/libgphoto2 #751: Build failure: "Inconsistent values for GETTEXT_PACKAGE_LIBGPHOTO2_PORT and po/Makevars DOMAIN"

trilader, 2021-12-13:
**Describe the bug**

After cloning the git repository, running `autoreconf --install --symlink` as per `INSTALL`, creating an installation directory `inst`, a build directory `build` next to it, running configure froḿ inside it (`../configure --prefix=$(realpath ../inst) --with-libexif=no --with-camlibs=ptp2 --disable-serial --disable-disk`) and starting the build (`make`, with or without `-j6` doesn't make a difference.) it fails with the error message in the subject:
`Inconsistent values for GETTEXT_PACKAGE_LIBGPHOTO2_PORT and po/Makevars DOMAIN`

I'm building on Windows with MinGW and this procedure/combination of software version has worked successfully in the past.
Versions of tools:
- `gcc`: 9.2.0
- `gettext`: 0.19.8.1
- `automake`: 1.16.2
- `autoconf`: 2.69
- `libtool`: 2.4.6

ndim, 2021-12-14:
Obviously, this check should have been conditional on `USE_NLS` from the start.

However, there is also something else going wrong here which I would like to find out and handle correctly.

Fix/Debug code to follow.

ndim, 2021-12-16:
Could you try `git pull` and `make all-local && make -C libgphoto2_port all-local && make all-local`?

trilader, 2021-12-16:
I just had a go at it.

<details>
<summary>Output of <pre>make all-local && make -C libgphoto2_port all-local && make all-local</pre></summary>

```
$ make all-local && make -C libgphoto2_port all-local && make all-local
CDPATH="${ZSH_VERSION+.}:" && cd .. && /bin/sh /c/work/libgphoto2-git/auto-aux/missing aclocal-1.16 -I auto-m4 -I libgphoto2_port/gphoto-m4
 cd .. && /bin/sh /c/work/libgphoto2-git/auto-aux/missing automake-1.16 --foreign
CDPATH="${ZSH_VERSION+.}:" && cd .. && /bin/sh /c/work/libgphoto2-git/auto-aux/missing autoconf
/bin/sh ./config.status --recheck
running CONFIG_SHELL=/bin/sh /bin/sh ../configure --prefix=/c/work/libgphoto2-git/inst --with-libexif=no --with-camlibs=ptp2 --disable-serial --disable-disk PKG_CONFIG_PATH=/mingw64/lib/pkgconfig:/mingw64/share/pkgconfig --no-create --no-recursion
configure: loading site script /mingw64/etc/config.site
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for POSIX sh $() command substitution... yes
checking original CC value... ""
checking original CXX value... ""
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
checking how to run the C preprocessor... gcc -E
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking for a sed that does not truncate output... /usr/bin/sed
checking for ar... ar
checking the archiver (ar) interface... ar
checking for rsync... no
/* confdefs.h */
#define PACKAGE_NAME "libgphoto2 photo camera library"
#define PACKAGE_TARNAME "libgphoto2"
#define PACKAGE_VERSION "2.5.27.1"
#define PACKAGE_STRING "libgphoto2 photo camera library 2.5.27.1"
#define PACKAGE_BUGREPORT "gphoto-devel@lists.sourceforge.net"
#define PACKAGE_URL ""
#define PACKAGE "libgphoto2"
#define VERSION "2.5.27.1"
configure: saving original confdefs.h and creating empty confdefs.h
checking CFLAGS value before... -g -O2
checking CXXFLAGS value before... -g -O2
checking whether CFLAGS can append -std=c90... yes
checking whether CFLAGS can append -pedantic -Wall -Wextra -Werror... yes
checking whether to test pedantic -std=c90... yes
checking whether CFLAGS can append -std=c99... yes
checking whether CFLAGS can append -pedantic -Wall -Wextra -Werror... yes
checking whether to test pedantic -std=c99... yes
checking whether CFLAGS can append -std=c11... yes
checking whether CFLAGS can append -pedantic -Wall -Wextra -Werror... yes
checking whether to test pedantic -std=c11... yes
checking whether CFLAGS can append -std=c2x... yes
checking whether CFLAGS can append -pedantic -Wall -Wextra -Werror... yes
checking whether to test pedantic -std=c2x... yes
checking whether CXXFLAGS can append -std=c++98... yes
checking whether CXXFLAGS can append -pedantic -Wall -Wextra -Werror... yes
[... 626 lines cut ...]
config.status: creating libgphoto2_port/Makefile
config.status: creating libgphoto2_port.pc
config.status: creating libgphoto2_port-uninstalled.pc
config.status: creating gphoto2-port-config
config.status: creating tests/Makefile
config.status: creating doc/Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
config.status: executing libtool commands
config.status: executing po-directories commands
config.status: creating po/POTFILES
config.status: creating po/Makefile
  MAKEVARS_DOMAIN=libgphoto2_port-12
  GETTEXT_PACKAGE_LIBGPHOTO2_PORT=libgphoto2_port-12
Good: Matching gettext domain values (libgphoto2_port-12)
make: Leaving directory '/c/work/libgphoto2-git/build/libgphoto2_port'
  MAKEVARS_DOMAIN=libgphoto2-6
  GETTEXT_PACKAGE_LIBGPHOTO2=libgphoto2-6
Good: Matching gettext domain values (libgphoto2-6)
```

</details>

It also now builds without error for me.

ndim, 2021-12-17:
> ```
> $ make all-local && make -C libgphoto2_port all-local && make all-local
> [...]
>   MAKEVARS_DOMAIN=libgphoto2_port-12
>   GETTEXT_PACKAGE_LIBGPHOTO2_PORT=libgphoto2_port-12
> Good: Matching gettext domain values (libgphoto2_port-12)
> make: Leaving directory '/c/work/libgphoto2-git/build/libgphoto2_port'
>   MAKEVARS_DOMAIN=libgphoto2-6
>   GETTEXT_PACKAGE_LIBGPHOTO2=libgphoto2-6
> Good: Matching gettext domain values (libgphoto2-6)
> ```
> 
> It also now builds without error for me.

I have good news and bad news.

The good news: The problem you have reported appears to be fixed.

The bad news: My latest change was intended to just provide some basic debugging information. I do not know how this change could have actually fixed the problem.

Anyway, if something similar comes up again, the build output will now contain a bit more information to help with finding the actual issue, and we can deal with that if and when it actually happens.

Please reopen this issue if the
```
Error: Mismatching gettext domain values
```
problem reappears during the next few days.

Thank you very much for reporting and for testing!

closed 2021-12-17 by ndim