gphoto/libgphoto2 #755: Canon EOS M50: --capture-image takes forever because it enumerates all files and folders on the SD card

antocuni, 2021-12-31:
**Describe the bug**

if I try to capture an image, it takes much longer than expected. In the following example, the shutter speed was set to 1/160 and the capture itself was very quick on the camera, but it took 16 seconds to complete:

```
$ time gphoto2 --set-config capturetarget=1 --capture-image 
New file is in location /store_00020001/DCIM/100CANON/IMG_9500.CR3 on the camera
New file is in location /store_00020001/DCIM/100CANON/IMG_9500.JPG on the camera

real	0m16.294s
user	0m12.835s
sys	0m0.260s
```

I tried to run the same command with `--debug`, the `stderr` log is attached.
[Uploading stderr.zip…]()

By looking at the log, it seems to me that it takes such a long time because it tries to enumerate all files and folders on the SD card. For example, I see the following:

```
3.953990 get_folder_from_handle      (2): (20001,90000000,/store_00020001/)
3.954147 get_folder_from_handle      (2): (20001,0,/store_00020001/)
3.954301 gp_filesystem_append        (2): Append /store_00020001/DCIM/100CANON/IMG_9499.CR3 to filesystem
3.954410 lookup_folder               (2): Lookup folder '/store_00020001/DCIM/100CANON'...
3.954514 lookup_folder               (2): Folder / is dirty. Listing folders in there to make folder clean...
3.954742 gp_filesystem_list_folders  (2): Listing folders in /
3.954850 lookup_folder               (2): Lookup folder '/'...
3.954943 lookup_folder               (2): Found! / is 0xc08530
3.955028 gp_filesystem_list_folders  (2): ... is dirty, getting from camera
3.955122 folder_list_func            (2): folder_list_func(/)
3.955282 delete_all_folders          (2): Internally deleting all folders from '/'...
3.955387 lookup_folder               (2): Lookup folder '/'...
3.955481 lookup_folder               (2): Found! / is 0xc08530
3.955573 recurse_delete_folder       (2): Recurse delete folder 0xc08530//
3.955661 append_folder_one           (2): Append one folder store_00020001
3.955764 gp_filesystem_list_folders  (2): Folder / contains 1 subfolders.
3.955860 lookup_folder               (2): Done making folder / clean...
3.956000 lookup_folder               (2): Folder /store_00020001/ is dirty. Listing folders in there to make folder clean...
3.956109 gp_filesystem_list_folders  (2): Listing folders in /store_00020001/
3.956206 lookup_folder               (2): Lookup folder '/store_00020001/'...
3.956303 lookup_folder               (2): Found! /store_00020001/ is 0xc25a80
3.956390 gp_filesystem_list_folders  (2): ... is dirty, getting from camera
3.956483 folder_list_func            (2): folder_list_func(/store_00020001/)
3.956644 ptp                         (2): (storage=0x00020001, handle=0x00000000)
3.956758 folder_list_func            (2): after list folder (storage=0x00020001, handler=0x080)
3.956859 folder_list_func            (2): adding 0x90000000 / ob=0xc30068 to folder
3.957044 ptp_usb_sendreq             (2): Sending PTP_OC 0x1008 (Get object info) (0x919251b1) request...
3.957160 gp_port_write               (3): Writing 16 = 0x10 bytes to port...
3.957498 gp_port_write               (3): Wrote   16 = 0x10 bytes to port: (hexdump of 16 bytes)
```

after which there is a long list of `gp_port_read` and other messages which list of files on the SD card.

I also tried to delete all files on the SD card, and I confirm that after this the capture is much quicker:
```
$ time gphoto2 --set-config capturetarget=1 --capture-image 
New file is in location /store_00020001/DCIM/100CANON/IMG_9501.CR3 on the camera
New file is in location /store_00020001/DCIM/100CANON/IMG_9501.JPG on the camera

real	0m3.222s
user	0m0.160s
sys	0m0.152s
```

**Name the camera**
Canon EOS M50

**libgphoto2 and gphoto2 version**

```
$ gphoto2 --version
gphoto2 2.5.20

Copyright (c) 2000-2018 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.20         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.25.1       standard camlibs (SKIPPING lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.

antocuni, 2021-12-31:
attaching stderr [stderr.zip](https://github.com/gphoto/libgphoto2/files/7796703/stderr.zip)

msmeissn, 2022-01-01:
yes, due to PTP filesystem nature to find out the right directory / filename we need to read a lot of the image informations. I did "on demand" loading, but if all images are in 1 directory, we need to read all of them :/

i would recommend using SDRAM capture if you just download immediately anyway, or an empty sd card as you tried.

antocuni, 2022-01-01:
What I don't understand is why we need to find the directory/filename. Is it just to be able to print `New file is in location ...` ?
If so, what about adding an option to suppress it? For my specific use case knowing the filename is not necessary at all, because what I do is to shoot images continuously for 1 hour or so, saving them on the SD card.
If the idea sounds reasonable, I'd also be happy to try to make a PR