gphoto/libgphoto2 #756: Questions or corrections in "supported cameras" and "remote control" tables.

user0022, 2022-01-01:
Hello!
For my first post here, I just would like to say thanks for your work, and also I don't have computing knowledge and haven't yet used your software, so I hope I don't say mistakes.

**Cameras in "supported cameras" table missing in "remote control" table?**
As I watched these 2 tables, I saw many missing cameras in "Remote controlling cameras" http://www.gphoto.org/doc/remote/, their compatible state for being included in this table being yet writen in "libgphoto2 :: supported cameras" http://www.gphoto.org/proj/libgphoto2//support.php .
First example we can see is at the top of "supported cameras" table, most Agfa cameras are able of "Image Capture, Liveview, Configuration", but none is present in the "remote controling cameras" table.
And first right example is Canon Digital IXUS 30, "Image Capture, Liveview, Configuration" capable, and effectively present also in "remote controlling cameras".
Ctrl+F says 635 cameras in "supported cameras" table do at least "Image Capture", which is more than the ~200 in "remote control" table.

You say "The list is incomplete and might be incorrect.", but is it voluntary? Or something I didn't understand? Or maybe it is time missing with too much work (if it is the case, as it is non(technical I can add the 400 missing cameras for you in the table, and even try to fill in the Megapixel cells, I would put the html code in this page).
Too bad if someone came accross the "remote controlling" table, and wouldn't find his camera here, and would then give up a test or a purchasing because of this.
(I don't know if the same occurs in the opposite direction, it seems not but I checked only the 15 first cameras.)

**Misc. small errors seen in "remote controling" page.**
- "otherwise it will gone" (re-foundable with Ctrl+F) -> "go"? or "be gone"?
- "controling" spells "controlling" I think (myself making many spelling or english mistakes excuse me I'm french).

**And in table.**
- "Canon Digital IXUS 500" line is duplicated.
- Some lines don't seem in the right place (if you have ordered cameras alphabetically): Canon IXY Digital 300, Canon EOS 1D C,  Sony SLT-A58, Sony DSC RX10 M3, Sony DSC RX100 M5 (not checked everything). There are other lines that might be rearranged but it's maybe a bit complicated (after 50 comes 60 or 500?).
- "PowerShot A300" line have missing cells or something.
The Liveview cell seems to have disappeared. Fortunately I know a little bit html, so  a small edition in the source sould work, replacing
 ```
<tr>
<td>Canon PowerShot A300</td>
<td>Yes</td>
<td>All, Viewfinder</td>
<td>3.2</td>
<td>&nbsp;</td>
</tr>
```
with
 ```
<tr>
<td>Canon PowerShot A300</td>
<td>Yes</td>
<td>All, Viewfinder</td>
<td>&nbsp;</td>
<td>3.2</td>
<td>&nbsp;</td>
</tr>
```
- "Canon PowerShot A400" has inversion in Liveview and Megapixel cells text.

**More compact view for the tables?**
(If I dare make this suggestion! But well it's just an idea.)
These tables are a bit long with until 2777 cameras (fortunately, the more cameras the better), and should become yet longer with time. So their readability can be a bit hard, because of obligation of scrolling and scrolling.
Personnally it annoyed me so I copied them in a WYSIWYG page so I could edit them for more readability.
Modifications I made were mainly to avoid line breaks in "remote controlling" table (height was reduced of 30 or 50% I would say), because they can add a line you have to scroll, for sometimes only 1 character or 3 pixels.
So 2 things I tried to reduce width of a table line (and then thanks to it their height):
- Get cellspacing to 0. So instead of "| |" separation betweeen table cells, you only have a single "|", that avoids pixels leaks (vertically too) and (to me) is more beautiful and visually clearer.
- Remove brand names (Canon, Fuji...) in first column cells (I did this only on "remote control" table, "as "supported cameras" table e.g. has no line breaks and then is less needy. But if it was my site I should have the same presentation for all tables of course, for coherence).
I simply did it with Ctrl+H in a text editor, on the source code, replacing for the 8 current brands in this table:
```
<tr>
<td>Canon
```
(With the space after the name.)
with
```
<tr>
<td>
```
(Space removed too.)
That prevents many line breaks.
And put the brand name only in a new table line (with its cells merged to be clearer) that I added above concerned lines. So after line [Nikon_____] comes line [CoolPix 880__], and then line [CoolPix 2500_], etc.
Extra advantage can be to see more clearly cameras of a same brand (personnally I can appreciate it in my WYSIWYG).
Disadvantages could be 1/ that some readers can be a bit lost not knowing if the name they see is a Canon or a Nikon, and 2/ that typing precisely with the quotes operator (e. g. "Canon 20D") in a search engine can prevent to lead to this page.

Other possibilities (trained stuntmen only, don't do this at home!):
- Reduce font size or choose a more narrow font for the tables.
- Get cellpading to 0 (text sticks to the cell border, you can bare it or not).

msmeissn, 2022-01-01:
The gphoto.org/doc/remote page is manually edited and updated. Currently I do it when I get across entries, but as you see its not fully aligned with the libgpoto2 supported page, which is autogenerated.

I fixed your typo comments on the remote page now.

i need to see if I can adjust to your other recommendations.

msmeissn, 2022-01-01:
the support page does not close the table tag ... it needs a bit more fixup

user0022, 2022-01-04:
Thank you for the taking into consideration!
And well I understand better the differences between the two tables now, it surprised me that the "supported" table is automatically made (probably from another package or software then).

May I add yet that I saw the corrections that are OK, but 2 typos remaining:
- "otherwise it will gone" is now "otherwise it go", but you probably mean "otherwise it goes" or "otherwise it will go"...
- 2 "controling" are still with 1 l (it's in titles).

I'm afraid I don't really understand "the support page does not close the table tag", but well anyway I don't think I could help, my html is too poor...
Thank you.