gphoto/libgphoto2 #759: libgphoto-2.5.28 fails to configure with --with-jpeg

mattst88, 2022-01-09:
**Describe the bug**
Caused by commit https://github.com/gphoto/libgphoto2/commit/ea00220b6f44a00c2efa792c538f1804ebe183c2
````
./configure --with-jpeg` fails with: 
....
checking whether to build with libjpeg... configure: error: 
Unhandled value given to --with-jpeg / --without-jpeg: 'yes'
````

`configure` should continue accepting plain `--with-jpeg` or `--with-jpeg=yes`.

Cc: @ndim

ndim, 2022-01-10:
The behaviour from before commit ea00220b6f44a00c2efa792c538f1804ebe183c2 was

  * `--with-jpeg=no` or `--without-jpeg`

    Do not use libjpeg. Do not autodetect it either.

  * `--with-jpeg=<any value but "no">` or no jpeg related argument given at all

    Autodetect libjpeg. If found, use it.

I think we can safely assume that when someone runs `configure` with `--with-jpeg=yes` and libjpeg cannot be both found and used, continuing the build without using libjpeg is an outcome the user does not expect.

Therefore I would like to propose the following behaviour, where "autodetect" means the autodetect sequence introduced by commit ea00220 (`LIBJPEG_CFLAGS|LIBS` vars if defined, `pkg-config` `libjpeg.pc` if found, default location like before commit ea00220):

  * `--with-jpeg=no` or `--without-jpeg`

    Do not use libjpeg. Do not autodetect it either.

    This matches the behaviour before commit ea00220.

  * `--with-jpeg` or `--with-jpeg=yes`

    Autodetect libjpeg. If not found, abort.

    If libjpeg can be found and used, this matches the behaviour before commit ea00220.
    If libjpeg cannot be found or cannot be used, this differs from the behaviour before commit ea00220 (aborts now instead of silently ignoring the user's wish to build with libjpeg).

  * `--with-jpeg=auto` or `--with-jpeg=autodetect` or no jpeg related argument given

    Autodetect libjpeg. If found, use it.

    This matches the behaviour before commit ea00220.

  * Any other argument, e.g. `--with-jpeg=/path/to/somewhere`

    Abort the build.
    
    We allow the user to set LIBJPEG_CFLAGS and/or LIBJPEG_LIBS instead if they need to specify which libjpeg to compile with and link against.

    This differs from the behaviour before commit ea00220, where `configure` would just silently ignore the `--with-jpeg=ARG` argument (unexpected behaviour which should be fixed) and build with whatever header `#include <jpeglib.h>` happens to include and whatever library `-ljpeg` happens to link against, or not building with libjpeg if compiling or linking were not possible.

@mattst88 Are you OK with the proposed behaviour?

mattst88, 2022-01-10:
Yep, that sounds exactly like what I'd suggest. Thanks!

closed 2022-01-10 by ndim