gphoto/libgphoto2 #761: error: unknown warning group '-Woverride-init', ignored [-Werror,-Wunknown-warning-option]

ryandesign, 2022-01-14:
**Describe the bug**
libgphoto fails to build on macOS 10.13.6 with Xcode 9.4.1 or on macOS 10.14.6 with Xcode 10.3:

```
pentax/js0n.c:14:32: error: unknown warning group '-Woverride-init', ignored [-Werror,-Wunknown-warning-option]
#pragma GCC diagnostic ignored "-Woverride-init"
                               ^
```

Here are full build logs:

* https://build.macports.org/builders/ports-10.13_x86_64-builder/builds/137816/steps/install-port/logs/stdio
* https://build.macports.org/builders/ports-10.14_x86_64-builder/builds/114384/steps/install-port/logs/stdio

**Name the camera**
none

**libgphoto2 and gphoto2 version**
2.5.28

**To Reproduce**
compile libgphoto 2.5.28 on macOS 10.13.6 with Xcode 9.4.1 or on macOS 10.14.6 with Xcode 10.3

ndim, 2022-01-16:
I might need to build on a Mac to reproduce, examine, and fix this, so I need to coordinate with someone else for access to their Mac. That might take a few days.

ryandesign, 2022-01-16:
Note it's only certain versions of macOS (or probably it's caused by certain versions of clang) with which it fails.

On OS X 10.11.6 with Xcode 8.2.1 it builds fine.
On macOS 10.12.6 with Xcode 9.2 it fails as above.
On macOS 10.13.6 with Xcode 9.4.1 it fails as above.
On macOS 10.14.6 with Xcode 10.3 it fails as above.
On macOS 10.15.7 with Xcode 11.7 it builds fine.

ndim, 2022-01-16:
This looks useful to find out the clang/llvm versions: https://gist.github.com/yamaya/2924292. I have gathered the following information from there:

Xcode 8.2 -> Apple LLVM 8.0.0
Xcode 9.2 -> Apple LLVM 9.0.0
Xcode 9.4.1 -> Apple LLVM 9.1.0
Xcode 10.3 -> Apple LLVM 10.0.1
Xcode 11.7 -> Apple clang 11.0.3

This gives me an idea what compiler versions to try with a minimal working example on godbolt.org: clang 9 and clang 10.

ryandesign, 2022-01-16:
> This looks useful to find out the clang/llvm versions: https://gist.github.com/yamaya/2924292.

Or https://trac.macports.org/wiki/XcodeVersionInfo

> This gives me an idea what compiler versions to try with a minimal working example on godbolt.org: clang 9 and clang 10.

Understand that Apple Clang uses a different version numbering scheme than open source Clang.

ndim, 2022-01-16:
FTR, this is the failing compile command from one of the macports logs linked above:

 libtool: compile:  /usr/bin/clang -DHAVE_CONFIG_H -I. -I.. -D_GPHOTO2_INTERNAL_CODE -DLOCALEDIR=\"/opt/local/share/locale\" -DCAMLIBS=\"/opt/local/lib/libgphoto2/2.5.28\" -I.. -I.. -I../libgphoto2_port -I/opt/local/include -Wno-unused -pipe -Os -arch x86_64 -Werror=unknown-warning-option -Wall -Wextra -Weverything -Wno-error=documentation-deprecated-sync -Wno-unused-parameter -DLIBGPHOTO2 -DPKTDATADIR=\"/\" -pipe -Os -arch x86_64 -Werror=unknown-warning-option -Wall -Wextra -Weverything -Wno-error=documentation-deprecated-sync -Wno-unused-parameter -MT pentax/la-js0n.lo -MD -MP -MF pentax/.deps/la-js0n.Tpo -c pentax/js0n.c  -fno-common -DPIC -o pentax/.libs/la-js0n.o

And this is from the other one:

libtool: compile:  /usr/bin/clang -DHAVE_CONFIG_H -I. -I.. -D_GPHOTO2_INTERNAL_CODE -DLOCALEDIR=\"/opt/local/share/locale\" -DCAMLIBS=\"/opt/local/lib/libgphoto2/2.5.28\" -I.. -I.. -I../libgphoto2_port -I/opt/local/include -isysroot/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk -Wno-unused -pipe -Os -isysroot/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk -arch x86_64 -Werror=unknown-warning-option -Wall -Wextra -Weverything -Wno-error=documentation-deprecated-sync -Wno-unused-parameter -DLIBGPHOTO2 -DPKTDATADIR=\"/\" -pipe -Os -isysroot/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk -arch x86_64 -Werror=unknown-warning-option -Wall -Wextra -Weverything -Wno-error=documentation-deprecated-sync -Wno-unused-parameter -MT pentax/la-js0n.lo -MD -MP -MF pentax/.deps/la-js0n.Tpo -c pentax/js0n.c  -fno-common -DPIC -o pentax/.libs/la-js0n.o

ndim, 2022-01-16:
One possible fix for any gcc specific CPP stuff is to test for clang first using `#if`, and only if it is not clang, do the gcc detection. Otherwise, the clang will try to behave like gcc, and the gcc detection logic will treat clang like gcc.

This might be useful here, unless someone wants to rewrite the jump address table logic.

ryandesign, 2022-01-17:
It doesn't strike me so much as a Clang vs. GCC issue but rather an issue of what is supported by the current version of whatever the current compiler is.

Here, we're talking about `-Woverride-init`, a warning which evidently at least some versions of GCC support and which Apple Clang 9 and 10 do not recognize. The current version of the [Clang docs do mention `-Woverride-init`](https://clang.llvm.org/docs/DiagnosticsReference.html#woverride-init) which I guess is why the most recent versions of Apple Clang do not complain about this code. The docs do mention that `-Woverride-init` is a synonym for `-Winitializer-overrides` which is enabled by default.

libgphoto2 is requesting for an error to occur when an unknown warning option is used (`-Werror=unknown-warning-option`) so on those certain older Clang versions, it's just doing what you asked. Maybe there are some old GCC versions that don't support `-Woverride-init`; if so, they probably fail on this code too.

I tested recent open source GCC and Clang and they both fail the same way on a simple test program that uses `#pragma GCC diagnostic ignored "-Wfoobar"` so at least there's no Clang/GCC difference there:

```
$ gcc-mp-11 -Werror=unknown-warning-option -o /tmp/test /tmp/test.c
cc1: error: '-Werror=unknown-warning-option': no option '-Wunknown-warning-option'
```
```
$ clang-mp-13 -Werror=unknown-warning-option -o /tmp/test /tmp/test.c
/tmp/test.c:1:32: error: unknown warning group '-Wfoobar', ignored [-Werror,-Wunknown-warning-option]
#pragma GCC diagnostic ignored "-Wfoobar"
                               ^
1 error generated.
```

The libgphoto2 configure script already performs some testing to see what warnings can be enabled. A test for `-Woverride-init` could be added there, and then this js0n.c code could use a conditional with that new define around this pragma.

ryandesign, 2022-01-17:
Actually I should pay attention to what I write. They didn't fail the same way, did they? GCC evidently doesn't support `-Werror=unknown-warning-option` while Clang does.

If I remove `-Werror=unknown-warning-option` then GCC says:

```
$ gcc-mp-11 -o /tmp/test /tmp/test.c
/tmp/test.c:1:32: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
    1 | #pragma GCC diagnostic ignored "-Wfoobar"
      |                                ^~~~~~~~~~
```

But of course since it's just a warning, it won't cause the build to fail.

ndim, 2022-01-18:
I would like to propose a solution like the below minimum working example (MWE).

This solution treats clang specially as the first conditional branch, and only in the non-clang case keeps the behaviour for GCC compilers exactly as it is. This allows us to handle clang separately, i.e. we are not having clang potentially choke on GCC specifics it is not quite as compatible with as it thought it was.

Could you check this MWE on your set of Xcode installations?

    $ clang -o /dev/null -std=c99 -pedantic -Wall -Wextra -Weverything -Werror -Wno-gnu-designator -c mwe.c

```
#ifdef __clang__

#pragma clang diagnostic push
#pragma clang diagnostic ignored "-Winitializer-overrides"

#else /* not clang */

// gcc started warning for the init syntax used here, is not helpful so don't generate the spam, suppressing the warning is really inconsistently supported across versions
#if defined(__GNUC__) && (__GNUC__ > 4 || (__GNUC__ == 4 && __GNUC_MINOR__ >= 6))
#pragma GCC diagnostic push
#endif
#pragma GCC diagnostic ignored "-Wunknown-pragmas"
#pragma GCC diagnostic ignored "-Wpragmas"
#pragma GCC diagnostic ignored "-Winitializer-overrides"
#pragma GCC diagnostic ignored "-Woverride-init"

#endif /* not clang */

static const int list[] = {
  [0 ... 255] = 42,
  [';']       = 23,
};

#ifdef __clang__

#pragma clang diagnostic pop

#else /* not clang */

#if defined(__GNUC__) && (__GNUC__ > 4 || (__GNUC__ == 4 && __GNUC_MINOR__ >= 6))
#pragma GCC diagnostic pop
#endif

#endif /* not clang */

int main()
{
  return 0*list['x'];
}
```

ryandesign, 2022-01-22:
> Could you check this MWE on your set of Xcode installations?

**10.11**

```
$ clang -v
Apple LLVM version 8.0.0 (clang-800.0.42.1)
Target: x86_64-apple-darwin15.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```
```
$ clang -o /dev/null -std=c99 -pedantic -Wall -Wextra -Weverything -Werror -Wno-gnu-designator -c mwe.c
```

**10.12**

```
$ clang -v
Apple LLVM version 9.0.0 (clang-900.0.39.2)
Target: x86_64-apple-darwin16.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```
```
$ clang -o /dev/null -std=c99 -pedantic -Wall -Wextra -Weverything -Werror -Wno-gnu-designator -c mwe.c
```

**10.13**

```
$ clang -v
Apple LLVM version 9.1.0 (clang-902.0.39.2)
Target: x86_64-apple-darwin17.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```
```
$ clang -o /dev/null -std=c99 -pedantic -Wall -Wextra -Weverything -Werror -Wno-gnu-designator -c mwe.c
```

**10.14**

```
$ clang -v
Apple LLVM version 10.0.1 (clang-1001.0.46.4)
Target: x86_64-apple-darwin18.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```
```
$ clang -o /dev/null -std=c99 -pedantic -Wall -Wextra -Weverything -Werror -Wno-gnu-designator -c mwe.c
```

**10.15**

```
% clang -v
Apple clang version 11.0.3 (clang-1103.0.32.62)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```
```
% clang -o /dev/null -std=c99 -pedantic -Wall -Wextra -Weverything -Werror -Wno-gnu-designator -c mwe.c
```

**11**

```
% clang -v
Apple clang version 12.0.5 (clang-1205.0.22.11)
Target: x86_64-apple-darwin20.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```
```
% clang -o /dev/null -std=c99 -pedantic -Wall -Wextra -Weverything -Werror -Wno-gnu-designator -c mwe.c
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
1 error generated.
```
```
% clang -o /dev/null -std=c99 -pedantic -Wall -Wextra -Weverything -Werror -Wno-gnu-designator -Wno-poison-system-directories -c mwe.c
```

ndim, 2022-01-22:
@ryandesign Thanks a lot!

I would not blame that minimum working example for the `-Wno-poison-system-directories` issue. That source file does not even include any files, or link to anything, so there is no part of it demanding input from /usr/local/include.

So I guess this means that this kind of fix is a go.

closed 2022-01-22 by ndim