gphoto/libgphoto2 #762: Support Sony A7 mark 2

gmerceron1000r, 2022-01-17:
Hello
Please could take a look about this camera support?

**Name of the camera*
Sony A7II

**USB IDs**
054c:0a6a

**camera summary output**
Please find file attached
[summary.txt](https://github.com/gphoto/libgphoto2/files/7881434/summary.txt)

**camera configuration output**
Please find file attached
[list-all-config.txt](https://github.com/gphoto/libgphoto2/files/7881437/list-all-config.txt)

**test capture**
  gphoto2 --capture-image-and-download -> works but --interval generate Segmentation fault error, please find log files attached
  gphoto2 --capture-preview -> works

[gphoto_export.txt](https://github.com/gphoto/libgphoto2/files/7881439/gphoto_export.txt)
[gphoto_export_debug.txt](https://github.com/gphoto/libgphoto2/files/7881725/gphoto_export_debug.txt)

Thanks in advance.

msmeissn, 2022-01-17:
can you perhaps get a gdb backtrace of the crash?

msmeissn, 2022-01-17:
or valgrind output,. which should also show the crash location and reason

gmerceron1000r, 2022-01-17:
Hello, 

Thanks for your feedback, I'm not confident with gdb nor valgrind so I take look on previous ticket to understand how it works.
I started gphoto with the following:

gdb gphoto2
r --auto-detect -B 2 --capture-image-and-download --interval 5
... wait for the crash ...
bt

then here is the direct output:
gdb gphoto2 
GNU gdb (Raspbian 8.2.1-2) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from gphoto2...(no debugging symbols found)...done.
(gdb) r --auto-detect -B 2 --capture-image-and-download --interval 5
Starting program: /usr/bin/gphoto2 --auto-detect -B 2 --capture-image-and-download --interval 5
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0xb6253460 (LWP 3940)]
[Thread 0xb6253460 (LWP 3940) exited]
[New Thread 0xb6253460 (LWP 3941)]                                                                                                                                                                   
[Thread 0xb6253460 (LWP 3941) exited]
[New Thread 0xb6253460 (LWP 3942)]
[Thread 0xb6253460 (LWP 3942) exited]
[New Thread 0xb6253460 (LWP 3943)]
[Thread 0xb6253460 (LWP 3943) exited]
[New Thread 0xb6253460 (LWP 3944)]
[Thread 0xb6253460 (LWP 3944) exited]
[New Thread 0xb6253460 (LWP 3945)]
Model                          Port            
----------------------------------------------------------
Sony ILCE-7M2 (Control)        usb:001,014     
Time-lapse mode enabled (interval: 5s).
Bulb mode enabled (exposure time: 2s).
Capturing frame #1...
New file is in location /capt0000.arw on the camera                                                                                                                                                  
File capt0000.arw exists. Overwrite? [y|n] y
Saving file as capt0000.arw
Deleting file /capt0000.arw on the camera
not sleeping (3 seconds behind schedule)
Capturing frame #2...
New file is in location /capt0001.arw on the camera                                                                                                                                                  
File capt0001.arw exists. Overwrite? [y|n] y
Saving file as capt0001.arw
Deleting file /capt0001.arw on the camera
not sleeping (0 seconds behind schedule)
Capturing frame #3...
New file is in location /capt0002.arw on the camera                                                                                                                                                  
Saving file as capt0002.arw
Deleting file /capt0002.arw on the camera
not sleeping (3 seconds behind schedule)
Capturing frame #4...
not sleeping (0 seconds behind schedule)
Capturing frame #5...
New file is in location /capt0003.arw on the camera                                                                                                                                                  
Saving file as capt0003.arw
[... 106 lines cut ...]
New file is in location /capt0025.arw on the camera                                                                                                                                                  
Saving file as capt0025.arw
Deleting file /capt0025.arw on the camera
not sleeping (1 seconds behind schedule)
Capturing frame #27...
Waiting for next capture slot 1 seconds...
New file is in location /capt0026.arw on the camera                                                                                                                                                  
Saving file as capt0026.arw
Deleting file /capt0026.arw on the camera
Capturing frame #28...

Thread 1 "gphoto2" received signal SIGSEGV, Segmentation fault.
strlen () at ../sysdeps/arm/armv6/strlen.S:26
26	../sysdeps/arm/armv6/strlen.S: No such file or directory.
(gdb) bt
#0  strlen () at ../sysdeps/arm/armv6/strlen.S:26
#1  0xb6bee558 in __GI___strdup (s=0x0) at strdup.c:41
#2  0xb59c08c0 in ?? () from /usr/lib/arm-linux-gnueabihf/libgphoto2/2.5.27.1/ptp2.so
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 

The current vertion is gphoto2 2.5.20
OS is Stellarmate 5.10.17 running on raspberry pi4

I try to export a valgrind output

hfiguiere, 2022-01-17:
it's passing `nullptr` to a `strlen()`

(also it's on arm 32-bits which mean the bug might not be revealed in x86_64)

gmerceron1000r, 2022-01-17:
These is a beta test on x64. I try it now then come back to you if I can reproduce the issue.

gmerceron1000r, 2022-01-17:
Here is the x64 test feedback, you will find
- OS specs
- gphoto2 version
- gdb output
- valgrind output

---------------------------------------------------------------------------------------------------------------
###  OS SPECS ###
---------------------------------------------------------------------------------------------------------------
stellarmate@stellarmate:~ $ uname -a
Linux stellarmate 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux

---------------------------------------------------------------------------------------------------------------
###  GPHOTO2 VERSION ###
---------------------------------------------------------------------------------------------------------------
stellarmate@stellarmate:~ $ gphoto2 -v
gphoto2 2.5.27

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.27         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.27.1       standard camlibs, gcc, no ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking

---------------------------------------------------------------------------------------------------------------
### GDB OUTPUT ###
---------------------------------------------------------------------------------------------------------------
stellarmate@stellarmate:~ $ gdb gphoto2 
GNU gdb (Debian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from gphoto2...
(No debugging symbols found in gphoto2)
(gdb) r --auto-detect -B 2 --capture-image-and-download --interval 5
Starting program: /usr/bin/gphoto2 --auto-detect -B 2 --capture-image-and-download --interval 5
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ff71d11c0 (LWP 2413)]
[Thread 0x7ff71d11c0 (LWP 2413) exited]
[New Thread 0x7ff71d11c0 (LWP 2414)]                                            
[Thread 0x7ff71d11c0 (LWP 2414) exited]
[New Thread 0x7ff71d11c0 (LWP 2415)]
[Thread 0x7ff71d11c0 (LWP 2415) exited]
[New Thread 0x7ff71d11c0 (LWP 2416)]
[Thread 0x7ff71d11c0 (LWP 2416) exited]
[New Thread 0x7ff71d11c0 (LWP 2417)]
Model                          Port            
----------------------------------------------------------
Sony ILCE-7M2 (Control)        usb:001,006     
Time-lapse mode enabled (interval: 5s).
Bulb mode enabled (exposure time: 2s).
Capturing frame #1...
Waiting for next capture slot 1 seconds...
[... 113 lines cut ...]
==3763==    by 0x4B24EDB: ??? (in /usr/lib/aarch64-linux-gnu/libpopt.so.0.0.1)
==3763==    by 0x4B24F6B: ??? (in /usr/lib/aarch64-linux-gnu/libpopt.so.0.0.1)
==3763==    by 0x4B2674B: poptGetNextOpt (in /usr/lib/aarch64-linux-gnu/libpopt.so.0.0.1)
==3763==    by 0x10FFE7: ??? (in /usr/bin/gphoto2)
==3763==  If you believe this happened as a result of a stack
==3763==  overflow in your program's main thread (unlikely but
==3763==  possible), you can try to increase the size of the
==3763==  main thread stack using the --main-stacksize= flag.
==3763==  The main thread stack size used in this run was 8388608.
==3763== 
==3763== HEAP SUMMARY:
==3763==     in use at exit: 6,706,720 bytes in 302 blocks
==3763==   total heap usage: 28,467 allocs, 28,165 frees, 2,711,733,815 bytes allocated
==3763== 
==3763== LEAK SUMMARY:
==3763==    definitely lost: 32 bytes in 1 blocks
==3763==    indirectly lost: 48 bytes in 2 blocks
==3763==      possibly lost: 5,736 bytes in 34 blocks
==3763==    still reachable: 6,700,904 bytes in 265 blocks
==3763==         suppressed: 0 bytes in 0 blocks
==3763== Rerun with --leak-check=full to see details of leaked memory
==3763== 
==3763== For lists of detected and suppressed errors, rerun with: -s
==3763== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Segmentation fault

hfiguiere, 2022-01-17:
that's not x86_64. That's aarch64.

(just to clarify)

gmerceron1000r, 2022-01-17:
Yes you right.
I would also add the valgrind --leak-check=full as suggested by the tool itself.

stellarmate@stellarmate:~ $ valgrind --leak-check=full gphoto2 --auto-detect -B 2 --capture-image-and-download --interval 5
==3780== Memcheck, a memory error detector
==3780== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==3780== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==3780== Command: gphoto2 --auto-detect -B 2 --capture-image-and-download --interval 5
==3780== 
Model                          Port                                            
----------------------------------------------------------
Sony ILCE-7M2 (Control)        usb:001,006     
Time-lapse mode enabled (interval: 5s).
Bulb mode enabled (exposure time: 2s).
Capturing frame #1...
New file is in location /capt0000.arw on the camera                            
File capt0000.arw exists. Overwrite? [y|n] y
Saving file as capt0000.arw
Deleting file /capt0000.arw on the camera
==3780== Invalid read of size 1
==3780==    at 0x484D32C: __GI_strlen (vg_replace_strmem.c:460)
==3780==    by 0x4BBFE93: strdup (strdup.c:41)
==3780==    by 0x716BA57: ??? (in /usr/lib/aarch64-linux-gnu/libgphoto2/2.5.27.1/ptp2.so)
==3780==    by 0x713B937: ??? (in /usr/lib/aarch64-linux-gnu/libgphoto2/2.5.27.1/ptp2.so)
==3780==    by 0x48892B7: gp_camera_wait_for_event (in /usr/lib/aarch64-linux-gnu/libgphoto2.so.6.2.0)
==3780==    by 0x11BF67: ??? (in /usr/bin/gphoto2)
==3780==    by 0x11C3D7: ??? (in /usr/bin/gphoto2)
==3780==    by 0x11D2CF: ??? (in /usr/bin/gphoto2)
==3780==    by 0x4B24EDB: ??? (in /usr/lib/aarch64-linux-gnu/libpopt.so.0.0.1)
==3780==    by 0x4B24F6B: ??? (in /usr/lib/aarch64-linux-gnu/libpopt.so.0.0.1)
==3780==    by 0x4B2674B: poptGetNextOpt (in /usr/lib/aarch64-linux-gnu/libpopt.so.0.0.1)
==3780==    by 0x10FFE7: ??? (in /usr/bin/gphoto2)
==3780==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==3780== 
==3780== 
==3780== Process terminating with default action of signal 11 (SIGSEGV)
==3780==  Access not within mapped region at address 0x0
==3780==    at 0x484D32C: __GI_strlen (vg_replace_strmem.c:460)
==3780==    by 0x4BBFE93: strdup (strdup.c:41)
==3780==    by 0x716BA57: ??? (in /usr/lib/aarch64-linux-gnu/libgphoto2/2.5.27.1/ptp2.so)
==3780==    by 0x713B937: ??? (in /usr/lib/aarch64-linux-gnu/libgphoto2/2.5.27.1/ptp2.so)
==3780==    by 0x48892B7: gp_camera_wait_for_event (in /usr/lib/aarch64-linux-gnu/libgphoto2.so.6.2.0)
==3780==    by 0x11BF67: ??? (in /usr/bin/gphoto2)
==3780==    by 0x11C3D7: ??? (in /usr/bin/gphoto2)
==3780==    by 0x11D2CF: ??? (in /usr/bin/gphoto2)
==3780==    by 0x4B24EDB: ??? (in /usr/lib/aarch64-linux-gnu/libpopt.so.0.0.1)
==3780==    by 0x4B24F6B: ??? (in /usr/lib/aarch64-linux-gnu/libpopt.so.0.0.1)
==3780==    by 0x4B2674B: poptGetNextOpt (in /usr/lib/aarch64-linux-gnu/libpopt.so.0.0.1)
==3780==    by 0x10FFE7: ??? (in /usr/bin/gphoto2)
==3780==  If you believe this happened as a result of a stack
==3780==  overflow in your program's main thread (unlikely but
==3780==  possible), you can try to increase the size of the
==3780==  main thread stack using the --main-stacksize= flag.
==3780==  The main thread stack size used in this run was 8388608.
==3780== 
==3780== HEAP SUMMARY:
==3780==     in use at exit: 6,707,352 bytes in 307 blocks
==3780==   total heap usage: 27,853 allocs, 27,546 frees, 2,708,056,072 bytes allocated
==3780== 
==3780== 24 bytes in 1 blocks are possibly lost in loss record 50 of 135
==3780==    at 0x4849E4C: malloc (vg_replace_malloc.c:307)
==3780==    by 0x5A1E923: ??? (in /usr/lib/aarch64-linux-gnu/libusb-1.0.so.0.3.0)
==3780==    by 0x5A1EC2F: ??? (in /usr/lib/aarch64-linux-gnu/libusb-1.0.so.0.3.0)
==3780==    by 0x5A19F9B: libusb_init (in /usr/lib/aarch64-linux-gnu/libusb-1.0.so.0.3.0)
==3780==    by 0x59E5C13: ??? (in /usr/lib/aarch64-linux-gnu/libgphoto2_port/0.12.0/usb1.so)
==3780==    by 0x49199E3: gp_port_set_info (in /usr/lib/aarch64-linux-gnu/libgphoto2_port.so.12.0.0)
==3780==    by 0x4885CB7: gp_camera_set_port_info (in /usr/lib/aarch64-linux-gnu/libgphoto2.so.6.2.0)
==3780==    by 0x115257: ??? (in /usr/bin/gphoto2)
==3780==    by 0x110A9B: ??? (in /usr/bin/gphoto2)
==3780==    by 0x4B63217: (below main) (libc-start.c:308)
[... 131 lines cut ...]
==3780==    at 0x484C164: calloc (vg_replace_malloc.c:760)
==3780==    by 0x5A1CDBF: libusb_alloc_transfer (in /usr/lib/aarch64-linux-gnu/libusb-1.0.so.0.3.0)
==3780==    by 0x59E7BCB: ??? (in /usr/lib/aarch64-linux-gnu/libgphoto2_port/0.12.0/usb1.so)
==3780==    by 0x49174AB: gp_port_open (in /usr/lib/aarch64-linux-gnu/libgphoto2_port.so.12.0.0)
==3780==    by 0x488615B: gp_camera_init (in /usr/lib/aarch64-linux-gnu/libgphoto2.so.6.2.0)
==3780==    by 0x4886DB3: gp_camera_get_single_config (in /usr/lib/aarch64-linux-gnu/libgphoto2.so.6.2.0)
==3780==    by 0x117077: ??? (in /usr/bin/gphoto2)
==3780==    by 0x11C54F: ??? (in /usr/bin/gphoto2)
==3780==    by 0x11D2CF: ??? (in /usr/bin/gphoto2)
==3780==    by 0x4B24EDB: ??? (in /usr/lib/aarch64-linux-gnu/libpopt.so.0.0.1)
==3780==    by 0x4B24F6B: ??? (in /usr/lib/aarch64-linux-gnu/libpopt.so.0.0.1)
==3780==    by 0x4B2674B: poptGetNextOpt (in /usr/lib/aarch64-linux-gnu/libpopt.so.0.0.1)
==3780== 
==3780== LEAK SUMMARY:
==3780==    definitely lost: 32 bytes in 1 blocks
==3780==    indirectly lost: 48 bytes in 2 blocks
==3780==      possibly lost: 5,736 bytes in 34 blocks
==3780==    still reachable: 6,701,536 bytes in 270 blocks
==3780==         suppressed: 0 bytes in 0 blocks
==3780== Reachable blocks (those to which a pointer was found) are not shown.
==3780== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==3780== 
==3780== For lists of detected and suppressed errors, rerun with: -s
==3780== ERROR SUMMARY: 12 errors from 12 contexts (suppressed: 0 from 0)
Segmentation fault

msmeissn, 2022-01-18:
is it possible to build / install with debug symbols so we can see where the strlen is called from?

msmeissn, 2022-01-18:
if you run configure prepend CFLAGS="-g -O2"   or so 

CFLAGS="-g -O2" ./configure .....

gmerceron1000r, 2022-01-18:
For posterity, I used the following process:
- remove valgrind installed package form sources : apt remove valgrind
- download tar.bz2 from the official website https://valgrind.org/downloads/current.html
- extract the archive : cd Downloads && tar jxvf "archive_name"
- compile and install : cd "extracted_folder" && CFLAGS="-g -O2" ./configure && make && sudo make install 
- then retry : valgrind gphoto2 --auto-detect -B 2 --capture-image-and-download --interval 5

output still prompt ??? I continue to look why?

#############
## OUTPUT ##
#############

stellarmate@stellarmate:~ $ rm *.arw
stellarmate@stellarmate:~ $ valgrind gphoto2 --auto-detect -B 2 --capture-image-and-download --interval 5
==2069== Memcheck, a memory error detector
==2069== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==2069== Using Valgrind-3.18.1 and LibVEX; rerun with -h for copyright info
==2069== Command: gphoto2 --auto-detect -B 2 --capture-image-and-download --interval 5
==2069== 
Model                          Port                                            
----------------------------------------------------------
Sony ILCE-7M2 (Control)        usb:001,006     
Time-lapse mode enabled (interval: 5s).
Bulb mode enabled (exposure time: 2s).
Capturing frame #1...
Waiting for next capture slot 1 seconds...
New file is in location /capt0000.arw on the camera                            
Saving file as capt0000.arw
Deleting file /capt0000.arw on the camera
Capturing frame #2...
not sleeping (0 seconds behind schedule)
Capturing frame #3...
New file is in location /capt0001.arw on the camera                            
Saving file as capt0001.arw
Deleting file /capt0001.arw on the camera
not sleeping (20 seconds behind schedule)
Capturing frame #4...
Waiting for next capture slot 2 seconds...
New file is in location /capt0002.arw on the camera                            
Saving file as capt0002.arw
Deleting file /capt0002.arw on the camera
Capturing frame #5...
Waiting for next capture slot 0 seconds...
Capturing frame #6...
New file is in location /capt0003.arw on the camera                            
Saving file as capt0003.arw
Deleting file /capt0003.arw on the camera
not sleeping (17 seconds behind schedule)
Capturing frame #7...
New file is in location /capt0004.arw on the camera                            
Saving file as capt0004.arw
Deleting file /capt0004.arw on the camera
not sleeping (21 seconds behind schedule)
Capturing frame #8...
New file is in location /capt0005.arw on the camera                            
Saving file as capt0005.arw
Deleting file /capt0005.arw on the camera
not sleeping (19 seconds behind schedule)
Capturing frame #9...
New file is in location /capt0006.arw on the camera                            
Saving file as capt0006.arw
Deleting file /capt0006.arw on the camera
not sleeping (23 seconds behind schedule)
Capturing frame #10...
New file is in location /capt0007.arw on the camera                            
Saving file as capt0007.arw
Deleting file /capt0007.arw on the camera
not sleeping (22 seconds behind schedule)
Capturing frame #11...
[... 29 lines cut ...]
==2069==    by 0x4B29EDB: ??? (in /usr/lib/aarch64-linux-gnu/libpopt.so.0.0.1)
==2069==    by 0x4B29F6B: ??? (in /usr/lib/aarch64-linux-gnu/libpopt.so.0.0.1)
==2069==    by 0x4B2B74B: poptGetNextOpt (in /usr/lib/aarch64-linux-gnu/libpopt.so.0.0.1)
==2069==    by 0x10FFE7: ??? (in /usr/bin/gphoto2)
==2069==  If you believe this happened as a result of a stack
==2069==  overflow in your program's main thread (unlikely but
==2069==  possible), you can try to increase the size of the
==2069==  main thread stack using the --main-stacksize= flag.
==2069==  The main thread stack size used in this run was 8388608.
==2069== 
==2069== HEAP SUMMARY:
==2069==     in use at exit: 6,726,474 bytes in 437 blocks
==2069==   total heap usage: 41,648 allocs, 41,211 frees, 23,664,900,952 bytes allocated
==2069== 
==2069== LEAK SUMMARY:
==2069==    definitely lost: 32 bytes in 1 blocks
==2069==    indirectly lost: 48 bytes in 2 blocks
==2069==      possibly lost: 5,736 bytes in 34 blocks
==2069==    still reachable: 6,720,658 bytes in 400 blocks
==2069==         suppressed: 0 bytes in 0 blocks
==2069== Rerun with --leak-check=full to see details of leaked memory
==2069== 
==2069== For lists of detected and suppressed errors, rerun with: -s
==2069== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Segmentation fault

msmeissn, 2022-01-18:
did not yet really work: ... would not have ??? here
 ==2069== by 0x4BC4E93: strdup (strdup.c:41)
==2069== by 0x7170A57: ??? (in /usr/lib/aarch64-linux-gnu/libgphoto2/2.5.27.1/ptp2.so)

I put some hardening into 2.5.28.1 current git state, if you can update to that?

gmerceron1000r, 2022-01-19:
Hello, 

Here is the process used.
- Download the Code Zipped from the main page
- sudo apt remove gphoto2 to remove the previous version

$ sudo apt remove gphoto2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following package was automatically installed and is no longer required:
  libcdk5nc6
Use 'sudo apt autoremove' to remove it.
The following packages will be REMOVED:
  gphoto2
0 upgraded, 0 newly installed, 1 to remove and 84 not upgraded.
After this operation, 970 kB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 133470 files and directories currently installed.)
Removing gphoto2 (2.5.27-1) ...
Processing triggers for man-db (2.9.4-2) ...

- Read the INSTALL file from main page
- Install the Step 0 dependancies : $ sudo apt install automake autoconf pkg-config autopoint gettext libtool
- Move zip file in /usr/local : $ mv libgphoto2-master.zip /usr/local/
- Unzip the archive:  $ sudo unzip libgphoto2-master.zip 
- Enter in the folder: cd libgphoto2-master
- Run autoreconf $ sudo autoreconf --install --symlink
- Run configure: $ sudo ./configure --prefix=/usr/local
- Run make : $ sudo make
- Install : $ sudo make install

Now I'm no longer able to run "gphoto2" commands as I did previously in my commands.
BTW the version seems updated to the suggested one.

stellarmate@stellarmate:~ $ gphoto2-config --version
libgphoto2 2.5.28.1

stellarmate@stellarmate:~ $ gphoto2-
gphoto2-config       gphoto2-port-config

msmeissn, 2022-01-19:
you should not need to remove the "gphoto2" commandline tool if you rebuiold the library, can you reinstall it?

gmerceron1000r, 2022-01-19:
I reinstalled gphoto as suggested : sudp apt install -y gphoto2

stellarmate@stellarmate:~ $ gphoto2 --version
gphoto2 2.5.27

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.27         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.27.1       standard camlibs, gcc, no ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking

Then I re-run the compilation process:

$ mv /usr/local/libgphoto2-master (unzipped form github)
$ sudo autoreconf --install --symlink
$ sudo ./configure --prefix=/usr/local
$ sudo make
$ sudo make install

Here is the result:

stellarmate@stellarmate:~ $ gphoto2 --version
gphoto2 2.5.27

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.27         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.27.1       standard camlibs, gcc, no ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking
stellarmate@stellarmate:~ $ gphoto2-config --version
libgphoto2 2.5.28.1

the error still remain

stellarmate@stellarmate:~ $ valgrind gphoto2 --auto-detect -B 2 --capture-image-and-download --interval 5
==2071== Memcheck, a memory error detector
==2071== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==2071== Using Valgrind-3.18.1 and LibVEX; rerun with -h for copyright info
==2071== Command: gphoto2 --auto-detect -B 2 --capture-image-and-download --interval 5
==2071== 
Model                          Port                                            
----------------------------------------------------------
Sony ILCE-7M2 (Control)        usb:001,006     
Time-lapse mode enabled (interval: 5s).
Bulb mode enabled (exposure time: 2s).
Capturing frame #1...
Waiting for next capture slot 1 seconds...
New file is in location /capt0000.arw on the camera                            
Saving file as capt0000.arw
Deleting file /capt0000.arw on the camera
Capturing frame #2...
not sleeping (1 seconds behind schedule)
Capturing frame #3...
New file is in location /capt0001.arw on the camera                            
Saving file as capt0001.arw
Deleting file /capt0001.arw on the camera
not sleeping (20 seconds behind schedule)
Capturing frame #4...
Waiting for next capture slot 1 seconds...
New file is in location /capt0002.arw on the camera                            
[... 135 lines cut ...]
==2071==    by 0x4B29EDB: ??? (in /usr/lib/aarch64-linux-gnu/libpopt.so.0.0.1)
==2071==    by 0x4B29F6B: ??? (in /usr/lib/aarch64-linux-gnu/libpopt.so.0.0.1)
==2071==    by 0x4B2B74B: poptGetNextOpt (in /usr/lib/aarch64-linux-gnu/libpopt.so.0.0.1)
==2071==    by 0x10FFE7: ??? (in /usr/bin/gphoto2)
==2071==  If you believe this happened as a result of a stack
==2071==  overflow in your program's main thread (unlikely but
==2071==  possible), you can try to increase the size of the
==2071==  main thread stack using the --main-stacksize= flag.
==2071==  The main thread stack size used in this run was 8388608.
==2071== 
==2071== HEAP SUMMARY:
==2071==     in use at exit: 6,727,034 bytes in 441 blocks
==2071==   total heap usage: 65,281 allocs, 64,840 frees, 60,794,565,632 bytes allocated
==2071== 
==2071== LEAK SUMMARY:
==2071==    definitely lost: 32 bytes in 1 blocks
==2071==    indirectly lost: 48 bytes in 2 blocks
==2071==      possibly lost: 5,736 bytes in 34 blocks
==2071==    still reachable: 6,721,218 bytes in 404 blocks
==2071==         suppressed: 0 bytes in 0 blocks
==2071== Rerun with --leak-check=full to see details of leaked memory
==2071== 
==2071== For lists of detected and suppressed errors, rerun with: -s
==2071== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Segmentation fault

glundby, 2022-01-19:
Reminds me of Segmentation Fault i keep getting both with Sony A7 and Sony 5000. Tried loading newest Kstars 3.5.7 but find the error keeps happening after 40-73 captures. It kills the whole system and it restarts.

msmeissn, 2022-01-21:
it went back to 2.5.27.1
 libgphoto2 2.5.27.1 standard camlibs, gcc, no ltdl, EXIF

either you build both libgphoto2 or gphoto2 locally, or you overwrite the installed libgphoto2 version by your own build.

SiuyuenChow, 2022-05-07:
@gmerceron1000r  
hello ,
Have you solved this problem yet? I recently encountered this problem with a SONY compatible camera model :A7M4

*** Contexterror ***              
You need to specify a folder starting with /store_xxxxxxxxx/
@
*** Contexterror ***

msmeissn, 2022-05-08:
@gmerceron1000r this is unrelated, please keep this in your seperate issue

I fixed another crash reason in current libgphoto2 master git. Need to run a release at some point in near future. :/