gphoto/libgphoto2 #767: Changing focus on Canon R - there should be a better way to do it?

pootle, 2022-01-22:
I'm writing a program to adjust focus for focus stacking, with my  camera the valid values of  'manualfocusdrive' are
['Near 1', 'Near 2', 'Near 3', 'None', 'Far 1', 'Far 2', 'Far 3']

It would be really useful to be able to request a repeat number when doing this: 
    xxx.change_focus('Near 2', 4) 

ATM  if I change setting (say) 'Far 2' several times, then only the first takes effect.

It seems you have to alternate 'Far 2' with 'None' for subsequent settings to take effect.  Is this the proper way to rack the focus by multiple steps?

Watching the lens' focus while running this, the focus shifts only once in the first set of tests, but every change shifts focus on the second set of tests. I'm running the standard distro version (Fedora 34) libgphoto2-2.5.27-1.fc34.x86_64 and I get the same result running on a Raspberry Pi which is more up to date - This is the full stack including  Jim Easterbrook's Python binding:

    python -m gphoto2
    python-gphoto2 version: 2.3.1
    libgphoto2 version: ['2.5.28.1', 'standard camlibs (SKIPPING docupen lumix)', 'gcc', 'no ltdl', 'no EXIF']
    libgphoto2_port version: ['0.12.0', 'iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi', 'gcc', 'ltdl', 'EXIF', 'USB', 'serial without locking']
    python-gphoto2 examples: /usr/local/lib/python3.9/dist-packages/gphoto2/examples

Here is the test (python) program

    #!/usr/bin/env python
    
    import gphoto2 as gp, time
    
    wtest = 'manualfocusdrive'
    
    def main():
        camera=gp.Camera()
        try:
            camera.init()
        except gp.GPhoto2Error as gpe:
            if gpe.code == -105:
                print('ERROR: Unable to contact camera - is it connected? has it timed out?', file=sys.stderr)
                sys.exit(1)
            else:
                raise
        cam_cfg=camera.get_config()
        fchangewidg = cam_cfg.get_child_by_name(wtest)
        for x in range(6):         # this cycle only changes focus once
            fchangewidg.set_value('Far 2')
            camera.set_config(cam_cfg)
            time.sleep(.5)
        for x in range(6):        # this cycle only changes focus once
            fchangewidg.set_value('Near 2')
            camera.set_config(cam_cfg)
            time.sleep(.5)
        time.sleep(1)
        for x in range(6):        # this cycle changes focus every time
            fchangewidg.set_value('None')
            camera.set_config(cam_cfg) 
            fchangewidg.set_value('Far 2')
            camera.set_config(cam_cfg)
            time.sleep(.5)
        for x in range(6):        # this cycle changes focus every time
            fchangewidg.set_value('None')
            camera.set_config(cam_cfg) 
            fchangewidg.set_value('Near 2')
            camera.set_config(cam_cfg)
            time.sleep(.5)
    if __name__ == '__main__':
        main()

msmeissn, 2022-01-22:
If you fetch the config tree just once, but set the same value multiple times you need to mark the widget as changed inbetween calls:

gp_widget_set_changed(widget,1); would be the libgphoto2 API.

fchangewidg.set_changed(1)  is probably the python binding equivalent.

pootle, 2022-01-23:
Just calling set_changed after each update doesn't update the camera, but using camera.set_single_config(....) after each set_value call works fine and hopefully is a bit more efficient.

closed 2022-01-23 by pootle