gphoto/libgphoto2 #769: Cant conect nikon d3000

Manuudatu00, 2022-02-04:
I have been trying to conect my nikon d3000 but the raspberry doesn’t recognize the camera, i have tested it with diferent cables and nothing. Something to do?

glundby, 2022-02-05:
Tried with generic indi_gphoto as driver? I have this problem with my Sony a5000, It does not work with Sony DSLR as device but with gphoto generic it works. My Sony a7 works with Sony DSLR. Sendt fra E-post for Windows Fra: Manuudatu00Sendt: fredag 4. februar 2022 kl. 23.31Til: gphoto/libgphoto2Kopi: SubscribedEmne: [gphoto/libgphoto2] Cant conect nikon d3000 (Issue #769) I have been trying to conect my nikon d3000 but the raspberry doesn’t recognize the camera, i have tested it with diferent cables and nothing. Something to do?

msmeissn, 2022-02-07:
can you report
gphoto2 --version

?

is it shown with 

lsusb

when it is attached?