gphoto/libgphoto2 #770: Canon gp_camera_capture returns incorrect path

jim-easterbrook, 2022-02-09:
Running `gp_camera_capture` saves the image to SD card (e.g. `/store_00020001/DCIM/100CANON/IMG_7755.JPG`) but the path returned by `gp_camera_capture` has an empty folder and the name `capt0000.jpg`.

This is on my Canon 100D and a python-gphoto2 user's Canon D5.

The problem was introduced when upgrading from libgphoto2 v2.5.27 to v2.5.28.

msmeissn, 2022-02-11:
could you add a debug trace?
e.g.
gphoto2 --debug --debug-logfile=capture.log --capture-image

jim-easterbrook, 2022-02-11:
I don't have the gphoto2 command line tool installed at the moment. I've got a log file from the Python script that uses `gp_camera_capture`. As it's 6.6 MByte I've used zip to compress it.
[log.txt.zip](https://github.com/gphoto/libgphoto2/files/8048793/log.txt.zip)
.

jim-easterbrook, 2022-02-11:
Here's a similar log file done with libgphoto2 v 2.5.27.
[log_2.5.27.txt.zip](https://github.com/gphoto/libgphoto2/files/8048834/log_2.5.27.txt.zip)

msmeissn, 2022-02-12:
i found a place where i refactored a bit too much. can you check libgphoto2 git master branch with above commit?

jim-easterbrook, 2022-02-12:
Will do so in a couple of days time. Thanks for investigating the problem.

jim-easterbrook, 2022-02-12:
OK, I lied. I've just tried it and v 2.5.28.1 works fine for me.

closed 2022-02-13 by msmeissn