gphoto/libgphoto2 #772: mem leak when camera init failed

czln, 2022-02-17:
**Describe the bug**

I'm trying sample-capture, yet got memory leak with leaksanitizer

```
Camera init.  Takes about 10 seconds.

*** Contexterror ***
Could not detect any camera
  Retval of capture_to_file: -105

=================================================================
==28261==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 4096 byte(s) in 1 object(s) allocated from:
    #0 0x7f8f871b1b in malloc (/usr/lib/aarch64-linux-gnu/libasan.so.4+0xcfb1b)
    #1 0x7f8ca957ff  (<unknown module>)
    #2 0x7f8ca98c17  (<unknown module>)
    #3 0x7f8cc3011b  (<unknown module>)
    #4 0x7f8cc2e93b  (<unknown module>)
    #5 0x7f8cc26057  (<unknown module>)
    #6 0x7f8ce99643  (<unknown module>)
    #7 0x7f8f46b4d7  (/usr/lib/aarch64-linux-gnu/libgphoto2_port.so.12+0x24d7)
    #8 0x7f8f453ab7  (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x3ab7)
    #9 0x7f8f4534a3  (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x34a3)
    #10 0x7f8f4540b3 in lt_dlforeachfile (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x40b3)
    #11 0x7f8f46b98f in gp_port_info_list_load (/usr/lib/aarch64-linux-gnu/libgphoto2_port.so.12+0x298f)
    #12 0x7f8f71a533 in gp_camera_init (/usr/lib/aarch64-linux-gnu/libgphoto2.so.6+0x9533)
    #13 0x55787a3083 in main /home/zelin/debug/example.c:112
    #14 0x7f8f5d871f in __libc_start_main (/lib/aarch64-linux-gnu/libc.so.6+0x2071f)
    #15 0x55787a26e3  (/home/zelin/debug/a.out+0x16e3)

Direct leak of 4096 byte(s) in 1 object(s) allocated from:
    #0 0x7f8f871b1b in malloc (/usr/lib/aarch64-linux-gnu/libasan.so.4+0xcfb1b)
    #1 0x7f8ca957ff  (<unknown module>)
    #2 0x7f8ca98f53  (<unknown module>)
    #3 0x7f8ca9cf07  (<unknown module>)
    #4 0x7f8ca9d35f  (<unknown module>)
    #5 0x7f8ca9f19f  (<unknown module>)
    #6 0x7f8ca9f4b3  (<unknown module>)
    #7 0x7f8caa0b67  (<unknown module>)
    #8 0x7f8cc30143  (<unknown module>)
    #9 0x7f8cc2e93b  (<unknown module>)
    #10 0x7f8cc26057  (<unknown module>)
    #11 0x7f8ce97d9b  (<unknown module>)
    #12 0x7f8f46e007 in gp_port_set_info (/usr/lib/aarch64-linux-gnu/libgphoto2_port.so.12+0x5007)
    #13 0x7f8f717313 in gp_abilities_list_detect (/usr/lib/aarch64-linux-gnu/libgphoto2.so.6+0x6313)
    #14 0x7f8f71a54f in gp_camera_init (/usr/lib/aarch64-linux-gnu/libgphoto2.so.6+0x954f)
    #15 0x55787a3083 in main /home/zelin/debug/example.c:112
    #16 0x7f8f5d871f in __libc_start_main (/lib/aarch64-linux-gnu/libc.so.6+0x2071f)
    #17 0x55787a26e3  (/home/zelin/debug/a.out+0x16e3)

Direct leak of 4096 byte(s) in 1 object(s) allocated from:
    #0 0x7f8f871b1b in malloc (/usr/lib/aarch64-linux-gnu/libasan.so.4+0xcfb1b)
    #1 0x7f8ca957ff  (<unknown module>)
    #2 0x7f8ca98c17  (<unknown module>)
    #3 0x7f8cc3011b  (<unknown module>)
    #4 0x7f8cc2e93b  (<unknown module>)
    #5 0x7f8cc26057  (<unknown module>)
    #6 0x7f8ce97d9b  (<unknown module>)
    #7 0x7f8f46e007 in gp_port_set_info (/usr/lib/aarch64-linux-gnu/libgphoto2_port.so.12+0x5007)
    #8 0x7f8f717313 in gp_abilities_list_detect (/usr/lib/aarch64-linux-gnu/libgphoto2.so.6+0x6313)
    #9 0x7f8f71a54f in gp_camera_init (/usr/lib/aarch64-linux-gnu/libgphoto2.so.6+0x954f)
    #10 0x55787a3083 in main /home/zelin/debug/example.c:112
    #11 0x7f8f5d871f in __libc_start_main (/lib/aarch64-linux-gnu/libc.so.6+0x2071f)
    #12 0x55787a26e3  (/home/zelin/debug/a.out+0x16e3)

Direct leak of 4096 byte(s) in 1 object(s) allocated from:
    #0 0x7f8f871b1b in malloc (/usr/lib/aarch64-linux-gnu/libasan.so.4+0xcfb1b)
    #1 0x7f8ca957ff  (<unknown module>)
    #2 0x7f8ca98f53  (<unknown module>)
    #3 0x7f8ca9cf07  (<unknown module>)
    #4 0x7f8ca9d35f  (<unknown module>)
    #5 0x7f8ca9f19f  (<unknown module>)
    #6 0x7f8ca9f4b3  (<unknown module>)
    #7 0x7f8caa0b67  (<unknown module>)
    #8 0x7f8cc30143  (<unknown module>)
    #9 0x7f8cc2e93b  (<unknown module>)
    #10 0x7f8cc26057  (<unknown module>)
    #11 0x7f8ce99643  (<unknown module>)
    #12 0x7f8f46b4d7  (/usr/lib/aarch64-linux-gnu/libgphoto2_port.so.12+0x24d7)
    #13 0x7f8f453ab7  (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x3ab7)
    #14 0x7f8f4534a3  (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x34a3)
    #15 0x7f8f4540b3 in lt_dlforeachfile (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x40b3)
    #16 0x7f8f46b98f in gp_port_info_list_load (/usr/lib/aarch64-linux-gnu/libgphoto2_port.so.12+0x298f)
    #17 0x7f8f71a533 in gp_camera_init (/usr/lib/aarch64-linux-gnu/libgphoto2.so.6+0x9533)
    #18 0x55787a3083 in main /home/zelin/debug/example.c:112
    #19 0x7f8f5d871f in __libc_start_main (/lib/aarch64-linux-gnu/libc.so.6+0x2071f)
    #20 0x55787a26e3  (/home/zelin/debug/a.out+0x16e3)

SUMMARY: AddressSanitizer: 16384 byte(s) leaked in 4 allocation(s).
```

here the `example.c` is basically `sample-capture.c`

libgphoto2 version: 2.5.28

**To Reproduce**
I compile the code with gcc -g -fsanitize=address and run the programme, but you may also reproduce it with valgrind

czln, 2022-02-17:
when init succeeded, still got memory leak.

**camera model**
ILCE-5100

```
Camera init.  Takes about 10 seconds.
Capturing.
  Retval: 0
Pathname on the camera: //capt0000.jpg
       file info reported flags: 157
    info reported mtime: 1645064685
     info reported size: 2703516
     info reported type: image/jpeg
  Retval: 0
  Retval: 0
Deleting.
  Retval: 0

=================================================================
==28331==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 4096 byte(s) in 1 object(s) allocated from:
    #0 0x7fb45b9b1b in malloc (/usr/lib/aarch64-linux-gnu/libasan.so.4+0xcfb1b)
    #1 0x7fb17dc7ff  (/lib/aarch64-linux-gnu/libudev.so.1+0x67ff)
    #2 0x7fb17dfc17  (/lib/aarch64-linux-gnu/libudev.so.1+0x9c17)
    #3 0x7fb193011b  (/lib/aarch64-linux-gnu/libusb-1.0.so.0+0x1011b)
    #4 0x7fb192e93b  (/lib/aarch64-linux-gnu/libusb-1.0.so.0+0xe93b)
    #5 0x7fb1926057 in libusb_init (/lib/aarch64-linux-gnu/libusb-1.0.so.0+0x6057)
    #6 0x7fb1bec643 in gp_port_library_list (/usr/lib/aarch64-linux-gnu/libgphoto2_port/0.12.0/usb1.so+0x4643)
    #7 0x7fb41b34d7  (/usr/lib/aarch64-linux-gnu/libgphoto2_port.so.12+0x24d7)
    #8 0x7fb419bab7  (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x3ab7)
    #9 0x7fb419b4a3  (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x34a3)
    #10 0x7fb419c0b3 in lt_dlforeachfile (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x40b3)
    #11 0x7fb41b398f in gp_port_info_list_load (/usr/lib/aarch64-linux-gnu/libgphoto2_port.so.12+0x298f)
    #12 0x7fb4462533 in gp_camera_init (/usr/lib/aarch64-linux-gnu/libgphoto2.so.6+0x9533)
    #13 0x558a67e083 in main /home/zelin/debug/example.c:112
    #14 0x7fb432071f in __libc_start_main (/lib/aarch64-linux-gnu/libc.so.6+0x2071f)
    #15 0x558a67d6e3  (/home/zelin/debug/a.out+0x16e3)

Direct leak of 4096 byte(s) in 1 object(s) allocated from:
    #0 0x7fb45b9b1b in malloc (/usr/lib/aarch64-linux-gnu/libasan.so.4+0xcfb1b)
    #1 0x7fb17dc7ff  (/lib/aarch64-linux-gnu/libudev.so.1+0x67ff)
    #2 0x7fb17dff53  (/lib/aarch64-linux-gnu/libudev.so.1+0x9f53)
    #3 0x7fb17e3f07  (/lib/aarch64-linux-gnu/libudev.so.1+0xdf07)
    #4 0x7fb17e435f  (/lib/aarch64-linux-gnu/libudev.so.1+0xe35f)
    #5 0x7fb17e619f  (/lib/aarch64-linux-gnu/libudev.so.1+0x1019f)
    #6 0x7fb17e64b3  (/lib/aarch64-linux-gnu/libudev.so.1+0x104b3)
    #7 0x7fb17e7b67 in udev_enumerate_scan_devices (/lib/aarch64-linux-gnu/libudev.so.1+0x11b67)
    #8 0x7fb1930143  (/lib/aarch64-linux-gnu/libusb-1.0.so.0+0x10143)
    #9 0x7fb192e93b  (/lib/aarch64-linux-gnu/libusb-1.0.so.0+0xe93b)
    #10 0x7fb1926057 in libusb_init (/lib/aarch64-linux-gnu/libusb-1.0.so.0+0x6057)
    #11 0x7fb1bec643 in gp_port_library_list (/usr/lib/aarch64-linux-gnu/libgphoto2_port/0.12.0/usb1.so+0x4643)
    #12 0x7fb41b34d7  (/usr/lib/aarch64-linux-gnu/libgphoto2_port.so.12+0x24d7)
    #13 0x7fb419bab7  (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x3ab7)
    #14 0x7fb419b4a3  (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x34a3)
    #15 0x7fb419c0b3 in lt_dlforeachfile (/usr/lib/aarch64-linux-gnu/libltdl.so.7+0x40b3)
    #16 0x7fb41b398f in gp_port_info_list_load (/usr/lib/aarch64-linux-gnu/libgphoto2_port.so.12+0x298f)
    #17 0x7fb4462533 in gp_camera_init (/usr/lib/aarch64-linux-gnu/libgphoto2.so.6+0x9533)
    #18 0x558a67e083 in main /home/zelin/debug/example.c:112
    #19 0x7fb432071f in __libc_start_main (/lib/aarch64-linux-gnu/libc.so.6+0x2071f)
    #20 0x558a67d6e3  (/home/zelin/debug/a.out+0x16e3)

Direct leak of 4096 byte(s) in 1 object(s) allocated from:
    #0 0x7fb45b9b1b in malloc (/usr/lib/aarch64-linux-gnu/libasan.so.4+0xcfb1b)
    #1 0x7fb17dc7ff  (/lib/aarch64-linux-gnu/libudev.so.1+0x67ff)
    #2 0x7fb17dff53  (/lib/aarch64-linux-gnu/libudev.so.1+0x9f53)
    #3 0x7fb17e3f07  (/lib/aarch64-linux-gnu/libudev.so.1+0xdf07)
    #4 0x7fb17e435f  (/lib/aarch64-linux-gnu/libudev.so.1+0xe35f)
    #5 0x7fb17e619f  (/lib/aarch64-linux-gnu/libudev.so.1+0x1019f)
[... 186 lines cut ...]
    #2 0x7fb4461f13 in gp_camera_set_port_info (/usr/lib/aarch64-linux-gnu/libgphoto2.so.6+0x8f13)
    #3 0x7fb446293f in gp_camera_init (/usr/lib/aarch64-linux-gnu/libgphoto2.so.6+0x993f)
    #4 0x558a67e083 in main /home/zelin/debug/example.c:112
    #5 0x7fb432071f in __libc_start_main (/lib/aarch64-linux-gnu/libc.so.6+0x2071f)
    #6 0x558a67d6e3  (/home/zelin/debug/a.out+0x16e3)

Indirect leak of 12 byte(s) in 1 object(s) allocated from:
    #0 0x7fb455c4d7 in __interceptor_strdup (/usr/lib/aarch64-linux-gnu/libasan.so.4+0x724d7)
    #1 0x7fb41b5f47 in gp_port_set_info (/usr/lib/aarch64-linux-gnu/libgphoto2_port.so.12+0x4f47)
    #2 0x7fb4461f13 in gp_camera_set_port_info (/usr/lib/aarch64-linux-gnu/libgphoto2.so.6+0x8f13)
    #3 0x7fb446293f in gp_camera_init (/usr/lib/aarch64-linux-gnu/libgphoto2.so.6+0x993f)
    #4 0x558a67e083 in main /home/zelin/debug/example.c:112
    #5 0x7fb432071f in __libc_start_main (/lib/aarch64-linux-gnu/libc.so.6+0x2071f)
    #6 0x558a67d6e3  (/home/zelin/debug/a.out+0x16e3)

Indirect leak of 2 byte(s) in 1 object(s) allocated from:
    #0 0x7fb455c4d7 in __interceptor_strdup (/usr/lib/aarch64-linux-gnu/libasan.so.4+0x724d7)
    #1 0x7fb446bc07 in gp_filesystem_new (/usr/lib/aarch64-linux-gnu/libgphoto2.so.6+0x12c07)
    #2 0x7fb4461bd3 in gp_camera_new (/usr/lib/aarch64-linux-gnu/libgphoto2.so.6+0x8bd3)
    #3 0x558a67e043 in main /home/zelin/debug/example.c:104
    #4 0x7fb432071f in __libc_start_main (/lib/aarch64-linux-gnu/libc.so.6+0x2071f)
    #5 0x558a67d6e3  (/home/zelin/debug/a.out+0x16e3)

SUMMARY: AddressSanitizer: 25245 byte(s) leaked in 29 allocation(s).
```

Sija, 2024-10-02:
Is this still a case?

czln, 2025-05-20:
> Is this still a case?

well, I changed my job years ago. So sorry that I can't test this any more.