gphoto/libgphoto2 #774: Heap use-after-free in ptp2 camlib

trilader, 2022-03-10:
To help with investigating a memory leak in my application I've built libgphoto2 with address sanitizer enabled and noticed that it aborted execution when calling `gp_camera_wait_for_event` in a loop and triggering a capture with the camera while there is an SD card in the camera. This issue doesn't happen without an SD card in the camera (or at least I've not seen it happen).

The camera I've tested this with is a Canon EOS 77D.

This is based on 30de5cf6509ab5cf9f4b4f672123722e15daa5ce. The issue seems to be (at least the free happens in it) a not quite correct `realloc`, or something not getting updated after it, in `ptp_list_folder_eos`.

I've attached the output of my testing application (with debug log level) and the output of the address sanitizer as well.
Log is attached as a file rather than inline comment as comments can't have more than 65536 characters in them...

[libgphoto2_use_after_free.txt](https://github.com/gphoto/libgphoto2/files/8222114/libgphoto2_use_after_free.txt)

msmeissn, 2022-03-10:
i think i fixed that place , another fallout from refactoring.

msmeissn, 2022-03-10:
can you recheck?

closed 2022-03-10 by msmeissn