gphoto/libgphoto2 #777: Segmentation fault in print-camera-list

Keruspe, 2022-03-22:
**Describe the bug**

No matter how print-camera-list is invoked, it will segfault.

**Name the camera**

No camera involved

**libgphoto2 and gphoto2 version**

2.5.29

**To Reproduce**
Steps to reproduce the behavior:

Launch e.g. `/usr/lib/libgphoto2/print-camera-list human-readable`
Expected output appears, then `Segmentation fault      (core dumped) `

Backtrace is not really helpful:

```
(gdb) bt
#0  0x0000000000000000 in ?? ()
#1  0x00007f974caacd24 in _dl_fini () from /usr/x86_64-pc-linux-gnu/lib/ld-linux-x86-64.so.2
#2  0x00007f974c848c78 in __run_exit_handlers () from /usr/x86_64-pc-linux-gnu/lib/libc.so.6
#3  0x00007f974c848d7e in exit () from /usr/x86_64-pc-linux-gnu/lib/libc.so.6
#4  0x00007f974c831655 in __libc_start_call_main () from /usr/x86_64-pc-linux-gnu/lib/libc.so.6
#5  0x00007f974c831709 in __libc_start_main () from /usr/x86_64-pc-linux-gnu/lib/libc.so.6
#6  0x0000000000204c05 in _start ()
```

Removing the latest call to `lt_dlexit` and the cal to `lt_dlclose` withing the valgrind check inside `libgphoto2/gphoto2-abilities-list.c` "fixes" the problem. (both are needed, removing only one of those two doesn't change a thing)

msmeissn, 2022-03-23:
it kind of weird, i cannot reproduce this on my side.

is it outputting cameras, or just crashing?

is gphoto2 --list-cameras  working?

Keruspe, 2022-03-23:
It prints the cameras *then* crashes, when glibc calls all the cleanup stuff at the end of execution, from what I get

Keruspe, 2022-03-23:
I do not have gphoto2 installed but I can try.

I also can try adding some kind of sleep right at the end of print-camera-list to make sure it crashes after that indeed

msmeissn, 2022-03-24:
its weird, it is a NULL pointer dereference on exit, but its not clear where it might come from. 

run
LD_DEBUG=all /usr/lib/libgphoto2/print-camera-list human-readable   >logfile 2>&1

so we can see in which destructor it crashes

Keruspe, 2022-03-24:
End of output without LD_DEBUG:

```
2606|ptp2                |PTP                 |Zuk Z2 Pro
---+--------------------+--------------------+-------------------------------------------
No.|camlib              |driver name         |camera model
zsh: segmentation fault (core dumped)  /usr/lib/libgphoto2/print-camera-list human-readable
```

With what you asked:
```
keruspe@Lou ~ % LD_DEBUG=all /usr/lib/libgphoto2/print-camera-list human-readable >logfile 2>&1                     
zsh: segmentation fault (core dumped)  LD_DEBUG=all /usr/lib/libgphoto2/print-camera-list human-readable > logfile 2
```

End of logfile:
```
2592|ptp2                |PTP                 |ZiiLABS Zii EGG
2593|digigr8              229798:       symbol=gp_abilities_list_free;  lookup in file=/usr/lib/libgphoto2/print-camera-list [0]
    229798:     symbol=gp_abilities_list_free;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libgphoto2.so.6 [0]
    229798:     binding file /usr/lib/libgphoto2/print-camera-list [0] to /usr/x86_64-pc-linux-gnu/lib/libgphoto2.so.6 [0]: normal symbol `gp_abilities_list_free'
    229798:
    229798:     calling fini: /usr/lib/libgphoto2/print-camera-list [0]
    229798:
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/lib/libgphoto2/print-camera-list [0]
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libgphoto2.so.6 [0]
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libgphoto2_port.so.12 [0]
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libc.so.6 [0]
    229798:     binding file /usr/lib/libgphoto2/print-camera-list [0] to /usr/x86_64-pc-linux-gnu/lib/libc.so.6 [0]: normal symbol `__cxa_finalize' [GLIBC_2.2.5]
    229798:
    229798:     calling fini: /usr/x86_64-pc-linux-gnu/lib/libgphoto2.so.6 [0]
    229798:
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/lib/libgphoto2/print-camera-list [0]
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libgphoto2.so.6 [0]
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libgphoto2_port.so.12 [0]
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libc.so.6 [0]
    229798:     binding file /usr/x86_64-pc-linux-gnu/lib/libgphoto2.so.6 [0] to /usr/x86_64-pc-linux-gnu/lib/libc.so.6 [0]: normal symbol `__cxa_finalize' [GLIBC_2.2.5]
    229798:
    229798:     calling fini: /usr/x86_64-pc-linux-gnu/lib/libgphoto2_port.so.12 [0]
    229798:
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/lib/libgphoto2/print-camera-list [0]
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libgphoto2.so.6 [0]
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libgphoto2_port.so.12 [0]
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libc.so.6 [0]
    229798:     binding file /usr/x86_64-pc-linux-gnu/lib/libgphoto2_port.so.12 [0] to /usr/x86_64-pc-linux-gnu/lib/libc.so.6 [0]: normal symbol `__cxa_finalize' [GLIBC_2.2.5]
    229798:
    229798:     calling fini: /usr/x86_64-pc-linux-gnu/lib/libltdl.so.7 [0]
    229798:
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/lib/libgphoto2/print-camera-list [0]
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libgphoto2.so.6 [0]
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libgphoto2_port.so.12 [0]
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libc.so.6 [0]
    229798:     binding file /usr/x86_64-pc-linux-gnu/lib/libltdl.so.7 [0] to /usr/x86_64-pc-linux-gnu/lib/libc.so.6 [0]: normal symbol `__cxa_finalize' [GLIBC_2.2.5]
    229798:
    229798:     calling fini: /usr/x86_64-pc-linux-gnu/lib/libexif.so.12 [0]
    229798:
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/lib/libgphoto2/print-camera-list [0]
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libgphoto2.so.6 [0]
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libgphoto2_port.so.12 [0]
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libc.so.6 [0]
    229798:     binding file /usr/x86_64-pc-linux-gnu/lib/libexif.so.12 [0] to /usr/x86_64-pc-linux-gnu/lib/libc.so.6 [0]: normal symbol `__cxa_finalize' [GLIBC_2.2.5]
    229798:
    229798:     calling fini: /usr/x86_64-pc-linux-gnu/lib/libm.so.6 [0]
    229798:
    229798:
    229798:     calling fini: /usr/x86_64-pc-linux-gnu/lib/libdl.so.2 [0]
    229798:
    229798:
    229798:     calling fini: /usr/x86_64-pc-linux-gnu/lib/libglib-2.0.so.0 [0]
    229798:
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/lib/libgphoto2/print-camera-list [0]
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libgphoto2.so.6 [0]
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libgphoto2_port.so.12 [0]
    229798:     symbol=__cxa_finalize;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libc.so.6 [0]
    229798:     binding file /usr/x86_64-pc-linux-gnu/lib/libglib-2.0.so.0 [0] to /usr/x86_64-pc-linux-gnu/lib/libc.so.6 [0]: normal symbol `__cxa_finalize' [GLIBC_2.2.5]
    229798:     symbol=__deregister_frame_info;  lookup in file=/usr/lib/libgphoto2/print-camera-list [0]
[... cut ...]
    229798:     symbol=__deregister_frame_info;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/ld-linux-x86-64.so.2 [0]
    229798:     symbol=__deregister_frame_info;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libdl.so.2 [0]
    229798:     symbol=__deregister_frame_info;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libglib-2.0.so.0 [0]
    229798:     symbol=__deregister_frame_info;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libpcre.so.1 [0]
    229798:     symbol=__deregister_frame_info;  lookup in file=/usr/x86_64-pc-linux-gnu/lib/libc.so.6 [0]
```

Keruspe, 2022-03-24:
Just to double confirm I added a sleep after the call to `iterate_camera_list`, before returning. I crashes *after* the sleep, so it really is during final cleanup

Keruspe, 2022-03-24:
Valgrind take on this:

```
==269402== Jump to the invalid address stated on the next line
==269402==    at 0x0: ???
==269402==    by 0x493AC77: __run_exit_handlers (in /usr/x86_64-pc-linux-gnu/lib/libc.so.6)
==269402==    by 0x493AD7D: exit (in /usr/x86_64-pc-linux-gnu/lib/libc.so.6)
==269402==    by 0x4923654: (below main) (in /usr/x86_64-pc-linux-gnu/lib/libc.so.6)
==269402==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==269402== 
==269402== 
==269402== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==269402==  Bad permissions for mapped region at address 0x0
==269402==    at 0x0: ???
==269402==    by 0x493AC77: __run_exit_handlers (in /usr/x86_64-pc-linux-gnu/lib/libc.so.6)
==269402==    by 0x493AD7D: exit (in /usr/x86_64-pc-linux-gnu/lib/libc.so.6)
==269402==    by 0x4923654: (below main) (in /usr/x86_64-pc-linux-gnu/lib/libc.so.6)
==269402== 
==269402== HEAP SUMMARY:
==269402==     in use at exit: 33,207 bytes in 44 blocks
==269402==   total heap usage: 1,471 allocs, 1,427 frees, 95,060,022 bytes allocated
==269402== 
==269402== LEAK SUMMARY:
==269402==    definitely lost: 0 bytes in 0 blocks
==269402==    indirectly lost: 0 bytes in 0 blocks
==269402==      possibly lost: 4,608 bytes in 2 blocks
==269402==    still reachable: 28,599 bytes in 42 blocks
==269402==         suppressed: 0 bytes in 0 blocks
==269402== Rerun with --leak-check=full to see details of leaked memory
==269402== 
==269402== For lists of detected and suppressed errors, rerun with: -s
==269402== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)

```

Keruspe, 2022-03-24:
Oh, interresting! Let's go up the chain of dependencies

```
file=libglib-2.0.so.0 [0];  needed by /usr/x86_64-pc-linux-gnu/lib/libharfbuzz.so.0 [0] 
file=libharfbuzz.so.0 [0];  needed by /usr/x86_64-pc-linux-gnu/lib/libfreetype.so.6 [0]
file=libfreetype.so.6 [0];  needed by /usr/x86_64-pc-linux-gnu/lib/libgd.so.3 [0]
file=libgd.so.3 [0];  needed by /usr/x86_64-pc-linux-gnu/lib/libgphoto2/2.5.29/ax203.so [0] 
```

Keruspe, 2022-03-24:
Rebuilding freetype with harfbuzz disabled for testing, the last fini called is now from libssl, because of curl, because of the lumix module, and still the same problem, so glib wasn't the culprit

Keruspe, 2022-03-24:
Just tried out `gphoto2 --list-cameras` and it hits the same thing

msmeissn, 2022-03-25:
hmm, my libgd is not linked against harfbuzz and glib2 , this is probably why it so far does not reproduce for me. i will need to try this on my side :/ 
until then you could remove the ax203.so  file if you do not use this picture frame to make your setup work as a workaround

kreaninw, 2022-07-26:
I am also having this issue on Fedora 36 (libgphoto2 2.5.30). 

`gphoto2 --auto-detect` detect my camera model correctly (Olympus E-M1)

`gphoto2 --abilities` run fine. But `gphoto2 --summary` returns `Segmentation fault (core dumped)`.

And downgrading to the latest working version (libgphoto2 2.5.27) is not possible on fedora, since only libgphoto2 2.5.29 is available for downgrading. Therefore, I can't use my camera as my webcam in Fedora 36 at all. On Ubuntu 22.04 LTS, libgphoto2 is still at 2.5.27, so it's working fine.

Please fix this, thanks.

msmeissn, 2022-07-26:
This crash might be unrelated

can you try this in gdb? 

gdb gphoto2
r --summary
(should crash)
bt

kreaninw, 2022-07-26:
> 

This is my return from running the gdb:

```
$ gdb gphoto2
GNU gdb (GDB) Fedora 12.1-1.fc36
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-redhat-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from gphoto2...

This GDB supports auto-downloading debuginfo from the following URLs:
https://debuginfod.fedoraproject.org/ 
Enable debuginfod for this session? (y or [n]) y
Debuginfod has been enabled.
To make this setting permanent, add 'set debuginfod enabled on' to .gdbinit.
Downloading 0.35 MB separate debug info for /home/archerallstars/gphoto2
Reading symbols from /home/archerallstars/.cache/debuginfod_client/c9e7df8eeb36c2179106eb32d978d83698cd9884/debuginfo...
(gdb) r --summary
Starting program: /usr/bin/gphoto2 --summary
Downloading 0.03 MB separate debug info for system-supplied DSO at 0x7ffff7fc4000
Downloading 0.38 MB separate debug info for /lib64/libgphoto2.so.6             
Downloading 0.07 MB separate debug info for /home/archerallstars/.cache/debuginfod_client/ed48b0ecf0bedcf6f2a4fe2e5d2e107d62cec66f/debuginfo
Downloading 0.11 MB separate debug info for /lib64/libgphoto2_port.so.12       
Downloading 0.49 MB separate debug info for /lib64/libexif.so.12               
Downloading 0.95 MB separate debug info for /lib64/libreadline.so.8            
Downloading 0.02 MB separate debug info for /home/archerallstars/.cache/debuginfod_client/524a24317a37b45b87ba67e30845656ef9ce3476/debuginfo
Downloading 0.18 MB separate debug info for /lib64/libpopt.so.0                
Downloading 0.39 MB separate debug info for /lib64/libgcc_s.so.1               
Downloading 40.56 MB separate debug info for /home/archerallstars/.cache/debuginfod_client/1e82df1fa0e0de4d8382b75c0fa730e0672f1a47/debuginfo
Downloading 7.41 MB separate debug info for /lib64/libc.so.6                   
[Thread debugging using libthread_db enabled]                                  
Using host libthread_db library "/lib64/libthread_db.so.1".
Downloading 0.12 MB separate debug info for /lib64/libltdl.so.7
Downloading 2.25 MB separate debug info for /lib64/libm.so.6                   
Downloading 0.38 MB separate debug info for /lib64/libtinfo.so.6               
Downloading 0.20 MB separate debug info for /home/archerallstars/.cache/debuginfod_client/954d12f7d8216fde821db122a4768ee255382a63/debuginfo
Downloading 0.03 MB separate debug info for /usr/lib64/libgphoto2_port/0.12.1/disk.so
Downloading 0.01 MB separate debug info for /usr/lib64/libgphoto2_port/0.12.1/ptpip.so
Downloading 0.04 MB separate debug info for /usr/lib64/libgphoto2_port/0.12.1/serial.so
Downloading 0.05 MB separate debug info for /lib64/liblockdev.so.1             
Downloading 0.01 MB separate debug info for /home/archerallstars/.cache/debuginfod_client/70e1cbaa5f99ef4b1c4442bcaac954dd5bc82121/debuginfo
Downloading 0.09 MB separate debug info for /usr/lib64/libgphoto2_port/0.12.1/usb1.so
Downloading 0.40 MB separate debug info for /lib64/libusb-1.0.so.0             
Downloading 0.01 MB separate debug info for /home/archerallstars/.cache/debuginfod_client/d2c387b3f8505ddfc3c28f3e495f5d633c179cb4/debuginfo
Downloading 1.15 MB separate debug info for /lib64/libudev.so.1                
Downloading 2.98 MB separate debug info for /home/archerallstars/.cache/debuginfod_client/002cce667c466b2f92130c99f55d21afa5669636/debuginfo
[New Thread 0x7fffea4fd640 (LWP 15841)]                                        
[Thread 0x7fffea4fd640 (LWP 15841) exited]
Downloading 0.04 MB separate debug info for /usr/lib64/libgphoto2_port/0.12.1/usbdiskdirect.so
Downloading 0.04 MB separate debug info for /usr/lib64/libgphoto2_port/0.12.1/usbscsi.so
Downloading 0.17 MB separate debug info for /usr/lib64/libgphoto2/2.5.30/ax203.so
Downloading 0.90 MB separate debug info for /lib64/libgd.so.3                  
Downloading 0.91 MB separate debug info for /lib64/libjpeg.so.62               
Downloading 0.15 MB separate debug info for /home/archerallstars/.cache/debuginfod_client/efe241196bd7d87a220b593b4aac911a1245eaa0/debuginfo
Downloading 0.83 MB separate debug info for /lib64/libpng16.so.16              
Downloading 0.24 MB separate debug info for /lib64/libz.so.1                   
Downloading 1.19 MB separate debug info for /lib64/libfontconfig.so.1          
[... 141 lines cut ...]
    data=0x7fffffff2de8, dt=<optimized out>) at ptp2/library.c:7323
#7  0x00007fffe9c49c75 in camera_summary (camera=<optimized out>, summary=<optimized out>, context=0x55555557b1d0) at ptp2/library.c:7819
#8  0x00007ffff7f2afcb in gp_camera_get_summary (camera=0x555555579460, summary=summary@entry=0x7fffffff2f50, context=0x55555557b1d0)
    at /usr/src/debug/libgphoto2-2.5.30-2.fc36.x86_64/libgphoto2/gphoto2-camera.c:1228
#9  0x000055555556ab94 in action_camera_summary.isra.0 (params=<optimized out>) at /usr/src/debug/gphoto2-2.5.27-3.fc36.x86_64/gphoto2/actions.c:827
#10 0x3334322c3639312c in ?? ()
[... cut ...]
#13 0x2c3533312c303831 in ?? ()
#14 0x322c39322c393231 in ?? ()
#15 0x33312c39392c3732 in ?? ()
#16 0x322c312c38352c36 in ?? ()
#17 0x3239312c33382c39 in ?? ()
#18 0x2c38342c3132312c in ?? ()
#19 0x2c3733322c313332 in ?? ()
#20 0x39312c34342c3039 in ?? ()
#21 0x32332c3135312c38 in ?? ()
#22 0x37312c30332c352c in ?? ()
#23 0x3332322c3035322c in ?? ()
#24 0x3035322c3935312c in ?? ()
#25 0x2c3930322c31362c in ?? ()
#26 0x312c32382c343432 in ?? ()
#27 0x34352c32352c3030 in ?? ()
#28 0x2c3230312c32342c in ?? ()
```

msmeissn, 2022-07-27:
can you in this backtrace, do

frame 8
print *summary

(Likely crash reason is that there is a static buffer, we have some buffer overflow checking, but here it seems to be buggy and the buffer actually overflows.)

kreaninw, 2022-07-28:
> 

Hi, here is my returns:

```
(gdb) frame 8
#8  0x00007ffff7f2afcb in gp_camera_get_summary (camera=0x555555579460, summary=summary@entry=0x7fffffff2f50, context=0x55555557b1d0)
    at /usr/src/debug/libgphoto2-2.5.30-2.fc36.x86_64/libgphoto2/gphoto2-camera.c:1228
Downloading 0.05 MB source file /usr/src/debug/libgphoto2-2.5.30-2.fc36.x86_64/libgphoto2/gphoto2-camera.c
1228            CHECK_RESULT_OPEN_CLOSE (camera, camera->functions->summary (camera,                                                                         
(gdb) print *summary
$1 = {
  text = "Manufacturer: OLYMPUS\nModel: E-M1\n  Version: 1.00\n  Serial Number: BFJ211600\nVendor Extension ID: 0xfffd (1.0)\nVendor Extension Description: \n\nCapture Formats: MS Wave MS AVI Unknown(b980) Defined Typ"...}
```

viviicat, 2022-11-14:
I am still having this issue. I attempted to do a git-bisect to figure out what commit caused the issue, but had difficulty doing so. And unfortunately debugging with GDB did not prove very useful due to the aforementioned issues.

This is causing me a lot of problems here on Gentoo.

ndim, 2022-11-14:
I have tried to reproduce the segfault on Fedora 37 by running `print-camera-list` from git HEAD. My `libgd.so.3` is linked against `libfreetype.so.6`, `libharfbuzz.so.0`, and `libglib-2.0.so.0` - but I cannot see any segfaults. What I do see, however, is when I run

```
valgrind --leak-check=full --show-leak-kinds=all -s /prefix/lib/libgphoto2/print-camera-list human-readable
```
there are a few memory leaks:

```
=443413== LEAK SUMMARY:
==443413==    definitely lost: 512 bytes in 16 blocks
==443413==    indirectly lost: 0 bytes in 0 blocks
==443413==      possibly lost: 0 bytes in 0 blocks
==443413==    still reachable: 108,415 bytes in 42 blocks
==443413==         suppressed: 0 bytes in 0 blocks
==443413== 
==443413== ERROR SUMMARY: 12 errors from 12 contexts (suppressed: 0 from 0)
```

Note that this is libgphoto2 compiled with `--with-camlibs=everything`, so there are more camlibs being loaded than in a default build.

Maybe I need to set up a VM running Gentoo. But then I am running Fedora 37, and there are reports from F35 and F36 with segfaults, so perhaps I should see the segfaults here as well?

viviicat, 2022-11-15:
When I have a moment, I will try to provide the full log including compile flags and link details, in case that helps.

Please let me know if there is any specific additional info I can provide, debugging operations to perform, or a better way to nail down a repro.

msmeissn, 2022-11-15:
we might have overrun the fixed length buffer in the summary code still. The logic seems fine in the code.

This part seems to underline it:  (there is no leading \0 but we cross a page boundary which is not allocated.
 (txt=0x7fffffffeffe "14"<error: Cannot access memory at address 0x7ffffffff000>

viviicat, 2022-11-21:
I get this too when running `--camera="Canon EOS 90D" --capture-image-and-download`.

So I wonder if it really is the summary code, does that code path get run with this set of args as well?

viviicat, 2022-11-21:
Good news. If I force the CAMERAS flags to be just "canon", I am able to build successfully. This suggests to me that it is one of the cameras. I will try to do a binary search to figure out which camera it might be.

viviicat, 2022-11-21:
This is strange. If I enable all cameras, I get the crash. If I remove one camera (and it doesn't seem to matter which!) it does not crash.

Perhaps there is a buffer affected by the number of cameras that is overflowing.

viviicat, 2022-11-21:
I was able to reproduce this with both gcc and clang. However when I enabled ASAN `-fsanitize=address` in an attempt to catch the error in action, it no longer produced an error at all. So that's a bit odd.

Let me know if someone has an idea for further experiements to narrow this down.

msmeissn, 2022-11-21:
fwiw the crash was in the ptp2 driver. 
anything if you run it with valgrind 
build with -DVALGRIND  in cflags before

viviicat, 2022-11-21:
Note that the error about "the camera cannot capture" is what I would expect as the camera is not actually attached right now (that doesn't seem to make a difference on whether the crash happens)

My valgrind with the canon options and built with -DVALGRIND
```
└─⦅ valgrind --leak-check=full gphoto2 --debug --debug-logfile=my-logfile.txt --camera="Canon EOS 90D" --capture-image-and-download
==1366703== Memcheck, a memory error detector
==1366703== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==1366703== Using Valgrind-3.19.0 and LibVEX; rerun with -h for copyright info
==1366703== Command: gphoto2 --debug --debug-logfile=my-logfile.txt --camera=Canon\ EOS\ 90D --capture-image-and-download
==1366703== 
                                                                               
*** Error ***              
This camera can not capture.
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***       

==1366703== Jump to the invalid address stated on the next line
==1366703==    at 0x0: ???
==1366703==    by 0x4C7F677: __run_exit_handlers (in /lib64/libc.so.6)
==1366703==    by 0x4C7F779: exit (in /lib64/libc.so.6)
==1366703==    by 0x4C68358: (below main) (in /lib64/libc.so.6)
==1366703==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==1366703== 
==1366703== 
==1366703== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==1366703==  Bad permissions for mapped region at address 0x0
==1366703==    at 0x0: ???
==1366703==    by 0x4C7F677: __run_exit_handlers (in /lib64/libc.so.6)
==1366703==    by 0x4C7F779: exit (in /lib64/libc.so.6)
==1366703==    by 0x4C68358: (below main) (in /lib64/libc.so.6)
==1366703== 
==1366703== HEAP SUMMARY:
==1366703==     in use at exit: 131,150 bytes in 189 blocks
==1366703==   total heap usage: 39,798 allocs, 39,609 frees, 116,918,302 bytes allocated
==1366703== 
==1366703== 15 bytes in 1 blocks are definitely lost in loss record 8 of 87
==1366703==    at 0x4843724: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==1366703==    by 0x4B5BC22: poptGetNextOpt (in /usr/lib64/libpopt.so.0.0.1)
==1366703==    by 0x11D45D: main (main.c:2239)
==1366703== 
==1366703== 15 bytes in 1 blocks are definitely lost in loss record 9 of 87
==1366703==    at 0x4843724: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==1366703==    by 0x4B5BC22: poptGetNextOpt (in /usr/lib64/libpopt.so.0.0.1)
==1366703==    by 0x11D557: main (main.c:2277)
==1366703== 
==1366703== 15 bytes in 1 blocks are definitely lost in loss record 10 of 87
==1366703==    at 0x4843724: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==1366703==    by 0x4B5BC22: poptGetNextOpt (in /usr/lib64/libpopt.so.0.0.1)
==1366703==    by 0x11D586: main (main.c:2283)
==1366703== 
==1366703== 15 bytes in 1 blocks are definitely lost in loss record 11 of 87
==1366703==    at 0x4843724: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==1366703==    by 0x4B5BC22: poptGetNextOpt (in /usr/lib64/libpopt.so.0.0.1)
==1366703==    by 0x11D60C: main (main.c:2322)
==1366703== 
==1366703== 15 bytes in 1 blocks are definitely lost in loss record 12 of 87
==1366703==    at 0x4843724: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==1366703==    by 0x4B5BC22: poptGetNextOpt (in /usr/lib64/libpopt.so.0.0.1)
==1366703==    by 0x11DB1A: main (main.c:2323)
==1366703== 
==1366703== 15 bytes in 1 blocks are definitely lost in loss record 13 of 87
==1366703==    at 0x4843724: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==1366703==    by 0x4B5BC22: poptGetNextOpt (in /usr/lib64/libpopt.so.0.0.1)
==1366703==    by 0x11DB41: main (main.c:2324)
==1366703== 
==1366703== 15 bytes in 1 blocks are definitely lost in loss record 14 of 87
==1366703==    at 0x4843724: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==1366703==    by 0x4B5BC22: poptGetNextOpt (in /usr/lib64/libpopt.so.0.0.1)
==1366703==    by 0x11E596: main (main.c:2514)
[... 63 lines cut ...]
[Thread 0x7ffff76466c0 (LWP 1367123) exited]
[New Thread 0x7ffff76466c0 (LWP 1367124)]
[Thread 0x7ffff76466c0 (LWP 1367124) exited]
[New Thread 0x7ffff76466c0 (LWP 1367125)]
[Thread 0x7ffff76466c0 (LWP 1367125) exited]
[New Thread 0x7ffff76466c0 (LWP 1367126)]
[Thread 0x7ffff76466c0 (LWP 1367126) exited]

*** Error ***              
This camera can not capture.
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***       

Thread 1 "gphoto2" received signal SIGSEGV, Segmentation fault.
0x0000000000000000 in ?? ()
(gdb) bt
#0  0x0000000000000000 in ?? ()
#1  0x00007ffff7fcf8a3 in _dl_fini () from /lib64/ld-linux-x86-64.so.2
#2  0x00007ffff7a3d678 in __run_exit_handlers () from /lib64/libc.so.6
#3  0x00007ffff7a3d77a in exit () from /lib64/libc.so.6
#4  0x00007ffff7a26359 in __libc_start_call_main () from /lib64/libc.so.6
#5  0x00007ffff7a2640c in __libc_start_main () from /lib64/libc.so.6
#6  0x000055555555aeb1 in _start ()
```

msmeissn, 2022-11-22:
Ok, i got a bit confused between users. The issue from @viviicat is a atexit handler that is NULL.

The issue from @kreaninw is likely a buffer overrun due to Olympus having a very long MTP summary.

viviicat, 2022-11-22:
And I believe my issue matches this particular issue's description, right?

msmeissn, 2022-11-22:
yes, NULL ptr deref in atexit is the description one too. :/ need to think of more debugging ways

msmeissn, 2022-11-23:
ok, bigger hammer. We run it in gdb, break at the atexit registering function and see what is getting registered

gdb gphoto2

break atexit
break on_exit
r --list-cameras
run

use "continue" when you hit a break point until the program finishes.

we need all entries looking like:
Breakpoint 1, atexit (func=func@entry=0x7ffff7174d7e <OPENSSL_cleanup>) at atexit.c:45
45      {

for me on openSUSE Tumbleweed it is only this one.

viviicat, 2022-11-23:
I'm currently having trouble breaking at `atexit` and/or `on_exit` -- even though I've rebuilt glibc with debugging symbols it doesn't seem to be finding those functions to break at. Trying to figure out how to do that properly on Gentoo

viviicat, 2022-11-23:
Okay if I break at `__internal_atexit` that works (presumably optimization happening).

However the function pointers passed in unfortunately do not have symbols, they're just raw function pointers:

```
(gdb) break __internal_atexit
Function "__internal_atexit" not defined.
Make breakpoint pending on future shared library load? (y or [n]) y
Breakpoint 2 (__internal_atexit) pending.
(gdb) break __on_exit
Function "__on_exit" not defined.
Make breakpoint pending on future shared library load? (y or [n]) y
Breakpoint 3 (__on_exit) pending.
(gdb) r --list-cameras
Starting program: /usr/bin/gphoto2 --list-cameras
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".

Breakpoint 2, __internal_atexit (func=0x7ffff7fce6e0 <_dl_fini>, arg=arg@entry=0x0, d=d@entry=0x0, listp=listp@entry=0x7ffff7c82820 <__exit_funcs>) at cxa_atexit.c:41
41	 assert (func != NULL);
(gdb) continue
Continuing.
Loading camera drivers from '/usr/lib64/libgphoto2/2.5.30'... |--------------------------------                                               \  40.0%          
Breakpoint 2, __internal_atexit (func=0x7ffff4e63450, arg=0x7ffff4fc7210, d=0x7ffff4fc7060, listp=0x7ffff7c82820 <__exit_funcs>) at cxa_atexit.c:41
41	 assert (func != NULL);
(gdb) 
Continuing.

Breakpoint 2, __internal_atexit (func=0x7ffff4e63420, arg=0x7ffff4fc7208, d=0x7ffff4fc7060, listp=0x7ffff7c82820 <__exit_funcs>) at cxa_atexit.c:41
41	 assert (func != NULL);
(gdb) 
Continuing.

Breakpoint 2, __internal_atexit (func=0x7ffff4e8ec00, arg=0x7ffff4fc71b0, d=0x7ffff4fc7060, listp=0x7ffff7c82820 <__exit_funcs>) at cxa_atexit.c:41
41	 assert (func != NULL);
(gdb) 
Continuing.

Breakpoint 2, __internal_atexit (func=0x7ffff4e90c20, arg=0x7ffff4fc71c8, d=0x7ffff4fc7060, listp=0x7ffff7c82820 <__exit_funcs>) at cxa_atexit.c:41
41	 assert (func != NULL);
(gdb) 
Continuing.

Breakpoint 2, __internal_atexit (func=0x7ffff4e90c30, arg=0x7ffff4fc71c0, d=0x7ffff4fc7060, listp=0x7ffff7c82820 <__exit_funcs>) at cxa_atexit.c:41
41	 assert (func != NULL);
(gdb) 
Continuing.

Breakpoint 2, __internal_atexit (func=0x7ffff4f322e0, arg=0x7ffff4fc71e8, d=0x7ffff4fc7060, listp=0x7ffff7c82820 <__exit_funcs>) at cxa_atexit.c:41
41	 assert (func != NULL);
(gdb) 
Continuing.

Breakpoint 2, __internal_atexit (func=0x7ffff4f322f0, arg=0x7ffff4fc71e0, d=0x7ffff4fc7060, listp=0x7ffff7c82820 <__exit_funcs>) at cxa_atexit.c:41
41	 assert (func != NULL);
(gdb) 
Continuing.

Breakpoint 2, __internal_atexit (func=0x7ffff4f32300, arg=0x7ffff4fc71d8, d=0x7ffff4fc7060, listp=0x7ffff7c82820 <__exit_funcs>) at cxa_atexit.c:41
41	 assert (func != NULL);
(gdb) 
Continuing.
Number of supported cameras: 2629                                                                                                                               
Supported cameras:
	" JL2005B/C/D camera" (EXPERIMENTAL)
	"A&K SE180"
	"A&K SR15"
    ...
```

viviicat, 2022-11-23:
Here it is with me running a backtrace each time:

```
(gdb) r --list-cameras
Starting program: /usr/bin/gphoto2 --list-cameras
\[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".

Breakpoint 2, __internal_atexit (func=0x7ffff7fce6e0 <_dl_fini>, arg=arg@entry=0x0, d=d@entry=0x0, listp=listp@entry=0x7ffff7c82820 <__exit_funcs>) at cxa_atexit.c:41
41	 assert (func != NULL);
(gdb) bt
#0  __internal_atexit (func=0x7ffff7fce6e0 <_dl_fini>, arg=arg@entry=0x0, d=d@entry=0x0, listp=listp@entry=0x7ffff7c82820 <__exit_funcs>) at cxa_atexit.c:41
#1  0x00007ffff7af691c in __GI___cxa_atexit (func=<optimized out>, arg=arg@entry=0x0, d=d@entry=0x0) at cxa_atexit.c:70
#2  0x00007ffff7ae03aa in __libc_start_main_impl (main=0x55555556b37e <main>, argc=2, argv=0x7fffffffde18, init=0x0, fini=<optimized out>, 
    rtld_fini=<optimized out>, stack_end=0x7fffffffde08) at ../csu/libc-start.c:333
#3  0x000055555555ae91 in _start ()
(gdb) continue
Continuing.
Loading camera drivers from '/usr/lib64/libgphoto2/2.5.30'... |--------------------------------                                               \  40.0%        5s
Breakpoint 2, __internal_atexit (func=0x7ffff4e63450, arg=0x7ffff4fc7210, d=0x7ffff4fc7060, listp=0x7ffff7c82820 <__exit_funcs>) at cxa_atexit.c:41
41	 assert (func != NULL);
(gdb) bt
#0  __internal_atexit (func=0x7ffff4e63450, arg=0x7ffff4fc7210, d=0x7ffff4fc7060, listp=0x7ffff7c82820 <__exit_funcs>) at cxa_atexit.c:41
#1  0x00007ffff4e61c9f in ?? () from /usr/lib/gcc/x86_64-pc-linux-gnu/11.3.0/libstdc++.so.6
#2  0x00007ffff7fceb2e in call_init (env=0x7fffffffde30, argv=0x7fffffffde18, argc=2, l=<optimized out>) at dl-init.c:70
#3  call_init (l=<optimized out>, argc=2, argv=0x7fffffffde18, env=0x7fffffffde30) at dl-init.c:26
#4  0x00007ffff7fcec14 in _dl_init (main_map=0x55555558b700, argc=2, argv=0x7fffffffde18, env=0x7fffffffde30) at dl-init.c:117
#5  0x00007ffff7bfeaad in __GI__dl_catch_exception (exception=<optimized out>, operate=<optimized out>, args=<optimized out>)
    at /usr/src/debug/sys-libs/glibc-2.36-r5/glibc-2.36/elf/dl-error-skeleton.c:182
#6  0x00007ffff7fd4e96 in dl_open_worker (a=a@entry=0x7fffffff13b0) at dl-open.c:808
#7  0x00007ffff7bfea50 in __GI__dl_catch_exception (exception=<optimized out>, operate=<optimized out>, args=<optimized out>)
    at /usr/src/debug/sys-libs/glibc-2.36-r5/glibc-2.36/elf/dl-error-skeleton.c:208
#8  0x00007ffff7fd521b in _dl_open (file=<optimized out>, mode=-2147483647, caller_dlopen=0x7ffff7ab7457, nsid=-2, argc=2, argv=0x7fffffffde18, 
    env=0x7fffffffde30) at dl-open.c:884
#9  0x00007ffff7b3a9d8 in dlopen_doit (a=a@entry=0x7fffffff1620) at dlopen.c:56
#10 0x00007ffff7bfea50 in __GI__dl_catch_exception (exception=exception@entry=0x7fffffff1580, operate=<optimized out>, args=<optimized out>)
    at /usr/src/debug/sys-libs/glibc-2.36-r5/glibc-2.36/elf/dl-error-skeleton.c:208
#11 0x00007ffff7bfeb0f in __GI__dl_catch_error (objname=0x7fffffff15d8, errstring=0x7fffffff15e0, mallocedp=0x7fffffff15d7, operate=<optimized out>, 
    args=<optimized out>) at /usr/src/debug/sys-libs/glibc-2.36-r5/glibc-2.36/elf/dl-error-skeleton.c:227
#12 0x00007ffff7b3a4b6 in _dlerror_run (operate=operate@entry=0x7ffff7b3a980 <dlopen_doit>, args=args@entry=0x7fffffff1620) at dlerror.c:138
#13 0x00007ffff7b3aa91 in dlopen_implementation (dl_caller=<optimized out>, mode=<optimized out>, file=<optimized out>) at dlopen.c:71
#14 ___dlopen (file=<optimized out>, mode=<optimized out>) at dlopen.c:81
#15 0x00007ffff7ab7457 in ?? () from /usr/lib64/libltdl.so.7
#16 0x00007ffff7ab4098 in ?? () from /usr/lib64/libltdl.so.7
#17 0x00007ffff7ab5eb1 in ?? () from /usr/lib64/libltdl.so.7
#18 0x00007ffff7ab65cc in lt_dlopenadvise () from /usr/lib64/libltdl.so.7
#19 0x00007ffff7ab6691 in lt_dlopenext () from /usr/lib64/libltdl.so.7
#20 0x00007ffff7f0f33a in gp_abilities_list_load_dir (list=0x555555585830, dir=0x7ffff7f2f346 "/usr/lib64/libgphoto2/2.5.30", context=0x5555555851a0)
    at /usr/src/debug/media-libs/libgphoto2-2.5.30/libgphoto2-2.5.30/libgphoto2/gphoto2-abilities-list.c:243
#21 0x00007ffff7f0f736 in gp_abilities_list_load (list=0x555555585830, context=0x5555555851a0)
    at /usr/src/debug/media-libs/libgphoto2-2.5.30/libgphoto2-2.5.30/libgphoto2/gphoto2-abilities-list.c:341
#22 0x0000555555566358 in gp_params_abilities_list (p=0x555555580be0 <gp_params>) at gp-params.c:303
#23 0x000055555555e651 in list_cameras_action (p=0x555555580be0 <gp_params>) at actions.c:570
#24 0x000055555556a8f9 in cb_arg_run (ctx=0x555555585230, reason=POPT_CALLBACK_REASON_OPTION, opt=0x7fffffffa7c0, arg=0x0, params=0x7fffffffa634)
    at main.c:1672
#25 0x000055555556b174 in cb_arg (ctx=0x555555585230, reason=POPT_CALLBACK_REASON_OPTION, opt=0x7fffffffa7c0, arg=0x0, data=0x7fffffffa634) at main.c:1873
#26 0x00007ffff7d69de7 in ?? () from /usr/lib64/libpopt.so.0
#27 0x00007ffff7d69e76 in ?? () from /usr/lib64/libpopt.so.0
#28 0x00007ffff7d6b5e7 in poptGetNextOpt () from /usr/lib64/libpopt.so.0
#29 0x000055555556f5b7 in main (argc=2, argv=0x7fffffffde18, envp=0x7fffffffde30) at main.c:2550
(gdb) continue
Continuing.

Breakpoint 2, __internal_atexit (func=0x7ffff4e63420, arg=0x7ffff4fc7208, d=0x7ffff4fc7060, listp=0x7ffff7c82820 <__exit_funcs>) at cxa_atexit.c:41
41	 assert (func != NULL);
(gdb) bt
#0  __internal_atexit (func=0x7ffff4e63420, arg=0x7ffff4fc7208, d=0x7ffff4fc7060, listp=0x7ffff7c82820 <__exit_funcs>) at cxa_atexit.c:41
[... cut ...]
#26 0x00007ffff7d69e76 in ?? () from /usr/lib64/libpopt.so.0
#27 0x00007ffff7d6b5e7 in poptGetNextOpt () from /usr/lib64/libpopt.so.0
#28 0x000055555556f5b7 in main (argc=2, argv=0x7fffffffde18, envp=0x7fffffffde30) at main.c:2550
(gdb) continue
Continuing.
Number of supported cameras: 2629                                                                                                                               
Supported cameras:
	" JL2005B/C/D camera" (EXPERIMENTAL)
	"A&K SE180"
	"A&K SR15"
```

viviicat, 2022-11-23:
Going to try to build without optimizations to hopefully get more info here, lots of optimizing happening.

viviicat, 2022-11-23:
The filename passed to lt_dlopenext is "/usr/lib64/libgphoto2/2.5.30/lumix" for all of the breakpoints hit.

msmeissn, 2022-11-24:
if you do break atexit  before run it will say it does not know, but you can still set it:

(gdb) break atexit
Function "atexit" not defined.
Make breakpoint pending on future shared library load? (y or [n]) y
Breakpoint 1 (atexit) pending.
(gdb)

does it then crash at the end?

msmeissn, 2022-11-24:
also if you move the lumix.so driver away, does that fix the crashes?

msmeissn, 2022-11-24:
(your traces itself look good, nowhere NULL is passed in ... but it is a lower level proxy function, the NULL ptr might be hiding in higher layers.)

YangKeao, 2023-11-04:
I think I have found the direct cause of this Segmentation Fault. It faults in [`__do_global_dtors_aux`](https://github.com/gcc-mirror/gcc/blob/eb83605be3db9e8246c73755eafcac5df32ddc69/libgcc/crtstuff.c#L434C50-L434C50). After loading `lumix.so` or `ptp2.so`, which depends on `libgcc_s.so`, the fini function of `libssl.so` will try to call `__deregister_frame_info`:

```
  if (__deregister_frame_info)
    __deregister_frame_info (__EH_FRAME_BEGIN__);
```

However, the `ld.so` (redirected from PLT of `__deregister_frame_info`) failed to lookup the function `__deregister_frame_info` and gives an 0, so calling 0 causes the segmentation fault. The backtrace is like the following. It's suggested to (conditionally) break in `dl-call_fini.c` and use `stepi`/`nexti` to reach this step, as it will give a more accurate position:

```
(gdb) nexti
156             jmp *%r11               # Jump to function address.
(gdb) info registers
r11            0x0                 0
(gdb) bt
#0  _dl_runtime_resolve_xsavec () at ../sysdeps/x86_64/dl-trampoline.h:156
#1  0x00007ffff7fce0e2 in _dl_call_fini (closure_map=closure_map@entry=0x55555555d480) at dl-call_fini.c:43
#2  0x00007ffff7fd1ecd in _dl_fini () at dl-fini.c:114
#3  0x00007ffff7c46375 in ?? () from /usr/lib64/libc.so.6
#4  0x00007ffff7c464ca in exit () from /usr/lib64/libc.so.6
#5  0x00007ffff7c2def1 in ?? () from /usr/lib64/libc.so.6
#6  0x00007ffff7c2dfa5 in __libc_start_main () from /usr/lib64/libc.so.6
#7  0x00005555555556f1 in _start ()
```

I'm not very familiar with GOT, PLT things, so I still have three uncertain parts:

1. Why is the `__deregister_frame_info` in GOT written a weird value, so `if (__deregister_frame_info)` condition meets and try to call `__deregister_frame_info (__EH_FRAME_BEGIN__);`. The related GOT entry is weird: `00000009b6f8  000000000008 R_X86_64_RELATIVE                    2356d`
2. Why cannot the `ld.so`  see `libgcc_s.so` and cannot find `__deregister_frame_info`? Is the order of uninstalling shared library wrong?
3. Why is `libssl.so` affected but other libraries linked with crt (nearly every one) are not? I have checked that the `__deregister_frame_info` in `libpcre` GOT is still zero, so it doesn't try to find `__deregister_frame_info` through PLT and doesn't panic when calling the `fini_array` of `libpcre`.

I have two workarounds for the end-users:

1. Use `LD_BIND_NOW=1`, so that all `PLT` lookup are done during bootstrap. 
2. Don't compile/install with `lumix.so` and `ptp2.so`

I'll need more time to understand deeper about this issue (re-learn about the dynamic-linking) and propose a fix.

YangKeao, 2023-11-08:
In my case, I'm using a nearly full LLVM / clang toolchain. However, the `libunwind` is mis-configured to be compiled with `libgcc`. After manually applying patch (removing every `-lgcc_s` in the build script) to avoid linking `libunwind` with `libgcc_s.so`, this problem disappeared. 

I guess there is some conflicts between `libgcc` and `compiler-rt` and they caused such a mystery behavior (but I'm not 100% sure). This library behaves quite well (at least in my environment) :+1: .