gphoto/libgphoto2 #778: Improve eosremoterelease choices

transistortim, 2022-03-23:
**Is your feature request related to a problem? Please describe.**
In my setup I need to capture photos with a Canon camera without triggering autofocus, which is why I dug into eosremoterelease.
I found two issues dealing with autofocus and eosremoterelease settings, [here](https://github.com/gphoto/libgphoto2/issues/695) and over at [gphoto2](https://github.com/gphoto/gphoto2/issues/161). After some try & error and problems with localisation I got it working for my use-case.
Nevertheless, in my opinion the eosremoterelease choice names are non-self-explanatory for users not looking into the low-level communication (info on AF is missing, Press 1, 2, 3). I felt invited by the comment [/* debugging */](https://github.com/gphoto/libgphoto2/blob/a06093e4953f1d48bb5fcc937ec7e08242c0fdf5/camlibs/ptp2/config.c#L7463) to do some tests and compile information from various places, see below.

**Describe the solution you'd like**
Draft proposal for new eosremoterelease choices:
choice name         | remotereleaseon | remotereleaseoff
--------------------|-----------------|-----------------
Half-Press no AF    | 1, 1            |
Full-Press no AF*   | 2, 1*           |
Instant-Press no AF | 3, 1            |
Half-Press AF       | 1, 0            |
Full-Press AF       | 2, 0            |
Instant-Press AF    | 3, 0            |
Half-Release        |                 | 1
Full-Release        |                 | 2
Instant-Release     |                 | 3

*remotereleaseon `2, 1` is currently missing from eosremoterelease options. If there is a logic behind this, this should be Full-Press without AF.
"Full-Press" means pressing the button "the other half" (1/2 -> 1) after already having performed a "Half-Press" (0 -> 1/2). "Instant-Press" is the instant press without halting halfway (0 -> 1). I'm not a native speaker, maybe someone knows better fitting terms.

**Describe alternatives you've considered**
I'm open to discussion.

**Additional context**
To understand eosremoterelease, first of all I looked at capturing. What [camera_trigger_canon_eos_capture](https://github.com/gphoto/libgphoto2/blob/a06093e4953f1d48bb5fcc937ec7e08242c0fdf5/camlibs/ptp2/library.c#L5804) does is:

* Regular EOS (not M):
    * `ptp_canon_eos_remotereleaseon (params, 1, 0), _("Canon EOS Half-Press failed")`
    * `ptp_canon_eos_remotereleaseon (params, 2, 0), _("Canon EOS Full-Press failed")`
    * `ptp_canon_eos_remotereleaseoff (params, 2), _("Canon EOS Full-Release failed")`
    * `ptp_canon_eos_remotereleaseoff (params, 1), _("Canon EOS Half-Release failed")`
* EOS M:
    * `ptp_canon_eos_remotereleaseon (params, 3, 0), _("Canon EOS M Full-Press failed")`
    * `ptp_canon_eos_remotereleaseoff (params, 3), _("Canon EOS M Full-Release failed")`

The remotereleaseon/-off parameters for `eosremoterelease` are defined [here](https://github.com/gphoto/libgphoto2/blob/a06093e4953f1d48bb5fcc937ec7e08242c0fdf5/camlibs/ptp2/config.c#L7494). They are listed in the following table and cross-referenced to the names in the debug messages of `camera_trigger_canon_eos_capture`.

Choice name | remotereleaseon | remotereleaseoff | Name in capture debug msg | Note
------------|-----------------|------------------|---------------------------|---------------------
Press Half  | 1, 1            |                  | not used                  |
Press Full  | 3, 1            |                  | not used                  |
Immediate   | 1, 1; 3, 1      |                  | not used                  |
Press 1     | 1, 0            |                  | EOS Half-Press            |
Press 2     | 2, 0            |                  | EOS Full-Press            |
Press 3     | 3, 0            |                  | EOS M Full-Press          |
Release 1   |                 | 1                | EOS Half-Release          | same as Release Half
Release 2   |                 | 2                | EOS Full-Release          |
Release 3   |                 | 3                | EOS M Full-Release        | same as Release Full
Release Half|                 | 1                | EOS Half-Release          |
Release Full|                 | 3                | EOS M Full-Release        |

It seems the values `1` & `2` are valid for EOS, while `3` is valid for EOS M. This is also mentioned in the code of [ptpy - a Python implementation of PTP](https://github.com/Parrot-Developers/sequoia-ptpy/blob/abf0004d608c069cc40e29da8903718e29950524/ptpy/extensions/canon/canon.py#L835).
I do not think there is a difference between EOS and EOS M, because...
* In the autofocus issues mentioned above, the camera models working with "Immediate" are EOS Rebel SL3 (EOS 250D where I live), Canon EOS 4000D and Canon EOS 1200D. As these are no EOS M models and "Immediate" uses remotereleaseon with value `3`, this contradicts the assumption of `3` being only for EOS M.
* EOS 7D user claims `3` works as Full-Press on [mailing list](https://sourceforge.net/p/gphoto/mailman/gphoto-devel/thread/4FFEB282.7070007%40gmail.com/#msg29531288).
* In my own tests with EOS M6 Mark II and EOS M50, all `1`, `2` and `3` were valid.

I'm pretty sure the second remotereleaseon parameter is AF, because
* `camera_trigger_canon_eos_capture` (second parameter `0`) triggers AF, as reported in the autofocus issues. Using eosremoterelease with second parameter `1` does not trigger AF.
* This also seems to be indicated by the Canon EDSDK. There is already a [comment on this in the libgphoto source code](https://github.com/gphoto/libgphoto2/blob/a06093e4953f1d48bb5fcc937ec7e08242c0fdf5/camlibs/ptp2/config.c#L7477).

Results of my own tests with EOS M6 Mark II and EOS M50 (both show the same behaviour):

Choice name | Shutter triggered | AF triggered  | Note
------------|-------------------|---------------|-----
Press Half  | ❌                | ❌
Press Full  | ✅                | ❌
Immediate   | ✅                | ❌
Press 1     | ❌                | ✅
Press 2     | ⚡                | ⚡            | Error: PTP Device Busy (0x2019)
Press 2     | ✅                | ❌            | after calling Press 1 without releasing
Press 3     | ✅                | ✅
Release Half| ❌                | ❌
Release Full| ❌                | ❌
Release 1   | ❌                | ❌
Release 2   | ❌                | ❌
Release 3   | ❌                | ❌

Open questions:
* Could somebody test all choices with a "normal" EOS model?
    * Remember to release after pressing, else everything returns PTP Device Busy (0x2019).
* Is the existing "Immediate" required?
    * I do not think so, as performing a "Half-Press" shouldn't matter if an "Instant-Press" is performed afterwards.
    * EOS 7D user reporting on [mailing list](https://sourceforge.net/p/gphoto/mailman/gphoto-devel/thread/4FFEB282.7070007%40gmail.com/#msg29531288) its unnecessary, "Instant-Press" suffices.
    * On M6 Mark II and M50 also "Instant-Press" suffices.
* Is there a difference between the Release commands?
    * I did not notice anything. Sending "Press 1" to EOS M6 Mark II is indicated on the camera screen by settings being "locked", like when the button is manually pressed and held down halfway. This lock is released by every Release setting. (As they all seem to work, it does no harm keeping all of them.)
* In my naming proposal from above, should "Full-Press no AF" and "Full-Press AF" really mention AF?
    * Observation on M6 Mark II and M50: "Half-Press" has to be send first and triggers AF. If I manually defocus the lens afterwards and send "Full-Press AF", no additional AF is triggered. Nevertheless, due to the differing second parameter I think the commands belong to the group "AF" resp. "no AF". I vote for including both, as just having a single "Full-Press" doesn't fit in the scheme and would be confusing.

If the questions are resolved, I could help with the implementation.

msmeissn, 2022-04-03:
sorry for not getting back earlier. i am very busy these days :/ 

i think that the second argument includes autofocusing might be true, but i need to go into debugging this locally too.
i hope i can find the time somewhen in next weeks :/

Sija, 2024-10-02:
Would be great to have this sorted out.

axxel, 2024-10-07:
@transistortim if you are still around for feedback: nice work. I have a question regarding your Instant-Press suggestion: What is the expected/real behavior of sending just one `3,0` if the lens is out of focus? Does it focus or doesn't it?

Is your conclusion from your tests that sending a `3` is exactly equivalent of sending first a `1`, then immediately a `2`?

I'd be in support of removing the EOS-M special casing if not required. As a random data point: The very latest EOSUtility [USB traces](https://github.com/gphoto/libgphoto2/issues/1028#issuecomment-2386520797) from an R5m2 shows the interesting pattern (1,1), (1), (1,1), (2,1), (2), (download the image!), (1).