gphoto/libgphoto2 #781: Add Support for Canon EOS Rebel T8i

razorbladex401, 2022-03-25:
Please add support for Canon EOS Rebel T8i.

gphoto2 --summary
Camera summary:                                                                
Manufacturer: Canon.Inc
Model: Canon EOS Rebel T8i
  Version: 3-1.0.0
  Serial Number: 72537f4c9e4fc3a0ad46b59f62ac7884
Vendor Extension ID: 0xb (1.0)
Vendor Extension Description: 

Capture Formats: JPEG
Display Formats: Association/Directory, Script, DPOF, MS AVI, MS Wave, JPEG, CRW, Unknown(b103), Unknown(bf02), Defined Type, Unknown(b104), Unknown(b105)

Device Capabilities:
	File Download, File Deletion, File Upload
	No Image Capture, No Open Capture, Canon EOS Capture, Canon EOS Capture 2
	Canon Wifi support

Storage Devices Summary:
store_00020001:
	StorageDescription: SD
	VolumeLabel: 
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Digital Camera Layout (DCIM)
	Access Capability: Read-Write
	Maximum Capability: 511801556992 (488092 MB)
	Free Space (Bytes): 506115915776 (482669 MB)
	Free Space (Images): -1

Device Property Summary:
Property 0xd402:(read only) (type=0xffff) 'Canon EOS Rebel T8i'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd406:(readwrite) (type=0xffff) 'Unknown Initiator'
Property 0xd303:(read only) (type=0x2) 1
Battery Level(0x5001):(read only) (type=0x2) Enumeration [100,0,75,0,50] value: 100% (100)

 $ lsusb
Bus 002 Device 002: ID 03c3:120e  
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 04a9:32f1 Canon, Inc. 
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

 $ gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
USB PTP Class Camera           usb:001,003     

When I try to capture an image I can hear a quick chirp from the camera and the lcd goes black for 1 sec but ultimately it fails.
 $ gphoto2 --capture-image-and-download
ERROR: Could not capture image.                                                
ERROR: Could not capture.

msmeissn, 2022-03-25:
can you state the USB ids of this Canon camera?

output of:
lsusb

razorbladex401, 2022-03-25:
Bus 001 Device 003: ID 04a9:32f1 Canon, Inc.

msmeissn, 2022-04-03:
can you run
gphoto2 --capture-image --debug --debug-logfile=capture.log

and attach capture.log if it still fails?

razorbladex401, 2022-04-10:
See attached
[capture.log](https://github.com/gphoto/libgphoto2/files/8459277/capture.log)