gphoto/libgphoto2 #783: Camera detection fails

milovangudelj, 2022-03-28:
I haven't been able to detect any camera attached to my computer using the code provided in the examples. The command line tool is able to detect and interact with my camera without any issues tho. Are the examples outdated or am I doing something wrong?

Here's my `autodetect` function:
```c++
void autodetect()
{
	GPContext *context = sample_create_context();
	Camera *camera;
	CameraText text;
	int ret;
	char *owner;

	std::cout << "Creating camera...\n";
	gp_camera_new(&camera);

	/* This call will autodetect cameras, take the
	 * first one from the list and use it. It will ignore
	 * any others... See the *multi* examples on how to
	 * detect and use more than the first one.
	 */
	ret = gp_camera_init(camera, context);
	if (ret < GP_OK)
	{
		printf("No camera auto detected.\n");
		gp_camera_free(camera);
		exit(0);
	}

	/* Simple query the camera summary text */
	ret = gp_camera_get_summary(camera, &text, context);
	if (ret < GP_OK)
	{
		printf("Camera failed retrieving summary.\n");
		gp_camera_free(camera);
		exit(0);
	}
	printf("Summary:\n%s\n", text.text);

	/* Simple query of a string configuration variable. */
	ret = get_config_value_string(camera, "owner", &owner, context);
	if (ret >= GP_OK)
	{
		printf("Owner: %s\n", owner);
		free(owner);
	}

	gp_camera_exit(camera, context);
	gp_camera_free(camera);
	gp_context_unref(context);
}
```
where `sample_create_context` and `get_config_value_string` come from [context.c](https://github.com/gphoto/libgphoto2/blob/master/examples/context.c) and [config.c](https://github.com/gphoto/libgphoto2/blob/master/examples/config.c) respectively.

Running the function prints this to the terminal:

```shell
Creating camera...

*** Context error ***
Could not detect any camera
No camera auto detected.
```

While running 
```shell
gphoto2 --auto-detect
```
returns
```shell
Model                          Port                                            
----------------------------------------------------------
Canon EOS 1100D                usb:001,023     
```

Any help and/or suggestion is welcome and appreciated.

milovangudelj, 2022-04-06:
No updates on this? Let me know if you need more information to better understand the issue.

msmeissn, 2022-04-07:
are both program using the same libgphoto2?

ldd `which gphoto2`
ldd yourprogram

to see if it loads libgphoto2.so from the same location.

milovangudelj, 2022-04-07:
No, actually they link two different files. My program gets the library from where I installed it while gphoto2 gets it from `/lib64`.
Running `which gphoto2` and then `ldd /usr/bin/gphoto2` outputs:
```shell
# which gphoto2
/usr/bin/gphoto2

# ldd /usr/bin/gphoto2
libgphoto2.so.6 => /lib64/libgphoto2.so.6 (0x00007f1c19c44000)
libgphoto2_port.so.12 => /lib64/libgphoto2_port.so.12 (0x00007f1c19c36000)
```
while running `ldd Documents/Projects/C++/webcam/build/webcam` outputs:
```shell
# ldd Documents/Projects/C++/webcam/build/webcam
libgphoto2.so.6 => /usr/local/lib/libgphoto2.so.6 (0x00007f21c170f000)
libgphoto2_port.so.12 => /usr/local/lib/libgphoto2_port.so.12 (0x00007f21c1701000)
```

If that's the issue, how can I tell CMake to use the library in `/lib64` instead of the one I built from source? It's been quite a fight to get CMake to work with gphoto2 and be able to use it at all. If you need to look at the code I can link you to my repository.

Thanks for helping! :innocent:

msmeissn, 2022-04-07:
It will likely help already if you install libusb 1 develoment header and libraries and rebuild your libgphoto2 copy.

milovangudelj, 2022-04-08:
Small update.

As suggested I installed libusb-devel (v0.1.7) and rebuilt libgphoto2 (v2.5.29) from scratch using the archive downloaded from the releases page and now the detection works as expected. Yay! :heart_eyes:

The summary output looks like this:
```shell
Creating context...
Creating camera...
Summary:
Manufacturer: Canon Inc.
Model: Canon EOS 1100D
  Version: 3-1.0.5
  Serial Number: # a long serial number here #
Vendor Extension ID: 0xb (2.0)
Vendor Extension Description: 

Capture Formats: JPEG
Display Formats: Association/Directory, Script, DPOF, MS AVI, MS Wave, JPEG, CRW, Unknown(b103), Unknown(bf02), Defined Type, Unknown(b104), Unknown(b105)

Device Capabilities:
        File Download, File Deletion, File Upload
        No Image Capture, No Open Capture, Canon EOS Capture, Canon EOS Capture 2

Storage Devices Summary:
store_00020001:
        StorageDescription: SD
        VolumeLabel: 
        Storage Type: Removable RAM (memory card)
        Filesystemtype: Digital Camera Layout (DCIM)
        Access Capability: Read-Write
        Maximum Capability: 31897157632 (30419 MB)
        Free Space (Bytes): 31726600192 (30256 MB)
        Free Space (Images): -1

Device Property Summary:
Property 0xd402:(read only) (type=0xffff) 'Canon EOS 1100D'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd406:(readwrite) (type=0xffff) 'Unknown Initiator'
Property 0xd303:(read only) (type=0x2) 1
Battery Level(0x5001):(read only) (type=0x2) Enumeration [100,0,75,0,50] value: 100% (100)

lookup widget failed: -2
``` 

The lookup widget failed and I'm not sure if that's bad or not tho...

msmeissn, 2022-04-10:
this means the camera has no "owner" field, so you probably need to replace it by "artist" or "copyright"

good to hear it worked otherwise,  I think we can close

closed 2022-04-10 by msmeissn